aws/amazon-freertos #3220: [BUG] Nordic CMake Generating Build Files hangs after C compiler identification

dtony, 2021-06-11:
**Describe the bug**
I am trying to use CMake to generates build files for the Nordic Demo but the process is stucked just after the C compiler identification.

**System information**
- Hardware board: nrf52840-dk
- IDE used: none
- Operating System: Windows 10
- Code version: 20210526_Archive-24-gf8a1fda60
- Project/Demo: MQTT_BLE_TRANSPORT_DEMO

**Expected behavior**
Build files should be generated

**Screenshots or console output**
```
cmake -DVENDOR=nordic -DBOARD=nrf52840-dk -DCOMPILER=arm-segger -DAFR_ENABLE_ALL_MODULES=1 -DAFR_ENABLE_TESTS=0 -DAFR_TOOLCHAIN_PATH="C:\Program Files\SEGGER\SEGGER Embedded Studio for ARM 5.50a\gcc\arm-none-eabi" -S ..\amazon-freertos\ -B . -G"Unix Makefiles"
-- The C compiler identification is GNU 10.2.1
```

**Steps to reproduce bug**
Try the same procedure on Windows ?

**Additional context**
I trued different version of Segger Arm Compiler 64 and 32bit, latest 5.50a but also 5.32a anb 4.42b

mingyue86010, 2021-06-14:
Hi @dtony, 

I setup the windows environment as the getting start guide and it looks it works. I have cmake version of 3.17.2 and Segger Embedded Studio v5.50a windows 64-bit. The command I used is `cmake -DVENDOR=nordic -DBOARD=nrf52840-dk -DCOMPILER=arm-segger -DAFR_ENABLE_ALL_MODULES=1 -DAFR_ENABLE_TESTS=0 -S . -B ./build -G"Unix Makefiles" -DAFR_TOOLCHAIN_PATH="C:\Program Files\SEGGER\SEGGER Embedded Studio for ARM 5.50a\gcc\arm-none-eabi"` which is build from root of amazon-freertos. But I tried build from the outside folder as your command and it also works...

›
[image]

Regards,
Ming

dtony, 2021-06-14:
I just downlgraded to CMake 3.17.2 and it works ! It is probably a bug in most recent release of CMake thank you for pointing me in the right direction.

mingyue86010, 2021-06-15:
Cool, that's Great!

closed 2021-06-15 by mingyue86010