aws/amazon-freertos #3223: Build, flash, and run the FreeRTOS demo project in Aws esp32 in VS code

aditi230, 2021-06-13:
**Describe the bug**
Issue at step Hey, Build, flash, and run the FreeRTOS demo project" . I followed [https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf33] this documentation.

**System information**
- Hardware board: [esp32]

- Operating System: [ Windows  ]

**Expected behavior**
I am stuck at CMake step could not figure out how to proceed .Do I require VS Code for it.

mingyue86010, 2021-06-13:
Hello @aditi230 ,

Could you provide more information about what probelm you have? If possiable could you paste the command you use and the error result you got? The VS Code is not required. Once you set up correctly, you can directly run the Cmake command from the amazon-freertos root derectory from a terminal.

Regards,
Ming

aditi230, 2021-06-14:
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B your-build-directory 

In this command what should be written under your-build-directory and also I am using esp32_devkitC

mingyue86010, 2021-06-14:
Hi @aditi230, 

the -B option is to set the folder location (can be a new folder which will be created during the build process) where to store the building artifacts and the build image. For example, if you build from the amazon-root directory and want to save the image under "./build" folder. Then you coulde set the -B as ./build. Then the command would be: 

`cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ./build`

Regards,
Ming

aditi230, 2021-06-15:
Got It ! Thank you

aditi230, 2021-06-15:
After this this is the error I ma facing.
[image]

other then that every thing is working fine
How will i resolve this.
[image]

aditi230, 2021-06-15:
Build and flash FreeRTOS on Windows, Linux, and MacOS (ESP-IDF v4.2)
Use the idf.py script to build the project and flash the binaries onto your device.

Note
Some setups might require that you use the port option "-p port-name" with idf.py to specify the correct port, as in the following example.

idf.py -p /dev/cu.usbserial-00101301B flash
To build and flash the project

Navigate to the root of your FreeRTOS download directory.

In a command line window, enter the following command to add the ESP-IDF tools to your terminal's PATH.

Windows

vendors\espressif\esp-idf\export.bat

This is giving me error:

`Setting IDF_PATH: I:\FreeRTOS\vendors\espressif\esp-idf

Adding ESP-IDF tools to PATH...
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "I:\FreeRTOS\vendors\espressif\esp-idf\tools\idf_tools.py", line 1358, in <module>
    main(sys.argv[1:])
  File "I:\FreeRTOS\vendors\espressif\esp-idf\tools\idf_tools.py", line 1354, in main
    action_func(args)
  File "I:\FreeRTOS\vendors\espressif\esp-idf\tools\idf_tools.py", line 980, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "I:\FreeRTOS\vendors\espressif\esp-idf\tools\idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Python39\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Python39\lib\subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', '--work-tree=I:\\FreeRTOS\\vendors\\espressif\\esp-idf', 'describe', '--tags']' returned non-zero exit status 128.`

closed 2021-06-24 by aditi230