aws/amazon-freertos #3225: OTA job fails on MQTT over BLE, while receiving OTA Packets from AWS on device, some packets are missed in between and complete file is not received.

kumarfirmware, 2021-06-14:
Hi Team,

**I am using AWS FreeRTOS version 202007 and ESP32.**

**I am seeing a Major/critical issue, Some times OTA jobs fail on MQTT over BLE, while receiving OTA Packets from AWS on device, some packets are missed in between and complete file is not received.**

For example If the device has to receive 40 OTA packets from cloud. File Block 0 received, File Block 1 received, File Block 2 received, suddenly File Block 38 received, File Block 39 received, File block 40 received. In between packets are missed. OTA is complete.

**This is disaster. This is stopping our production. This does not happen every time. This happens only some times.**

I need immediate solution for this. Why is this happening?  How this can happen? Is this a bug in the AWS OTA library? or Is this a bug in the mobile app?  I have never seen this kind of integrity issues in TCP. TCP takes care of integrity.

I suspect either it is a bug in AWS OTA library or it is a bug in mobile app.

On what basis cloud is sending packets? How the sequence of packets is maintained? Where is the chance of packet loss? 
who is taking care of the integrity? whether all the packets are received correctly or not who is taking care? Is it the AWS OTA library responsible for this? 

Is OTA library based on very weak algorithm, Is OTA library can be fooled so easily? File Block 0 received, File Block 1 received, File Block 2 received, suddenly File Block 38 received, File Block 39 received, File block 40 received How can OTA library say in this case OTA is complete and SUCCESS. There is no integrity check in OTA library?

How do I log OTA packets in mobile app? I need to confirm mobile app is not missing packets. Is it possible to log OTA packets in mobile app?

This needs immediate attention. Kindly help as soon as possible.

Thanks, Regards,
Kumar.

pvyawaha, 2021-06-14:
Hello @kumarfirmware,

The OTA library maintains state in form of a bitmap representing each each block and verifies the signature ( for authenticity and integrity )  of the file when complete file is received in the boards specific OTA PAL implementation. It can receive the blocks out of order. Are you observing that the number of blocks received are less than expected and crypto signature verification fails?

kumarfirmware, 2021-06-14:
Hi pvyawaha,

Thanks for your immediate response.

I am using ios application.

I am observing the number of blocks received are less than the expected clearly. Remember this is not always but considerable number of times.

I don't expect blocks to be received out of order, because I fetch one block at a time. This is regarding OTA for secondary processor. 

I am still not understanding, How is the integrity and authenticity verified for OTA for secondary processor when the complete file is received? How is it ensuring all the blocks received successfully? Where is the calculation done?

Do you rule out the bug in AWS OTA library? Do you think issue should be in the mobile app?

Thanks, Regards,
kumar.

ravibhagavandas, 2021-06-14:
Hi @kumarfirmware

It looks similar to the issue #3228 you have raised.

Which version/commit ID of the IOS SDK are you using ?

Mobile app by default logs the messages received from AWS before sending to device, and also messages from device before sending to AWS, into the terminal. Would you be able to run the mobile app through xcode and reproduce this issue ? If so, could you paste the logs from the xcode terminal ?

Also could you provide more details on the setup ? Which region is AWS cognito pool configured with ? Is the mobile app running in same region ? Is the AWS IoT broker endpoint in the same region ?

kumarfirmware, 2021-06-14:
Hi Ravi,

Thanks for your immediate response.

AWS cognito, Mobile app, AWS IoT broker endpoint, Device -  All of them are running in same region, us-east-1.

I can run my app on XCode. 

But I never see the messages from the cloud getting logged on the terminal and vice versa, from the device to the cloud.

Do you confirm we can see the messages even when the mobile app acts as just proxy between the device and cloud, i.e. MQTT over BLE.

or you mean to say, I need to subscribe to the messages explicitly in the mobile App?

Please confirm the same.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-06-14:
Hi Kumar

Here are the steps I followed to run IOS app:

Checked out SDK V1.2.0 > Opened Workspace `Example/AmazonFreeRTOSDemo/AmazonFreeRTOSDemo.xcworkspace` in Xcode 12.5 > Configured Cognito Credentials, IoT policy and region in the SDK > Build and Run code on iphone. I can see the app logs on xcode console in the bottom window:
Here are the logs when I run the shadow demo from the latest main commit in this repo:
```
2021-06-14 12:38:08.788949-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Connect(clientID: "TEST_DEVICE", brokerEndpoint: "a31zvyed820ljz-ats.iot.us-west-2.amazonaws.com", cleanSession: true)
2021-06-14 12:38:08.792127-0700 AmazonFreeRTOSDemo[429:10669] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] connectUsingWebSocket status: 1
2021-06-14 12:38:08.886927-0700 AmazonFreeRTOSDemo[429:10669] [connection] nw_endpoint_handler_set_adaptive_read_handler [C2.1 34.213.168.191:443 ready socket-flow (satisfied (Path is satisfied), viable, interface: en0, ipv4, dns)] unregister notification for read_timeout failed
2021-06-14 12:38:08.887082-0700 AmazonFreeRTOSDemo[429:10669] [connection] nw_endpoint_handler_set_adaptive_write_handler [C2.1 34.213.168.191:443 ready socket-flow (satisfied (Path is satisfied), viable, interface: en0, ipv4, dns)] unregister notification for write_timeout failed
2021-06-14 12:38:09.146312-0700 AmazonFreeRTOSDemo[429:10669] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Connack(messageType: 2, status: 2)
2021-06-14 12:38:09.262157-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Publish(messageType: 3, topic: "$aws/things/TEST_DEVICE/shadow/delete", msgID: 1, qoS: 1, payload: 0 bytes)
2021-06-14 12:38:09.293078-0700 AmazonFreeRTOSDemo[429:10669] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Puback(messageType: 4, msgID: 1)
2021-06-14 12:38:14.394436-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Subscribe(topics: ["$aws/things/TEST_DEVICE/shadow/update/delta"], msgID: 2, qoSs: [1])
2021-06-14 12:38:14.542501-0700 AmazonFreeRTOSDemo[429:10769] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Suback(messageType: 9, msgID: 2, status: 1)
2021-06-14 12:38:19.369642-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Subscribe(topics: ["$aws/things/TEST_DEVICE/shadow/update/accepted"], msgID: 3, qoSs: [1])
2021-06-14 12:38:19.422277-0700 AmazonFreeRTOSDemo[429:10669] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Suback(messageType: 9, msgID: 3, status: 1)
2021-06-14 12:38:24.469738-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Subscribe(topics: ["$aws/things/TEST_DEVICE/shadow/update/rejected"], msgID: 4, qoSs: [1])
2021-06-14 12:38:24.530990-0700 AmazonFreeRTOSDemo[429:10769] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Suback(messageType: 9, msgID: 4, status: 1)
2021-06-14 12:38:29.571776-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Publish(messageType: 3, topic: "$aws/things/TEST_DEVICE/shadow/update", msgID: 5, qoS: 1, payload: 59 bytes)
2021-06-14 12:38:29.605137-0700 AmazonFreeRTOSDemo[429:10854] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Puback(messageType: 4, msgID: 5)
2021-06-14 12:38:29.646597-0700 AmazonFreeRTOSDemo[429:10890] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Publish(messageType: 3, topic: "$aws/things/TEST_DEVICE/shadow/update/delta", msgID: 2, qoS: 1, payload: 129 bytes)
2021-06-14 12:38:29.649296-0700 AmazonFreeRTOSDemo[429:10889] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Publish(messageType: 3, topic: "$aws/things/TEST_DEVICE/shadow/update/accepted", msgID: 3, qoS: 1, payload: 153 bytes)
2021-06-14 12:38:34.822973-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Publish(messageType: 3, topic: "$aws/things/TEST_DEVICE/shadow/update", msgID: 6, qoS: 1, payload: 59 bytes)
2021-06-14 12:38:34.860786-0700 AmazonFreeRTOSDemo[429:10889] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Puback(messageType: 4, msgID: 6)
2021-06-14 12:38:34.910051-0700 AmazonFreeRTOSDemo[429:10898] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Publish(messageType: 3, topic: "$aws/things/TEST_DEVICE/shadow/update/accepted", msgID: 3, qoS: 1, payload: 155 bytes)
2021-06-14 12:38:39.983156-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Unsubscribe(topics: ["$aws/things/TEST_DEVICE/shadow/update/delta"], msgID: 7)
2021-06-14 12:38:39.984885-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Unsuback(messageType: 11, msgID: 7)
2021-06-14 12:38:45.020910-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Unsubscribe(topics: ["$aws/things/TEST_DEVICE/shadow/update/accepted"], msgID: 8)
2021-06-14 12:38:45.021251-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Unsuback(messageType: 11, msgID: 8)
2021-06-14 12:38:50.060987-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↑ Unsubscribe(topics: ["$aws/things/TEST_DEVICE/shadow/update/rejected"], msgID: 9)
2021-06-14 12:38:50.061605-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][MQTT] ↓ Unsuback(messageType: 11, msgID: 9)
[... cut ...]
2021-06-14 12:38:56.153700-0700 AmazonFreeRTOSDemo[429:10624] [AFR][BDA9F97C-169D-B206-BA9F-7D07F2BAA9B7][ERROR] afrCentralManagerDidDisconnectPeripheral: The specified device has disconnected from us.
```
 For a publish received from AWS and send to device you see a log starting with `↓ Publish`.

I think by default the SDK logging is disabled [here](https://github.com/aws/amazon-freertos-ble-ios-sdk/blob/master/AmazonFreeRTOS/AmazonFreeRTOSManager.swift#L13), could you enable sdk logging from your app as shown in demo app [here](https://github.com/aws/amazon-freertos-ble-ios-sdk/blob/master/Example/AmazonFreeRTOSDemo/AmazonFreeRTOSDemo/AppDelegate.swift#L14) ?

kumarfirmware, 2021-06-14:
Hi Ravi,

I will enable the logging in the app. I will test and come back with logs.

Thanks for your immediate response.

If I do not see logs, I will come back.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-06-28:
Hi @kumarfirmware 
Where you able to get the logs for the isssue?

ravibhagavandas, 2021-07-08:
I will close this issue due to inactivity. Please feel free to open a new issue if you have any queries.

closed 2021-07-08 by ravibhagavandas