aws/amazon-freertos #3226: Some times Shadow reporting/publishing API fails on device and shadow publishing fails

kumarfirmware, 2021-06-14:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

ESP32 is connected to the corporate network Wi-Fi. Very High speed network. No chance of unstable network.
Also ESP does not loose ping at all. Ping is very stable. 

When I am trying to send the shadow update to the cloud using the following function, sometimes I get the error "Failed to send Shadow update error MQTT LIBRARY ERROR".

Some times I get the error, Wait complete with result NETWORK ERROR, Wait complete with result TIMEOUT.

When will these errors occur? What will be the root cause of these errors?

I have a stable Wifi connection and when these errors occur, I check the ping of the IP address and everything is good.

_sendShadowUpdates(IotMqttConnection_t mqttConnection,
const char * const pThingName,
size_t thingNameLength );

static int _setShadowCallbacks(IotMqttConnection_t mqttConnection,
const char * pThingName,
size_t thingNameLength );

updateStatus = AwsIotShadow_TimedUpdate( mqttConnection,
&updateDocument,
AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS,
5000 );

Is clientToken must while publishing the shadow from the device?

Will I be able to publish the shadow when an OTA job is in progress?

Please let me know as soon as possible?

Thanks, Regards,
Kumar.

aggarw13, 2021-06-23:
Hi @kumarfirmware, sorry about the delay in responding. 
For us to be able to diagnose the problem, will you be able to share the device logs? 
Can you share information about your application architecture such as the number of tasks you're running and the relative priority of the task using the Shadow library? 

>Is clientToken must while publishing the shadow from the device?

The Client Token information is optional by the AWS IoT Device Shadow service requirements for communicating with the server, but the Shadow library in the 202007.00 release requires that the Shadow document passed to the `AwsIotShadow_TimedUpdate` API contains the entry (as present in the logic [here](https://github.com/aws/amazon-freertos/blob/202007.00/libraries/c_sdk/aws/shadow/src/aws_iot_shadow_api.c#L897-L910)).  If you do not want to send the client token in the request payload to the service, you can modify the library though it can involve non-trivial modifications.

yanjos-dev, 2021-06-30:
I'm closing this issue due to inactivity. Please reopen this issue if you would like any further assistance.

closed 2021-06-30 by yanjos-dev

kumarfirmware, 2021-07-11:
Hi Team,

I am running two tasks in my application and one more task is OTA task which is internal to AWS SDK. So in total three tasks.
The task which reports shadows has highest priority compared to the other application task. I am not sure about the priority of the OTA task as it is internal to AWS SDK.

I understand we can use same MQTT connection for Shadow publishing and normal messages publishing and OTA update.

But the question here is for example when OTA update is happening, at the same time I want to publish shadow, at the same time I want to publish application message. Is this possible? If possible how does this work? Please let me know as soon as possible.

Thanks, Regards,
Kumar.