aws/amazon-freertos #3228: Some times Shadow update is not received on device, using MQTT over BLE

kumarfirmware, 2021-06-14:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

Setup:- Device is connected to AWS using mobile app using MQTT over BLE.

Shadow is subscribed with QOS1 on device. When shadow is published from the cloud, some times device misses the shadow.

Is the mobile app missing the shadow from AWS? or Is the mobile app receiving the shadow from AWS but failing to push it to the device?

How can we log this on the mobile App?

This does not happen every time, but happens some times (considerable number of times).

This is creating huge problems for us.

Kindly let me know as soon as possible, how to proceed further?

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-06-14:
Hi @kumarfirmware 

Is mobile app an IOS/Android Application ? Which version/commit ID of the SDK are you using ?
Mobile app  by default logs the messages received from AWS before sending to device, and also messages from device before sending to AWS, into the terminal. Would you be able to run the mobile app through xcode/android studio and reproduce this issue ? If so, could you paste the logs from the xcode/android studio terminal ? 

Also could you provide more details on the setup ? Which region is AWS cognito pool configured with ? Is the mobile app running in same region ? Is the AWS IoT broker endpoint in the same region ?

dan4thewin, 2021-06-30:
This issue has been idle for a long while, so I'll close it.  If there's still a problem, please open a new issue.

closed 2021-06-30 by dan4thewin