aws/amazon-freertos #3230: Subscription error in OTA library, triggers loop of MQTT disconnect and connect and OTA job fails.

kumarfirmware, 2021-06-14:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

ESP32 is connected to the corporate network Wi-Fi. Very High speed network. No chance of unstable network.
Also ESP does not loose ping at all. Ping is very stable.

Some times OTA library shows subscription error in OTA.  After the keep alive time out, MQTT disconnect happens and reconnects again. After the MQTT reconnect again,  OTA library shows subscription error. After the keep alive time out, MQTT disconnect happens and reconnects again. 

The above pattern continues as a loop and OTA job does not go through.

Remember this does not happen all the time, this happens only some times.

This is stopping the production again.

Kindly help in this as soon as possible.

Thanks, Regards,
Kumar.

pvyawaha, 2021-06-15:
Hello @kumarfirmware ,

Can you please share the logs when this issue happens as this is not reproducible in our tests on AWS FreeRTOS version 202007 and ESP32?

dan4thewin, 2021-06-30:
This issue has been idle for a long while, so I'll close it. If there's still a problem, please open a new issue.

closed 2021-06-30 by dan4thewin