aws/amazon-freertos #3235: [General] <ERROR: tool xtensa-esp32-elf has no installed versions. Please run 'C:\FreeRTOS1\vendors\espressif\esp-idf\install.bat' to install it.>

abhibhatia98, 2021-06-15:
**Briefly summarize the issue being raised**
Hi, 
First of all this is not an issue, may be I am doing something wrong or missing out something. 
So I am trying to build aws FreeRTOS using [this](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf42) guide. But when I run  `vendors\espressif\esp-idf\export.bat` command it gives me error specified in subject.

```
C:\FreeRTOS1>vendors\espressif\esp-idf\export.bat
Setting IDF_PATH: C:\FreeRTOS1\vendors\espressif\esp-idf

Adding ESP-IDF tools to PATH...
ERROR: tool xtensa-esp32-elf has no installed versions. Please run 'C:\FreeRTOS1\vendors\espressif\esp-idf\install.bat' to install it.
ERROR: tool esp32ulp-elf has no installed versions. Please run 'C:\FreeRTOS1\vendors\espressif\esp-idf\install.bat' to install it.
ERROR: tool cmake has no installed versions. Please run 'C:\FreeRTOS1\vendors\espressif\esp-idf\install.bat' to install it.
ERROR: tool openocd-esp32 has no installed versions. Please run 'C:\FreeRTOS1\vendors\espressif\esp-idf\install.bat' to install it.
ERROR: tool mconf has no installed versions. Please run 'C:\FreeRTOS1\vendors\espressif\esp-idf\install.bat' to install it.
ERROR: tool ninja has no installed versions. Please run 'C:\FreeRTOS1\vendors\espressif\esp-idf\install.bat' to install it.
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "C:\FreeRTOS1\vendors\espressif\esp-idf\tools\idf_tools.py", line 1358, in <module>
    main(sys.argv[1:])
  File "C:\FreeRTOS1\vendors\espressif\esp-idf\tools\idf_tools.py", line 1354, in main
    action_func(args)
  File "C:\FreeRTOS1\vendors\espressif\esp-idf\tools\idf_tools.py", line 980, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "C:\FreeRTOS1\vendors\espressif\esp-idf\tools\idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\abhishekbhatia\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\Users\abhishekbhatia\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 487, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['git', '--work-tree=C:\\FreeRTOS1\\vendors\\espressif\\esp-idf', 'describe', '--tags']' returned non-zero exit status 128.
```

I have tried to build esp-idf hello world program that build correctly and works fine.

**System information**
- Hardware board: [ESP32-DevKitC]
- IDE used: [CMD]
- Operating System: [ Windows ]
- Code version: Downloaded code from AWS FreeRTOS console
- Project/Demo: [ mqtt_demo_mutual_auth]

**Code/Steps to reproduce the current status**
Set up esp-idf v4.2
set up mingw from [here ](https://sourceforge.net/projects/mingw/files/Installer/)
Going to freertos project directory 
and ran the command `source vendors/espressif/esp-idf/export.sh`

Thanks

aditi230, 2021-06-15:
Hello, I am facing the same issue. Even though mine is not solved yet.
I assume this is coming because of esp-idf installed at the time of tool setting. That is at this stage: [](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/windows-setup.html).Because if you open esp-idf cmd it wont open at freeRtos directory.

abhibhatia98, 2021-06-16:
@yourslab if you can help, I will be thankful.

Thanks 
Abhishek

abhibhatia98, 2021-06-18:
HI @aditi230, have you resolved this issue ?

if not, you can check https://github.com/aws/amazon-freertos/issues/3196 thread, this can be useful for you.

Thanks

dan4thewin, 2021-06-24:
Hi, @abhibhatia98 .  Did you use these instructions to install esp-idf, https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/windows-setup.html ?  Did they succeed?  Did you try the shortcut to export.bat in the Start menu that it should have installed?

yanjos-dev, 2021-07-02:
I'm closing this issue due to inactivity. Please reopen this issue if you would like any further assistance.

closed 2021-07-02 by yanjos-dev