aws/amazon-freertos #3238: [BUG] A assert fail problem occurred in iot_mqtt_api

strawberry3papa, 2021-06-16:
1. esp32 connected to AWS IoT platform.
2. MQTT PINGRESP is generated.
3. Network connection closed by PINGRESP
4. A assert fail problem occurred in iot_mqtt_api.

==============================================================================
[Wed Apr 28 08:18:07.161 2021] 
[Wed Apr 28 08:18:07.161 2021] 13721 4761169 [_scheduleIntern] [INFO ][HDC][47611690] +-hdc_timer_mqtt_telemetry_start(time=60000)[1]
[Wed Apr 28 08:18:07.161 2021] 
[Wed Apr 28 08:18:07.161 2021] 13722 4761169 [_scheduleIntern] [WARN ][MQTT][47611690] (MQTT connection 0x3f826994) Reference count changed from 3 to 2.
[Wed Apr 28 08:18:07.163 2021] 
[Wed Apr 28 08:19:05.407 2021] 13723 4766996 [_scheduleIntern] [ERROR][MQTT][47669960] (MQTT connection 0x3f826994) Failed to receive PINGRESP within 10000 ms.
[Wed Apr 28 08:19:05.407 2021] 
[Wed Apr 28 08:19:06.168 2021] 13724 4767068 [_scheduleIntern] [INFO ][MQTT][47670680] (MQTT connection 0x3f826994) Network connection closed.
[Wed Apr 28 08:19:06.168 2021] 
[Wed Apr 28 08:19:06.168 2021] 13725 4767068 [_scheduleIntern] [INFO ][HDC][47670680] Shadow Ne../libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:823 (_IotMqtt_DecrementConnectionReferences)- assert failed!
[Wed Apr 28 08:19:06.168 2021] 
[Wed Apr 28 08:19:06.168 2021] abort() was called at PC 0x400fa263 on core 0
[Wed Apr 28 08:19:06.168 2021] 
[Wed Apr 28 08:19:06.168 2021] Backtrace:0x40090b2f:0x3fff4570 0x4009116d:0x3fff4590 0x40095a1a:0x3fff45b0 0x400fa263:0x3fff4620 0x400fba65:0x3fff4650 0x40196a89:0x3fff4690 0x40197837:0x3fff46b0 0x400911dd:0x3fff46d0
[Wed Apr 28 08:19:06.178 2021] 
[Wed Apr 28 08:19:06.178 2021] 
[Wed Apr 28 08:19:06.178 2021] ELF file SHA256: 4ad6762f5ceeaded
[Wed Apr 28 08:19:06.178 2021] 
[Wed Apr 28 08:19:06.178 2021] Rebooting...
[Wed Apr 28 08:19:06.202 2021] ets Jul 29 2019 12:21:46
[Wed Apr 28 08:19:06.202 2021] 
[Wed Apr 28 08:19:06.202 2021] rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[Wed Apr 28 08:19:06.202 2021] configsip: 0, SPIWP:0xee
[Wed Apr 28 08:19:06.202 2021] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[Wed Apr 28 08:19:06.202 2021] mode:DIO, clock div:2
[Wed Apr 28 08:19:06.202 2021] load:0x3fff0030,len:4
[Wed Apr 28 08:19:06.202 2021] load:0x3fff0034,len:7700
[Wed Apr 28 08:19:06.202 2021] load:0x40078000,len:13724
[Wed Apr 28 08:19:06.202 2021] load:0x40080400,len:4628
[Wed Apr 28 08:19:06.202 2021] entry 0x40080704
[Wed Apr 28 08:19:06.248 2021] I (29) boot: ESP-IDF v4.2-15-g6abb2ca33-dirty 2nd stage bootloader
[Wed Apr 28 08:19:06.248 2021] I (29) boot: compile time 16:58:07
==============================================================================

gshvang, 2021-06-22:
Hi @strawberry3papa , We will investigate further and get back to you as soon as we are able to solve the problem. Also, I highly recommend that you have your application use coreMQTT instead if possible.

muneebahmed10, 2021-07-21:
Hi @strawberry3papa,

This deprecated version of the MQTT library is no longer in the `main` branch of this repository. I would recommend migrating to the latest release 202107.00, which uses the [coreMQTT](https://github.com/FreeRTOS/coreMQTT) and [coreMQTT-Agent](https://github.com/FreeRTOS/coreMQTT-Agent) libraries. You can find various examples for using the coreMQTT-Agent library [here](https://github.com/aws/amazon-freertos/tree/main/demos/coreMQTT_Agent) and [here](https://github.com/FreeRTOS/coreMQTT-Agent-Demos).

closed 2021-07-25 by yourslab