aws/amazon-freertos #3239: [BUG] <replace with your title>

strawberry3papa, 2021-06-16:
1. MQTT connect to AWS Platform
2. While the device is connected to the aws platform, only the internet is disconnected.
3. MQTT PINGRESP is triggered.
4. Network connection closed by PINGRESP
5. MQTT(IOT_MQTT_FLAG_CLEANUP_ONLY) disconect to AWS platform
6. Continue from 1 to 5 until the problem occurs.

The NO MEMORY problem occurs as shown below.
===========================================
2782 177850 [message_task] [INFO ][HDC][1778500] ++MqttConnect DeviceId[a1842010521cb6992a] MqttConnection[0x0]
2783 177850 [message_task] [INFO ][HDC][1778500] MQTT demo client identifier is a1842010521cb6992a (length 18).
2784 178053 [message_task] [INFO ][MQTT][1780530] Establishing new MQTT connection.
2785 178053 [message_task] [ERROR][MQTT][1780530] (MQTT connection 0x3f8378c4) **Failed to allocate memory for the MQTT context and the MQTT Connection Mapping. Update the MAX_NO_OF_MQTT2786 178053 [message_task] [ERROR][MQTT][1780530] Failed to establish new MQTT connection, error NO MEMORY.**
[00000011] DRAM(I[63124]+E[3949724])[4074220] IRAM[61372] free[3975420] FreeHeapSize [4073916k]
heap[4014356] task stack aws_app[0] light[2576] mot[1084] motu[680] cons[0] msg[4340]
2787 178154 [message_task] [INFO ][MQTT][1781540] Network connection closed on error.
2788 178154 [message_task] [WARN ][MQTT][1781540] (MQTT connection 0x3f8378c4) does not have an MQTT Context.
2789 178155 [message_task] [INFO ][MQTT][1781550] (MQTT connection 0x3f8378c4) Network connection destroyed.
2790 178155 [message_task] [ERROR][HDC][1781550] **MQTT CONNECT returned error NO MEMORY.**

gshvang, 2021-06-22:
Hi @strawberry3papa , We will investigate further and get back to you as soon as we are able to solve the problem. Also, I highly recommend that you have your application use coreMQTT instead if possible.

yanjos-dev, 2021-07-06:
I'm closing this issue due to inactivity. Please reopen this issue if you would like any further assistance.

closed 2021-07-06 by yanjos-dev