aws/amazon-freertos #3240: E (315385) ota_pal: signature verification failed (I cann't OTA renew.)

adrian-zulnedi, 2021-06-16:
Hi, Last time, I took this opportunity to teach you about FreeRTOS environment settings. 
This time I wan' to know how can I renew my OTA file.  When I was make FreeRTOS Job is doesn`t work.
I have some thing error like this. 

```
I (314545) ota_pal: No such certificate file: aws_ota_codesigner_certificate.h. Using aws_ota_codesigner_certificate.h.
E (315385) ota_pal: signature verification failed
```

I take shourcut how to make that FreeRTOS renew code.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

@yourslab hi, last time twice thanks to help me. Sorry to trouble you if you can help me again. I need your help.

yourslab, 2021-06-16:
Hi, I'm going to refer you to my colleague who is an expert of the OTA agent.

adrian-zulnedi, 2021-06-16:
@yourslab thank you so much.
@pvyawaha hi, nice to meet you. My name is Adrian. I am beginner of FreeRTOS. 
I have project in my company to do Over the Air using AWS FreeRTOS service.  

this is which I found references then I trying like him(youtube)
https://www.youtube.com/watch?v=WkTkbIOvLkw

So in this lecture↑ How to setup Enviroment FreeRTOS between How to Update data on AWS FreeRTOS Job.
I was done to try in Setup Enviroment on Linux Ubuntu with help Mr.yourslab. 
Then I just to trying renew data with see monitor on terminal. but when I was renew that data it doesn`t work because signature verification failed. 

That youtube references doesn`t tell details is like Omitted. Can I ask your opinion and teach me? Mr. @pvyawaha .

----------------------------------------------------------------------------------------------------------------

1 Another error.(Because I want to make not just only one firemware. I need Multiple to make Over the Air with ESP32)
So ,  I just want to trying setup again in FreeRTOS to make firmware but have error again like this issue.↓
https://github.com/aws/amazon-freertos/issues/3196
↓
https://github.com/aws/amazon-freertos/issues/3209

But in issue 3209 is when I install again my operything system is work.

------------------------------------------------------------------------------------------------------------------

pvyawaha, 2021-06-16:
Hello @adrian-zulnedi,

Can you please check if you have updated the signingcredentialSIGNING_CERTIFICATE_PEM in https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/platform/posix/ota_pal/source/ota_pal_posix.c

adrian-zulnedi, 2021-06-16:
@pvyawaha hi, I think this repository is have nothing to do with my projects. because I don`t use that [ iot-device-sdk-embedded-C ] But If you say just about SIGNING_CERTIFICATE_PEM I have already to update the signingcredential.

[image]

adrian-zulnedi, 2021-06-17:
@pvyawaha @yourslab hi guys, I'm very sorry because I didn`t read more details about FreeRTOS GUIDE.
I know right now how can I must to do it. Why when I was renew that JOB because in SIGNING_CERTIFICATE_PEM is not Certification Authority ESP32. So when I was update again is work available. I am very sorry with your guys. Thanks to help me.

closed 2021-06-17 by yourslab

abhibhatia98, 2021-06-19:
Hi @adrian-zulnedi, 

I am also stuck in some what same scenario, but the thing is I am unable to find this `aws_ota_codesigner_certificate.h` file. 

I have checked the location as mentioned in [this](https://docs.aws.amazon.com/freertos/latest/userguide/download-ota-esp.html) documentation. I have downloaded the FreeRTOS project from [here](https://github.com/aws/amazon-freertos).

You can check this repo doesn't has that file. 

Could you please help me once again like from where you have downloaded the FreeRTOS project or incase I am missing out some thing.

Thanks 
Abhi

adrian-zulnedi, 2021-06-19:
@abhibhatia98  Hi, I have same mistake like you about `aws_ota_codesigner_certificate.h`. I guess you mistake input that crypto key. Did you make Certificate Authority(CA)? If not you must make that. 

`ecdsasigner.crt` and `ecdsasigner.key` then you must put in `ecdsasigner.crt` on `aws_ota_codeginer_certificate.h`.

please check this reference.
https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert-win.html

abhibhatia98, 2021-06-19:
Hi @adrian-zulnedi Thanks for your reply.

I have generate the ecadasigner certificate with the help of guide you mention. But can't find the `aws_ota_codeginer_certificate.h` file in /demos/include, for copying certificate.

Do you made this file or it is already there ??

You can check in [this ](https://github.com/aws/amazon-freertos) repo, `aws_ota_codeginer_certificate.h` file is not available. 

Thanks 
Abhi

adrian-zulnedi, 2021-06-19:
@abhibhatia98 sorry can you show me that shourcut?

adrian-zulnedi, 2021-06-19:
@abhibhatia98 I'm not make that file . that file already generate when I download that FreeRTOS folder in AWS. 
I think when you make FreeRTOS folder in AWS Something mistake to Download.

adrian-zulnedi, 2021-06-19:
@abhibhatia98 Can you show me that how you make FreeRTOS on AWS service with shourcut? maybe I can help.

adrian-zulnedi, 2021-06-19:
I use this Device Software is called `Connect to AWS IoT - ESP-WROVER-KIT ESP32-DevKitC 202012.00 Quick connect`.

[image]

abhibhatia98, 2021-06-19:
Hi @adrian-zulnedi, 
 
I have downloaded the freeRTOS project from [github ](https://github.com/aws/amazon-freertos). That doesn't have this file

And if I download from AWS FreeRTOS console, that cause some error in build, like below : 

[image]

Thanks 
Abhi

adrian-zulnedi, 2021-06-19:
@abhibhatia98  Did you have install xtensa-esp32-elf? or you must reinstall again and delete folder on build all

abhibhatia98, 2021-06-22:
Hi @adrian-zulnedi, I hope you have done with OTA. Well I am stuck at some points in which I required your help. 

Below is the screenshot of generating code signing profile, well I am confused what should I enter in `Path name of code signing certificate` ??
[image]

Second is after generating code signing profile, what should I enter in `Path name of file on device` ??

[image]

Please provide what you done in those two.

Thanks 
Abhishek

adrian-zulnedi, 2021-06-22:
@abhibhatia98  Hi, I enter that `Path name of file on device` are /demos/include/aws_ota/aws_ota_codesigner_certificate.h
or just /demos/include. OR  you can Absolute path when you install FreeRTOS .  `FreeRTOS/demos/include/aws_ota/aws_ota_codesigner_certificate.h`