aws/amazon-freertos #3241: break while building esp32-at esp-idf

adrian-zulnedi, 2021-06-16:
Hi, guys  I have make issue alikely  about why not make in `make all -j4` But is not have corectly what I want to know it.
Because I want to runnning on mac OS (M1) not just only Linux Ubuntu20.04. 

Then I was run `make all -j4` something have problem with xtensa-esp32-elf.
I try in another issue But not working . Step by step to using this lecture↓ in ESPRESSIF I use 3.3V.

@pvyawaha Hi, When I was talk with Mr.youslab , He help me not full about make all -j4. If you know something cloud you teach me about why is cannnot work ? I am very glad can to talk with you.

[Build break while building esp32-at using release/v1.2.0.0 branch #215](https://github.com/espressif/esp-at/issues/215)

```
/Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
```

adrian-zulnedi-plugo, 2021-06-17:
@yourslab Hi I am very glad for your passion to teach me. thank you. I have trying that your comment link.
But still doesn`t work. Maybe I must initialize my operating sytem? I had try command in terminal. cloud you check my terminal? 

[image]
[image]
[image]
[image]

adrian-zulnedi, 2021-06-17:
@yourslab Hi I am very glad for your passion to teach me. thank you. I have trying that your comment link.
But still doesn`t work. Maybe I must initialize my operating sytem? I had try command in terminal. cloud you check my terminal? 

[image]
[image]
[image]
[image]

yourslab, 2021-06-18:
I am introducing @shubhamkulkarni97, engineer from Espressif. Have any other customers encountered this issue when compiling using M1 Mac? The customer was able to compile using Ubuntu but gets an error when following the same steps on M1 Mac.

adrian-zulnedi, 2021-06-19:
@yourslab Hi, Thanks for your help.
@shubhamkulkarni97 hi, nice to meet you. I need your help for how can I install extensa-esp32-elf on Mac(M1&intel). Intel have same error.

adrian-zulnedi, 2021-06-21:
@yourslab Hi, Mr.shubnamkulkarni97 not have reply.

adrian-zulnedi, 2021-06-21:
this esp-idf have many something trouble because when I was `. ./exporth.sh` that have error like this. 

[image]

adrian-zulnedi, 2021-06-21:
ERROR DOCUMENTS

```

Traceback (most recent call last):
  File "/Users/adrian_plugo/esp/esp-idf/tools/idf.py", line 822, in <module>
    main()
  File "/Users/adrian_plugo/esp/esp-idf/tools/idf.py", line 740, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var='_IDF.PY_COMPLETE')
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1057, in main
    self._main_shell_completion(extra, prog_name, complete_var)
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1132, in _main_shell_completion
    rv = shell_complete(self, ctx_args, prog_name, complete_var, instruction)
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 45, in shell_complete
    echo(comp.source())
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 324, in source
    self._check_version()
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 312, in _check_version
    raise RuntimeError(
RuntimeError: Shell completion is not supported for Bash versions older than 4.4.

```

shubhamkulkarni97, 2021-06-21:
Hi @adrian-zulnedi, sorry for late response.

1. Regarding main issue, it seems you are not using correct toolchain version for building the application. You can see a warning message in the logs you shared [here](https://github.com/aws/amazon-freertos/issues/3241#issuecomment-863436772). It says: `version 8.4.0 is not supported`, you should use `version 5.2.0`.
To fix this issue, it is recommended to run `./install.sh` and `. ./export.sh` commands in esp-idf folder in the repo.

2. Regarding issue mentioned [here](https://github.com/aws/amazon-freertos/issues/3241#issuecomment-864690833), default bash version supported in macOS is completely outdated. To use all features supported by ESP-IDF, you need to update bash to latest version. You can check current bash version by running `bash --version`. For upgrading bash, you can refer this [link](https://itnext.io/upgrading-bash-on-macos-7138bd1066ba).

You should upgrade bash version and then run commands mentioned in first point.

Hope this helps!

Thanks,
Shubham

adrian-zulnedi, 2021-06-21:
@shubhamkulkarni97 hi, thank for your reply. 
1. when I was download esp-idf is version 5.2.0 but when I have installed is make to 8.4.0 .how can I fix that?
2. I just try that link but doesn`t work.

[image]

shubhamkulkarni97, 2021-06-21:
@adrian-zulnedi, Could you share which repositories are you using to build this project? Also the branch used.

Are you using both Amazon FreeRTOS and ESP-IDF?

adrian-zulnedi, 2021-06-21:
@shubhamkulkarni97 sorry cloud you tell me that how to check it ? command line. Because I think I use both . Amazon FreeRTOS and ESP-IDF .

shubhamkulkarni97, 2021-06-22:
@adrian-zulnedi, If you are using Amazon FreeRTOS, required version of ESP-IDF is included in it. You can check it in `./vendors/espressif/esp-idf` directory. In this case you should not use external ESP-IDF.

Please note that you need to run `./install.sh` and `. ./export.sh` commands in `./vendors/espressif/esp-idf` directory in Amazon FreeRTOS (not in any other directory/repo).

To check version of Amazon FreeRTOS you are using, you can run `git describe --tags` in AFR repo.

Also, the environment setup by above commands will be valid for single terminal. If you use new terminal window, you need to run these commands again.

adrian-zulnedi, 2021-06-25:
@shubhamkulkarni97 Sorry I know why can running that make all -j4. Because In Mac M1 is not creally good for runnnig.
I must to use Mac (intel) and version is ESP-IDF (v3.3). Just only this. thanks for your help. I will close this issue.

closed 2021-06-25 by adrian-zulnedi