aws/amazon-freertos #3251: Error: Downgrade or same version not allowed

Carfer10, 2021-06-19:
Hello everyone,

I've been working in this for some weeks, I've followed the official AWS OTA Update tutorial and AWS Getting Started with FreeRTOS. I succeeded and my ESP32 was working with the OTA Demo running on it, but the problem comes when executing an OTA job.

To update the firmware, I do the following steps:

1. Change code in aws_iot_ota_update_demo.c (for example a comment or the MQTT topic to publish to, definitely not relevant lines that could involve malfunction in the OTA Agent)
2. Increment variable APP_VERSION_BUILD in file aws_application_version.h
3. Generate build files again using 
```
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build/
```
4. Build files inside build/ directory
```
make all -j4
```
5. Upload newly generated aws_demos.bin inside build folder to my AWS S3 bucket
6. Execute OTA job from CLI using the following JSON file (changing the version of the S3 object with the new one):
```
aws iot create-ota-update --cli-input-json file://cli_ota_update_job.json
```
```
{
    "otaUpdateId": "otajob1",
    "description": "This is a test job",
    "targets": [
      "arn:aws:iot:eu-west-1:XXXXXXXXXXXX:thing/Node2"
    ],
    "protocols": [
      "MQTT"
    ],
    "targetSelection": "SNAPSHOT",
    "awsJobExecutionsRolloutConfig": {
        "maximumPerMinute": 1000
    },
    "awsJobPresignedUrlConfig": { },
    "files": [
      {
        "fileName": "/",
        "fileLocation": {
            "s3Location": {
                "bucket": "ota-updates-tfm",
                "key": "aws_demos.bin",
                "version": NEW_VERSION_STRING
            }
        },
        "codeSigning": {
            "startSigningJobParameter": {
                "signingProfileParameter": {},
                "signingProfileName": "profile_esp32_2",
                "destination": {
                    "s3Destination": {
                        "bucket": "ota-updates-tfm",
                        "prefix": "SignedImages/"
                    }
                }
            }
        }
      }
    ],
    "roleArn": "arn:aws:iam::XXXXXXXXXXXX:role/OTA_Service_Role"
}
```

7. The ESP32 gets the job correctly, but when updating I get the following error message before rebooting to previous version:
```
53 1220 [OTA Agent Task] [prvValidateUpdateVersion] We rebooted and the version is still the same.
54 1220 [OTA Agent Task] [prvParseJobDoc] Downgrade or same version not allowed, rejecting the update & rebooting.
55 1224 [OTA Agent Task] [prvPublishStatusMessage] Msg: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x23000000"}}
```

**System information**
- Hardware board: ESP32-DevKitC
- Operating System: Ubuntu 20.04
- Code version: FreeRTOS version 202012.00, downloaded directly from AWS FreeRTOS console
- Project/Demo: ota demo, but modified as this [https://github.com/aws/amazon-freertos/blob/mchp/otaapp/demos/ota/aws_iot_ota_update_demo.c](url) as I wanted to use OTA Agent demo and publish to MQTT topic using same connection

**Expected behavior**
I expected to have my ESP32 updated with the new firmware as I updated the version in the aws_application_version.h file.

**Additional context**
I don't know if it would be relevant, but if I flash directly to the board, the new version is recognized as expected and do not reboot.

pvyawaha, 2021-06-21:
Hello @Carfer10 ,

Can you please check in the logs after the image is received and the device reboots what the version number is ? It will be great if you can share the logs.

Carfer10, 2021-06-21:
Hi @pvyawaha,

I can't share it right now, but the last time I checked the version received and the version when the device reboots. Both are the same.

This is not the case when I flash the device directly, the version received is the new one and it doesn't reboot. 

Tomorrow I'll be able to share the logs

Carfer10, 2021-06-22:
> Hello @Carfer10 ,
> 
> Can you please check in the logs after the image is received and the device reboots what the version number is ? It will be great if you can share the logs.

This is the change log after the ota update is received and until the device reboots:
```
4826 95722 [OTA Agent Task] [prvIngestDataBlock] Received file block 273, size 4096
4827 95722 [OTA Agent Task] [prvIngestDataBlock] Remaining: 1
4828 95722 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock] 
4829 95724 [OTA Agent Task] [INFO ][MQTT][957240] (MQTT connection 0x3ffb9cd0) MQTT PUBLISH operation queued.
4830 95724 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/Node2/streams/AFR_OTA-e4480498-43e8-4d6d-8315-77db3a813e24/get/cbor
4831 95724 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock] 
4832 95742 [OTA Agent Task] [prvIngestDataBlock] Received file block 274, size 3824
4833 95742 [OTA Agent Task] [prvIngestDataBlock] Received final expected block of file.
4834 95742 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
I (957497) ota_pal: No such certificate file: /. Using aws_ota_codesigner_certificate.h.

4835 95776 [iot_thread] State: WaitingForFileBlock  Received: 277   Queued: 0   Processed: 0   Dropped: 0
4836 95779 [iot_thread] [INFO ][MQTT][957790] (MQTT connection 0x3ffb9cd0) MQTT PUBLISH operation queued.
4837 95779 [iot_thread] [INFO ][MQTT][957790] (MQTT connection 0x3ffb9cd0, PUBLISH operation 0x3ffba01c) Waiting for operation completion.
4838 95916 [OTA Agent Task] [prvIngestDataBlock] File receive complete and signature is valid.
4839 95916 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
4840 95916 [OTA Agent Task] [prvPublishStatusMessage] Msg: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x90012"}}
4841 95917 [OTA Agent Task] [INFO ][MQTT][959170] (MQTT connection 0x3ffb9cd0) MQTT PUBLISH operation queued.
4842 95917 [OTA Agent Task] [INFO ][MQTT][959170] (MQTT connection 0x3ffb9cd0, PUBLISH operation 0x3ffba190) Waiting for operation completion.
4843 95921 [iot_thread] [INFO ][MQTT][959210] (MQTT connection 0x3ffb9cd0, PUBLISH operation 0x3ffba01c) Wait complete with result SUCCESS.
4844 95936 [OTA Agent Task] [INFO ][MQTT][959360] (MQTT connection 0x3ffb9cd0, PUBLISH operation 0x3ffba190) Wait complete with result SUCCESS.
4845 95936 [OTA Agent Task] [prvPublishStatusMessage] 'IN_PROGRESS' to $aws/things/Node2/jobs/AFR_OTA-pruebacliexterna33/update
4846 95936 [OTA Agent Task] Received eOTA_JobEvent_Activate callback from OTA Agent.
4847 95936 [OTA Agent Task] [INFO ][MQTT][959360] (MQTT connection 0x3ffb9cd0) Disconnecting connection.
4848 95936 [OTA Agent Task] [INFO] [MQTT] [core_mqtt.c:2149] 4849 95936 [OTA Agent Task] Disconnected from the broker.4850 95936 [OTA Agent Task] 
4851 95936 [OTA Agent Task] [INFO ][MQTT][959360] (MQTT connection 0x3ffb9cd0) Network connection closed.
I (959417) boot_comm: chip revision: 1, min. application chip revision: 0
I (959477) esp_image: segment 0: paddr=0x001a0020 vaddr=0x3f400020 size=0x32ab8 (207544) map
I (959647) esp_image: segment 1: paddr=0x001d2ae0 vaddr=0x3ffbdb60 size=0x039d0 ( 14800) 
I (959667) esp_image: segment 2: paddr=0x001d64b8 vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _WindowOverflow4 at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

I (959667) esp_image: segment 3: paddr=0x001d68c0 vaddr=0x40080400 size=0x09750 ( 38736) 
I (959707) esp_image: segment 4: paddr=0x001e0018 vaddr=0x400d0018 size=0xc0780 (788352) map
0x400d0018: _flash_cache_start at ??:?

/xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/freertos_kernel/queue.c:1477 (xQueueSemaphoreTake)- assert failed!
abort() was called at PC 0x4008f609 on4852 96021 [iot_thread] State: WaitingForFileBlock  Receive core 0
0x4008f609: xQueueSemaphoreTake at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/freertos_kernel/queue.c:2429

ELF file SHA256: e6199796c6985075

Backtrace: 0x4008d1d0:0x3ffb3c70 0x4008d401:0x3ffb3c90 0x4008f609:0x3ffb3cb0 0x401514d3:0x3ffb3cf0 0x401514ed:0x3ffb3d10 0x40146c50:0x3ffb3d30 0x40148660:0x3ffb3d60 0x4014704e:0x3ffb3d80 0x40147a0c:0x3ffb3db0 0x400d3091:0x3ffb3de0 0x400d31b2:0x3ffb3e20 0x400d3291:0x3ffb3e80 0x400d2e0e:0x3ffb3ea0 0x401513ab:0x3ffb3ed0
0x4008d1d0: invoke_abort at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008d401: abort at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008f609: xQueueSemaphoreTake at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/freertos_kernel/queue.c:2429

0x401514d3: prIotMutexTimedLock at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x401514ed: IotMutex_Lock at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x40146c50: _IotMqtt_IncrementConnectionReferences at /xxxxxxxxxxxxxxxxxxxxxxxxx/FreeRTOS-Repo-Node2/FreeRTOS/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:758

[... cut ...]
55 1146 [iot_thread] State: Ready  Received: 1   Queued: 0   Processed: 0   Dropped: 0
56 1147 [iot_thread] [INFO ][MQTT][11470] (MQTT connection 0x3ffb9cd0) MQTT PUBLISH operation queued.
57 1147 [iot_thread] [INFO ][MQTT][11470] (MQTT connection 0x3ffb9cd0, PUBLISH operation 0x3ffb9a78) Waiting for operation completion.
58 1161 [iot_thread] [INFO ][MQTT][11610] (MQTT connection 0x3ffb9cd0, PUBLISH operation 0x3ffb9a78) Wait complete with result SUCCESS.
59 1261 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
```

Carfer10, 2021-06-23:
What I've found until now, it's that this error happen after trying to send an ota update to more than one node with the same job and the same ".bin" file. 

This update to more than one node fails (with same error as mentioned) and, afterwards, those nodes stopped updating correctly as mentioned above. It seems like they get bugged after this ota job.

pvyawaha, 2021-06-23:
Hello @Carfer10 ,

When you are sending the same .bin file to multiple `Things` are you storing the credentials somewhere in non volatile memory and reading from it ? If it is still in the header file then the same .bin will have same credentials.

Carfer10, 2021-06-25:
Thanks for your response @pvyawaha 

I'm not storing the credentials in non volatile memory, how should I proceed to update an N number of nodes with same firmware image?

ravibhagavandas, 2021-12-07:
Hi @Carfer10,

Did the above suggestion help resolve the issue ? 

Additionally I see that demo uses an old MQTT library version which is no longer actively maintained. Is it possible for you to upgrade to the new MQTT and OTA library so that you can stay up to date with the latest features and fixes. Please take a look at the OTA demo using latest MQTT and OTA library (LTS) versions  [here](https://github.com/aws/amazon-freertos/blob/main/demos/ota/ota_demo_core_mqtt/ota_demo_core_mqtt.c)

pvyawaha, 2021-12-16:
Hello @Carfer10 ,

Closing this ticket now , please reopen if you have questions on the same topic or create a new one for other topics.

closed 2021-12-16 by pvyawaha