aws/amazon-freertos #3252: [General] <Can not find demos/include/aws_ota_codesigner_certificate.h.>

abhibhatia98, 2021-06-19:
Well  I have successfully compile freeRTOS mqtt demo project, and now I am working on OTA demo. 

And followed [this](https://docs.aws.amazon.com/freertos/latest/userguide/burn-initial-firmware-esp.html) guide. In second point it mention to copy the certificate in to `demos/include/aws_ota_codesigner_certificate.h` , but when I goes to freeRTOS project that I downloaded from [github](https://github.com/aws/amazon-freertos), this file is not there.

Can anyone help me to find if I am missing something ?

Thanks
Abhi

Carfer10, 2021-06-19:
You first have to run the script tools/aws_config_quick_start/SetupAWS.py, as explained in this [getting started with FreeRTOS guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html). This script will generate that file.

abhibhatia98, 2021-06-20:
Hi, Thanks for your response @Carfer10 please check below snapshot.

[image]

It not generate` demos/include/aws_ota_codesigner_certificate.h` .

Regards
Abhi

Yoimer, 2021-06-20:
@abhibhatia98 Check this out, all about OTA is well explained there https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ota-dev.html

abhibhatia98, 2021-06-21:
Hi @Yoimer , Thanks for your response.

I am followed the same guide and when reaches to [this](https://docs.aws.amazon.com/freertos/latest/userguide/burn-initial-firmware-esp.html) page at second point 

[image]

the git project doesn't have this file. That's the issue.

Thanks 
Abhi

pvyawaha, 2021-06-21:
Hello @abhibhatia98 ,

Are you using the main ? We are working on integrating the latest OTA library in main branch of this repository along with the documentation updates , so the aws_ota_codesigner_certificate.h is removed. You can use previous release  https://github.com/aws/amazon-freertos/releases/tag/202012.00

closed 2021-06-23 by pvyawaha