aws/amazon-freertos #3256: [BUG] OTA_UPDATE_DEMO on ESP32 DevKitC V4 version is still the same after reboot

nova2882, 2021-06-21:
**Describe the bug**
Hi all, I am trying to test OTA with ESP32 DevKitC V4 by following this tutorial https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ota-dev.html. The board is run on OTA demo version 0.9.8 and I did increase the firmware version to 2.9.8 in file aws_application_version.h the new firmware version 2.9.8 is upload to S3. The new firmware seems successfully downloaded via MQTT protocol to the device. However, after reboot the firmware version keep the same 0.9.8 lead to error 
"44 945 [OTA Agent Task] [prvValidateUpdateVersion] We rebooted and the version is still the same.
45 945 [OTA Agent Task] [prvParseJobDoc] Downgrade or same version not allowed, rejecting the update & rebooting.
46 957 [OTA Agent Task] [prvPublishStatusMessage] Msg: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x23000000"}}"

**System information**
- Hardware board: ESP32 DevKitC V4
- IDE used: idf.py
- Operating System: MacOS 
- Code version: 202012.00-302-g06abb47fd
- Project/Demo: OTA_UPDATE_DEMO

**Expected behavior**
Device reboot with new firmware version (2.9.8) and the job execution status successful on AWS IoT job console

**Screenshots or console output**

570 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ timestamp: 1624297385 ]

571 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ jobId: AFR_OTA-aws-iot-ota-demo5 ]

572 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ protocols: ["MQTT"] ]

573 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ streamname: AFR_OTA-bbaffdef-d067-4a13-b4cb-2cd19dda8fc4 ]

574 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ filepath: / ]

575 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ filesize: 1179744 ]

576 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ fileid: 0 ]

577 53216 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ certfile: aws_ota_codesigner_certificate.h ]

578 53218 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ sig-sha256-ecdsa: MEUCICHZ96x/Nbe2jjN8Keaw9w3h0pgM... ]

579 53218 [OTA Agent Task] [prvParseJobDoc] Job was accepted. Attempting to start transfer.

[0;32mI (532937) ota_pal: prvPAL_GetPlatformImageState[0m
[0;32mI (533007) esp_ota_ops: aws_esp_ota_get_boot_flags: 1[0m
[0;32mI (533017) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0[0m
[0;32mI (533027) ota_pal: Writing to partition subtype 17 at offset 0x1a0000[0m
580 53248 [iot_thread] [INFO ][DEMO][532480] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0

581 53348 [iot_thread] [INFO ][DEMO][533480] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0

582 53451 [iot_thread] [INFO ][DEMO][534510] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0

583 53564 [iot_thread] [INFO ][DEMO][535640] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0

584 53671 [iot_thread] [INFO ][DEMO][536710] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0

[0;32mI (537757) ota_pal: esp_ota_begin succeeded[0m
[0;32mI (537757) ota_pal: prvPAL_GetPlatformImageState[0m
[0;32mI (537757) esp_ota_ops: aws_esp_ota_get_boot_flags: 1[0m
[0;32mI (537757) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0[0m
585 53701 [OTA Agent Task] [prvSetDataInterface] Data interface is set to MQTT.

586 53701 [OTA Agent Task] [prvProcessJobHandler] Setting OTA data inerface.

587 53701 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForJob] Event [ReceivedJobDocument] New state [CreatingFile] 

588 53702 [OTA Agent Task] [INFO ][MQTT][537020] (MQTT connection 0x3ffb63f0) SUBSCRIBE operation scheduled.

589 53702 [OTA Agent Task] [INFO ][MQTT][537020] (MQTT connection 0x3ffb63f0, SUBSCRIBE operation 0x3ffd4adc) Waiting for operation completion.

590 53707 [OTA Agent Task] [INFO ][MQTT][537070] (MQTT connection 0x3ffb63f0, SUBSCRIBE operation 0x3ffd4adc) Wait complete with result SUCCESS.

591 53707 [OTA Agent Task] [prvInitFileTransfer_Mqtt] OK: $aws/things/ESP32-DevKitC-01/streams/AFR_OTA-bbaffdef-d067-4a13-b4cb-2cd19dda8fc4/data/cbor

[... 3052 lines cut ...]
2118 57637 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ESP32-DevKitC-01/streams/AFR_OTA-bbaffdef-d067-4a13-b4cb-2cd19dda8fc4/get/cbor

2119 57637 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock] 

2120 57643 [OTA Agent Task] [prvIngestDataBlock] Received file block 246, size 4096

2121 57644 [OTA Agent Task] [prvIngestDataBlock] Remaining: 42

2122 57644 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock] 

2123 57646 [OTA Agent Task] [INFO ][MQTT][576460] (MQTT connection 0x3ffb63f0) MQTT PUBLISH operation queued.

2124 57646 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ESP32-DevKitC-01/streams/AFR_OTA-bbaffdef-d067-4a13-b4cb-2cd19dda8fc4/get/cbor

2125 57646 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock] 

2126 57654 [OTA Agent Task] [prvIngestDataBlock] Received file block 247, size 4096

2127 57655 [OTA Agent Task] [prvIngestDataBlock] Remaining: 41

2128 57655 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock] 

2129 57656 [OTA Agent Task] [INFO ][MQTT][576560] (MQTT connection 0x3ffb63f0) MQTT PUBLISH operation queued.

2130 57656 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ESP32-DevKitC-01/strea

pvyawaha, 2021-06-21:
Hello @nova2882 ,

Thank you for reporting this , we are trying to reproduce it on the release 202012.00.

nova2882, 2021-06-22:
Hi @pvyawaha,  thanks for the response.

I try to clone release FreeRTOS 202012.00 which has ESP-IDF v3.3 and the OTA example work just fine for me. 

I think the issue I have could due to I check out the master branch a couple of months ago and the OTA example still has bugs.
[image]

I saw the OTA example is moved out of Master branch so where I should start the development/testing? Last release or master branch?

Thanks

pvyawaha, 2021-06-22:
Hello @nova2882 ,

We are working on integration of OTA LTS Ver 3 in main branch. You can use latest official release with OTA V1 https://github.com/aws/amazon-freertos/releases/tag/202012.00.

closed 2021-06-23 by henrytrh