aws/amazon-freertos #3259: LINUX UBUNTU20.04 error: expected initializer before '__result_use_check' ,'__alloc_align' when I was use make all -j4

adrian-zulnedi, 2021-06-22:
Hi, I have same problem when I was make issue on this repository but I don't clear understand when I was `make all -j4` still have this error.

```

s_flash/src/nvs_api.cpp:14:
/home/adrian-zulnedi/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/adrian-zulnedi/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~

```

I try to reinstall all and I change ESPRESSIF version is v4.2 but It doesn't work.
What I mistake? Because I check step by step .

this same problem in mac but still not correct.
https://github.com/aws/amazon-freertos/issues/3209

adrian-zulnedi, 2021-06-22:
[image]

yourslab, 2021-06-22:
Hi @adrian-zulnedi. Have you tried following this step? https://github.com/aws/amazon-freertos/issues/3241#issuecomment-865512993

adrian-zulnedi, 2021-06-25:
@yourslab @shubhamkulkarni97  hi, sorry to let talk about this issue. I want close this issue. Because I have correctly for setup in MacOS (Intel). Thanks a lot with your action.

HOW CAN I RUNNING THAT MAC OS WITH FREERTOS ESP32 (V3.3)
https://docs.aws.amazon.com/ja_jp/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf42

closed 2021-06-25 by adrian-zulnedi