aws/amazon-freertos #3260: [General] <Converting from .wav/mp3 to .bin file>

arantza12345, 2021-06-22:
I am struggling to convert a .wav file to a .bin file.
I have been unable to find a converter that does this conversion and I need it in order to be able to safe some files into a program. 

Does anyone know how to do it?

Thank you.

muneebahmed10, 2021-06-22:
Hi @arantza12345,

This forum might not be the best place for you to receive help on file conversion questions, as it is primarily focused towards embedded devices. Have you considered trying Stack Exchange? You might be more able to receive help for your question there.

With that said, what is the use case for your .bin files? If any binary file will do regardless of format, can you just rename the `.wav` extension to `.bin`? If you need a specific disk image format, e.g. from a CD or DVD, can you just create a virtual disk image and burn your files to the virtual CD?

yanjos-dev, 2021-06-30:
I'm closing this ticket due to inactivity. Please reopen this ticket if you would like any further assistance.

closed 2021-06-30 by yanjos-dev