aws/amazon-freertos #3261: idf.py : command not found

aditi230, 2021-06-22:
**Briefly summarize the issue being raised**
While performing "To build and flash the project" on Ubuntu facing the error as idf.py : command not found
**Describe the desired outcome**
idf.py -DVENDOR=espressif -DBOARD=esp32__devkitc -DCOMPILER=xtensa-esp32 build
This is the command I ruined on FreeRTOS directory on linux
**System information**
- Hardware board: [ esp-32 devkitc ]
- IDE used: [ ... ] performing on terminal
- Operating System: [Linux]
- Code version: (run ``git describe --tags`` to find it) while I ruined this error was - "Not a fir repository"

[image]

gshvang, 2021-06-22:
Hi @aditi230, Does the FreeRTOS directory contains the FreeRTOS code that you downloaded? It would be helpful is you can share the contents of FreeRTOS directory where you are running this command. Also to run git command you have to change directory to the root of the FreeRTOS code that you downloaded. There should be a folder named amazon-freertos from where you need to run the git commands.

aditi230, 2021-06-23:
hey! it worked after I clones the git repository .

closed 2021-06-23 by aditi230