aws/amazon-freertos #3265: Configure cmake in the build directory and build the firmware image with the following command.

aditi230, 2021-06-23:
**Briefly summarize the issue being raised**
Configure cmake in the build directory and build the firmware image with the following command.
**_idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 build_**
Where do I need to run this command
**System information**
- Hardware board: [ esp32 devkitc ]
- IDE used: [ ... ]none
- Operating System: [ Linux  ]

gshvang, 2021-06-23:
Hi @aditi230, It is a python script, you can navigate to the folder where this file is located in the terminal and run it from there.
Let us know if you face any issues.

aditi230, 2021-06-24:
Hi @gshvang ,at this location it gave the error relating CMakeLists.txt Not found.
Where as when I ruined the same in direct RTOS directory it runned fine.
Do I need to do any changes?

gshvang, 2021-06-24:
HI @aditi230 , It is fine to run it from the direct RTOS directory. No changes are required. I hope it solves your issue.

aditi230, 2021-06-29:
Yes,Thank you.

closed 2021-06-29 by aditi230