aws/amazon-freertos #3268: How to link prebuilt Library in aws/amazon-freertos V201920?

Raghav3107, 2021-06-24:
Hello,
I m using the `amazon-freertos v2021912` repository. There I want to link my prebuild library `(libalgobsec.a`) which is in the `/aws_demos/application_code/ `directory.
How should I link this?

Thanks for the help?

dan4thewin, 2021-06-24:
Hi, @Raghav3107 .    How you manage linking is going to depend on the toolchain you use, which is often dictated by the board you're using.  You may find our getting started guides helpful: https://docs.aws.amazon.com/freertos/latest/userguide/freertos-getting-started.html
I'd also encourage you to engage with our knowledgable and helpful community forums: https://forums.freertos.org/

Raghav3107, 2021-06-25:
Hello @dan4thewin  
I m using the `xtensa-esp32-elf-gcc8_4_0-esp-2020r3` toolchain and esp32 dev kit,
I link the library in one of my[ custom repo ](https://github.com/Raghav3107/amazon-freertos-workspace-v202012)where I m using amazon-freertos as an external library. There I know which CMake file is responsible for the build. 
I added the two links to link the library.
```
# Declaring the precompiled BSEC lib as imported global so that it doesn't need to be recompiled
add_library(bme680lib STATIC IMPORTED GLOBAL)
set_target_properties(bme680lib PROPERTIES IMPORTED_LOCATION "${PROJECT_SOURCE_DIR}/components/bme680/algo/normal_version/bin/esp/esp32/libalgobsec.a")

# For some reason, math library from newlib isn't implicitly imported so that needs to be manually
# done for now. You might be able to get rid of the hack at some point
add_library(mathlib STATIC IMPORTED GLOBAL)
set_target_properties(mathlib PROPERTIES IMPORTED_LOCATION "${PROJECT_SOURCE_DIR}/amazon-freertos/vendors/espressif/esp-idf/components/newlib/lib/libm.a")

target_link_libraries(afr_workshop 
    PRIVATE
        bme680lib 
        mathlib
)

```
But I m don't know how will link here.

dan4thewin, 2021-06-28:
Hi, @Raghav3107 .  I replied to your post on the forums.  Let's continue the discussion there ( https://forums.freertos.org/t/how-to-link-the-prebuilt-libraries-in-amazon-freertos-repository/12916 )

closed 2021-06-28 by dan4thewin