aws/amazon-freertos #3288: How can we connect MQTT over WIFI in AWS freertos in esp32

aditi230, 2021-06-29:
**Briefly summarize the issue being raised**/
I need to publish data from esp32 to aws console through mqtt over wifi.How could I perform that?

**System information**
- Hardware board: [ ESP32 DEVKITC ]
- IDE used: [ UBUNTU terminal ]
- Operating System: [ Linux

muneebahmed10, 2021-06-30:
Hi @aditi230,

You can find detailed instructions for getting started with ESP32 and FreeRTOS [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html). From there, you should be able to run the [coreMQTT Mutual Authentication](https://docs.aws.amazon.com/freertos/latest/userguide/mqtt-demo-ma.html) demo to connect to AWS IoT over MQTT.

aditi230, 2021-07-09:
I followed the same step but still facing the error. Do I need to any changes in cmake file?

ravibhagavandas, 2021-07-13:
Could you provide more details on the error or at what step of your setup its failing so that we can assist further in troubleshooting.

closed 2021-07-16 by aditi230