aws/amazon-freertos #3289: [General] <While running demo : core mqtt facing error in Cmake files>

aditi230, 2021-06-29:
**Briefly summarize the issue being raised**
Under FreeRtos directry I navigated to - > "amazon-freertos/demos/coreMQTT$" and to upload this in to esp32 devkitc I runned the command -> idf.py -p PORT flash 
After this received error -> 
No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as
CMake Error at CMakeLists.txt:2 (afr_demo_module):
  Unknown CMake command "afr_demo_module".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.16)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

**System information**
- Hardware board: [ esp32 devkitc]
- IDE used: [ ubuntu terminal ]
- Operating System: [ Linux ]
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [ coreMQTT]

yanjos-dev, 2021-06-30:
Hi @aditi230,

What version of this repository are you using?

There are some additional steps that are not listed in the sequence that you shared. Please see this website for the appropriate sequence of steps required to flash this demo to the esp32 devkitc platform. ([link](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif-overview))

aditi230, 2021-07-02:
hey @yanjos-dev ,
 the git version I am using is- 20210526_Archive-55-g265cd76d0.
I did follow the link provided still facing the issue.

yanjos-dev, 2021-07-02:
What directory are you attempting to run the demos from? In the case of these demos, you should be running the cmake commands from the root directory of the repository and not the demos directory.

Can you share the steps and errors that you are getting now? I would expect these to be different from your original post.

aditi230, 2021-07-09:
@yanjos-dev 
Configure cmake in the build directory and build the firmware image with the following command.
how to I configure cmake.
I am running demos from their folder like amazon-freertos/demos/coreMQTT.
And what is the cmake command you are referring to because I runned build command after setupAWS.py i.e. idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 build

aditi230, 2021-07-09:
[image]
This is the error what i am facing.

ravibhagavandas, 2021-07-11:
Hi @aditi230

To build the demo, you need to run cmake from the root folder, ie `amazon-freertos/`. To build and flash using cmake, please follow the instructions for the respective ESP IDF version (v4.2 or v3.3) in the [getting started guide for Espressif](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html). Amazon FreeRTOS v202012.00 uses ESP IDF v3.3 and the latest code in main branch uses IDF v4.2.

Please let us know if you have further queries.

aditi230, 2021-07-12:
I used the same link provided for it. and relating esp idf I downloaded v4.2 and amazon FreeRTOS  I cloned this git repository. Still I am facing the error.
I ruined build command in root directory that worked but then how will I flash the code in drive?

ravibhagavandas, 2021-07-13:
You can do build and flash with either `cmake` or Espressif `idf.py` tool. To use cmake, run:
```
make -C ./YOUR_BUILD_DIRECTORY erase_flash
make -C ./YOUR_BUILD_DIRECTORY flash
```
Noticed, there is a typo in the command you executed for: `earse_flash` instead of `erase_flash`.

For details take a look at the section **Build and flash FreeRTOS on Windows, Linux, and MacOS (ESP-IDF v4.2)** in ESP getting started guide: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf42

aditi230, 2021-07-13:
[image]
[image]
I tried both but still facing the error.

ravibhagavandas, 2021-07-13:
It seems you have generated CMake files using Ninja build system using the command :
```
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ./YOUR_BUILD_DIRECTORY -DAFR_ENABLE_ALL_MODULES=1 -DAFR_ENABLE_TESTS=0 -GNinja
````
If so you need to use ninja tool to erase and flash to the ESP board:
```
ninja -C <./YOUR_BUILD_DIRECTORY> erase_flash
ninja -C <./YOUR_BUILD_DIRECTORY> flash
```
See **With Ninja** section under step 4.

You can install ninja build tool from here: https://github.com/ninja-build/ninja/wiki/Pre-built-Ninja-packages

If you want to use `make` build system as above, remove the current build folder and recreate the build folder by providing option **-G"Unix Makefiles"**  as follows
```
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ./YOUR_BUILD_DIRECTORY -DAFR_ENABLE_ALL_MODULES=1 -DAFR_ENABLE_TESTS=0 -G"Unix Makefiles"
````

aditi230, 2021-07-14:
Thank was able to flash into device. But how Do I monitor now? As I flased coreMQTT project but how do I see the result in MQTT Test client.

ravibhagavandas, 2021-07-14:
Monitoring MQTT messages through AWS console is mentioned here:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#gsg-espressif-monitor-mqtt

For this you should have completed the prerequisite steps of creating and activating an AWS account and user as mentioned here:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-prereqs

Once an account is created you can use the `quick start` script to create a thing, certificate and policy in AWS and configure FreeRTOS demo with the details. The steps can be found under **Configure the FreeRTOS demo applications**

Finally, you can subscribe to the topic from test client in AWS console and then flash the demo onto the board.

Additionally you could also monitor logs on ESP32 by using the command:
```
make -C ./YOUR_BUILD_DIRECTORY monitor
```

aditi230, 2021-07-15:
Thank you so much for guidance.

closed 2021-07-16 by aditi230