aws/amazon-freertos #3293: [BUG] <While running build command in project directory facing error>

aditi230, 2021-06-30:
**Describe the bug**
Configure cmake in the build directory and build the firmware image with the following command.

idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build
directory is->amazon-freertos/vendors/espressif/esp-idf/examples/get-started/hello_world
When I runned this command i faced error as follow-:
**System information**
- Hardware board: [ esp32 devkitc ]
- IDE used: [ linux terminal]
- Operating System: [  Linux  ]
- Code version: (v4.2.1-21-gd58109d32)
- Project/Demo: [hello_word ]

How would I solve this?
Thank you in advance
[image]

yanjos-dev, 2021-06-30:
Hello @aditi230,

The example demo that you are attempting to run is part of the Espressif ESP-IDF repository and is not directly related to this repository. I'm unfamiliar with this demo, but it looks like the flags that you are using are specific to the demos in this directory:
https://github.com/aws/amazon-freertos/tree/main/demos

If you would like to run the demos inside of  ESP-IDF repository and are having issues, I'd recommend that you post a question to the [Espressif section of the FreeRTOS forums](https://forums.freertos.org/c/partners-sponsors/espressif/14).

aditi230, 2021-07-02:
okay Thankyou

closed 2021-07-06 by yanjos-dev