aws/amazon-freertos #3299: [General] AFR & ESP32: Firmware bug is causing OTA Update to fail. Still, any possibility of a successful OTA Update?

dipen-1337lab, 2021-07-02:
Namaste Developers,

### **Briefly summarize the issue being raised**

- Our product is at client site for testing. We have pushed multiple OTA updates, which have been successfully deployed on the device to test the newly implemented application features. OTA updates have always happened smoothly, until this time.

- The Amazon FreeRTOS specifics for our project are 
  - FreeRTOS Version is V202002.00.
  - coreHTTP library has been ported for use with V202002.00. More details [here](https://github.com/aws/amazon-freertos/issues/2911).
  - cmake is run with `-DAFR_ESP_LWIP=1` followed by `ninja` command.

- On-board Firmware Version: `0.9.8`. OTA update Firmware Version: `0.9.9` [This is our internal versioning.]

- The `Root Cause Analysis` on the OTA Update failure has been identified as a bug in the firmware, running on the board. 

- The OTA Update thread running on the device successfully downloads the entire firmware received over-the-air. It then attempts a reboot to load the new firmware. However, in the process of reboot, the MQTT disconnection via `IotMqtt_Disconnect()` function fails, causing the `Firmware Version: 0.9.8` to boot again instead of the newly received Firmware Version: `0.9.9`.

### **Describe the desired outcome**

- The bug fix corresponds to a **_code change in the firmware_**, which is running on the device at the client site. This has been validated locally and, the OTA Update is successful. Without this fix, the OTA is always failing.

- The product is at client site and, with the ESP32 DevKitC board assembled along with the application hardware inside an enclosure, opening the product to flash the new firmware over the USB-Serial would be our last preferred resort.

- **Considering the fact that the new firmware is successfully downloaded, is there a solution (which would be implemented in the new firmware being pushed via OTA), to ensure that the new firmware boots despite the current firmware crashing?**

### **System information**
- Hardware board: ESP32-DevKitC
- IDE used: Eclipse IDE for C/C++ Developers (includes Incubating components). [Version] 2020-06 (4.16.0). [Build Id] 20200615-1200
- Operating System: [Edition] Windows 10 Home Single Language ; [Version] 21H1 ; [OS Build] 19043.1052
- Code version: v0.9.4-108-g770be0b
- Project/Demo: Our custom application has the following 3 demos integrated - mqtt_demo_mutual_auth | coreHTTP | mqtt_demo

### **References**

Below are the **three console log blocks** associated with the issue. The entire log file is enclosed here.
[firmware_crash_causes_ota_update_failure.txt](https://github.com/aws/amazon-freertos/files/6753720/firmware_crash_causes_ota_update_failure.txt)

**Observation 1 - File receive complete and signature is valid.**

```
9170 39718 [OTA Agent Task] [prvIngestDataBlock] Received file block 1195, size 1024
9171 39719 [OTA Agent Task] [prvIngestDataBlock] Remaining: 2
9172 39719 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
9173 39721 [OTA Agent Task] [prvIngestDataBlock] Received file block 1196, size 1024
9174 39721 [OTA Agent Task] [prvIngestDataBlock] Remaining: 1
9175 39721 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
I (397305) dual_outlet: Timer 0 ID equals 0
9176 39724 [OTA Agent Task] [prvIngestDataBlock] Received file block 1197, size 208
I (397315) ota_pal: No such certificate file: /. Using aws_ota_codesigner_certificate.h.

9177 39725 [OTA Agent Task] [prvIngestDataBlock] Received final expected block of file.
9178 39725 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
9179 39740 [iot_thread] [INFO ][DEMO][397400] State: WaitingForFileBlock  Received: 2216   Queued: 0   Processed: 0   Dropped: 0

9180 39772 [iot_thread] [INFO ][DEMO][397720] INFO : Free Heap size = 24680.
9181 39772 [iot_thread] [INFO ][DEMO][397720] INFO : Minimum Ever Free Heap size = 7464.
I (397805) dual_outlet: Timer 0 ID equals 0
E (397885) dual_outlet: Outlet 1 - Error fetching Energy Meter readings.
I (397885) dual_outlet: EPOCH Time in seconds is: 1625125388
I (397885) dual_outlet: EPOCH Time in milliseconds is: 1625125388810
I (398305) dual_outlet: Timer 0 ID equals 0
9182 39840 [iot_thread] [INFO ][DEMO][398400] State: WaitingForFileBlock  Received: 2216   Queued: 0   Processed: 0   Dropped: 0

9183 39872 [iot_thread] [INFO ][DEMO][398720] INFO : Free Heap size = 21476.
9184 39872 [iot_thread] [INFO ][DEMO][398720] INFO : Minimum Ever Free Heap size = 7464.
I (398805) dual_outlet: Timer 0 ID equals 0
E (398885) dual_outlet: Outlet 1 - Error fetching Energy Meter readings.
I (398885) dual_outlet: EPOCH Time in seconds is: 1625125389
I (398885) dual_outlet: EPOCH Time in milliseconds is: 1625125389810
I (399305) dual_outlet: Timer 0 ID equals 0
[... 55 lines cut ...]

**Observation 3 - OTA Update Job failure**

```
46 1324 [OTA Agent Task] [prvParseJobDoc] In self test mode.
47 1324 [OTA Agent Task] [prvParseJobDoc] Failing job. We rebooted and the version is still the same.
I (13305) ota_pal: prvPAL_SetPlatformImageState, 3
W (13405) ota_pal: Set image as invalid!
I (13405) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (13415) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
W (13415) ota_pal: Image not in self test mode 2
I (13425) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (13435) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
48 1337 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x23000000"}}
49 1337 [OTA Agent Task] [INFO ][MQTT][13370] (MQTT connection 0x3ffe6fbc) MQTT PUBLISH operation queued.
50 1339 [OTA Agent Task] [INFO ][MQTT][13370] (MQTT connection 0x3ffe6fbc, PUBLISH operation 0x3fff14e0) Waiting for operation completion.
51 1361 [OTA Agent Task] [INFO ][MQTT][13610] (MQTT connection 0x3ffe6fbc, PUBLISH operation 0x3fff14e0) Wait complete with result SUCCESS.
52 1361 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] 'FAILED' to $aws/things/esp32DevKitC_DFAFE9/jobs/AFR_OTA-26/update
53 1361 [OTA Agent Task] [prvResetDevice] Attempting forced reset of device...
```

Let me know if any additional details are required on this issue.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-07-07:
Namaste,

> ### **Describe the desired outcome**
> * The bug fix corresponds to a **_code change in the firmware_**, which is running on the device at the client site. This has been validated locally and, the OTA [...]

Requesting assistance for a possible solution on this issue.

Thanks | Regards,
Dipen

ravibhagavandas, 2021-07-09:
Hi Dipen,
After checking the OTA ESP PAL code in that version, the boot partition is set to the new image in the final step after disconnecting the MQTT connection. So it boots up with old image from the old partition since crash occurs before that. 
I will ask my colleagues from OTA team to see if there are any workarounds.

dipen-1337lab, 2021-07-12:
Namaste @ravibhagavandas, 

Many thanks for responding with your analysis. So the behaviour is indeed as per the firmware logic implemented.

> I will ask my colleagues from OTA team to see if there are any workarounds.

A friendly follow-up for any potential workarounds.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-07-19:
Namaste @ravibhagavandas,

Requesting for an update on this issue.

Thanks | Regards,
Dipen

ravibhagavandas, 2021-07-19:
Hi @dipen-1337lab 
One of my peers @pvyawaha will be reaching out to you soon, to see if there are any workarounds. Apologies for the wait and thanks for the patience.

dipen-1337lab, 2021-07-23:
Namaste @pvyawaha ,

Requesting your assistance on this issue.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-08-11:
Namaste @pvyawaha and @ravibhagavandas ,

It's been more than a month since I've opened this issue. Is there any potential workaround?

Thanks | Regards,
Dipen

dipen-1337lab, 2021-08-12:
Namaste @pvyawaha ,

> Are you still facing this issue

- Short answer - Yes, we are still facing this issue.
- Long answer - Our product at client site is currently not in use, since we are unable to push an OTA update. The device is disassembled. However, the issue can definitely be reproduced again in our development environment. So again, yes we are still facing this issue.

> ### **Describe the desired outcome**

- For our unique scenario, we would like to understand whether the **new device firmware**, which is sent as an OTA, would be able to run despite the **existing device firmware** crashing during reboot.
- If feasible, the corresponding changes can be incorporated in the **new device firmware**, which we will send as an OTA update.

> Are you planning to move to the official release or looing to fix the integration that you have

At this juncture, we are not considering a migration. We are using an older version 202002.00 and hence, will prefer a fix to the integration we have.

Thanks | Regards,
Dipen

pvyawaha, 2021-08-24:
Hello Dipen, 

Regarding -

> For our unique scenario, we would like to understand whether the new device firmware, which is sent as an OTA, would be able to run despite the existing device firmware crashing during reboot.
> If [...]

The  new firmware when downloaded will be executed in self-test mode and accepted if it connects to the service and updates the status. So when device reboots after successful download of a firmware then it will be executing the new firmware and should not crash.

dipen-1337lab, 2021-08-24:
Namaste @pvyawaha,

Thank you for your response!

> The new firmware when downloaded will be executed in self-test mode and accepted if it connects to the service and updates the status.

As can be understood from the below logs, the **self-test** begins after a reboot. Please refer the `firmware_crash_causes_ota_update_failure.txt` the log file attached in the original post for complete logs, if required.

```
46 1324 [OTA Agent Task] [prvParseJobDoc] In self test mode.
47 1324 [OTA Agent Task] [prvParseJobDoc] Failing job. We rebooted and the version is still the same.
I (13305) ota_pal: prvPAL_SetPlatformImageState, 3
W (13405) ota_pal: Set image as invalid!
```

Unfortunately, the problem occurs before reboot, as aptly identified and explained by @ravibhagavandas though his below comment.

> Hi Dipen,
> After checking the OTA ESP PAL code in that version, the boot partition is set to the new image in the final step after disconnecting the MQTT connection. So it boots up with old image [...]

Due to a MQTT disconnection related bug in our application code, the MQTT disconnection failure causes the firmware to crash. At this point, the boot partition is not set to the new image. Hence, after reboot the same firmware loads and runs again!

### This issue is a very unique scenario

- As mentioned earlier, I understand this issue as being an exceptional scenario due to a bug in our application. This, in no way is related to OTA functionality, neither from ESP-IDF side nor from AFR side.

- If there is no workaround possible for this scenario, we may close this issue!

Thanks | Regards,
Dipen

pvyawaha, 2021-08-30:
Hello @dipen-1337lab ,

Thank you for sharing more details, if the application that is flashed on the device is having bug as you mentioned it will have to be re-flashed with fix. The self-test phase will help avoiding such bugs if the images are sent over the air. I am closing this issue now.

closed 2021-08-30 by pvyawaha