aws/amazon-freertos #3312: ESP32 second core SMP

smartnimbus, 2021-07-09:
HI All
I am currently running release version 202012.00. on ESP32-WROVER-IE (8MB). 
How can I use the second core with this version? I tried xTaskCreatePinnedToCore however this is just a reap of xTaskCreate
int xTaskCreatePinnedToCore( void (*pxTaskCode)(void*),
                            const char * const pcName,
                            const uint32_t usStackDepth,
                            void * const pvParameters,
                            unsigned uxPriority,
                            void * const pxCreatedTask,
                            const int xCoreID )
{
	return xTaskCreate(pxTaskCode, pcName, usStackDepth, pvParameters, uxPriority, pxCreatedTask);
}

And xTaskCreateStaticPinnedToCore is not part of the code.

I did go through this link; however, I am new to freertos and did not understand what I had to do to make it work. 
https://github.com/aws/amazon-freertos/pull/1248

**System information**
- Hardware board: [ ESP32-WROVER-IE (8MB).]
- IDE used: [ VS Code ]
- Operating System: [ Windows]
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [shadow_demo | custom application]

ravibhagavandas, 2021-07-09:
Hi @smartnimbus 

Currently this repository does not support the SMP version of FreeRTOS from Espressif. We are working on a SMP feature for FreeRTOS kernel. The blog post for this, can be found [here](https://www.freertos.org/2021/06/introducing-the-freertos-symmetric-multiprocessing-smp-github-branch.html) and the github SMP branch can be found [here](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/smp). It does not support SMP for ESP32 yet.

You can still use ESP32 dual core version of FreeRTOS from esp-idf [repository](https://github.com/espressif/esp-idf) along with our AWS IoT CSDKs.

muneebahmed10, 2021-07-21:
Hi @smartnimbus,

Did the above response answer your question? We will close this issue if we do not hear back from you.

smartnimbus, 2021-07-22:
as I am using greengrass i am not able to use the AWS IoT Device SDK for Embedded C, therefore i will have to Wait until  greengrass is part of the Device SDK or you have  SMP enabled in FreeRTOS .

closed 2021-07-22 by smartnimbus

samuelbles07, 2021-11-09:
Hi @ravibhagavandas , is there any estimation when SMP capability will be available to amazon-freertos project especially for ESP32?

Thanks