aws/amazon-freertos #3313: Some times Shadow reporting/publishing API fails on device and shadow publishing fails

kumarfirmware, 2021-07-11:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

ESP32 is connected to the corporate network Wi-Fi. Very High speed network. No chance of unstable network.
Also ESP does not loose ping at all. Ping is very stable.

When I am trying to send the shadow update to the cloud using the following function, sometimes I get the error "Failed to send Shadow update error MQTT LIBRARY ERROR".

Some times I get the error, Wait complete with result NETWORK ERROR, Wait complete with result TIMEOUT.

When will these errors occur? What will be the root cause of these errors?

I have a stable Wifi connection and when these errors occur, I check the ping of the IP address and everything is good.

_sendShadowUpdates(IotMqttConnection_t mqttConnection,
const char * const pThingName,
size_t thingNameLength );

static int _setShadowCallbacks(IotMqttConnection_t mqttConnection,
const char * pThingName,
size_t thingNameLength );

updateStatus = AwsIotShadow_TimedUpdate( mqttConnection,
&updateDocument,
AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS,
5000 );

Will I be able to publish the shadow when an OTA job is in progress?

I am running two tasks in my application and one more task is OTA task which is internal to AWS SDK. So in total three tasks.
The task which reports shadows has highest priority compared to the other application task. I am not sure about the priority of the OTA task as it is internal to AWS SDK.

I understand we can use same MQTT connection for Shadow publishing and normal messages publishing and OTA update.

But the question here is for example when OTA update is happening, at the same time I want to publish shadow, at the same time I want to publish application message. Is this possible? If possible how does this work? Please let me know as soon as possible.

Please let me know as soon as possible.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-07-13:
Hi @kumarfirmware 
You could share the mqtt connection (and underlying TLS connection) with  multiple tasks and the MQTT library in FreeRTOS v202007.00 takes care of thread safety of the libraries. Few questions, which would help troubleshoot this further:
1. Could you reproduce the issue by enabling debug logs for MQTT and Network by defining following configs in [iot_config.h](https://github.com/aws/amazon-freertos/blob/202007.00/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L49) as follows:
```
#define IOT_LOG_LEVEL_NETWORK                   IOT_LOG_DEBUG
#define IOT_LOG_LEVEL_MQTT                      IOT_LOG_DEBUG
```
2. Are you seeing NETWORK_ERROR only for shadow updates or for your application or OTA tasks as well ? What is the size of the shadow document being sent ?
3. When there is a NETWORK_ERROR do you see connection termination with broker ?

kumarfirmware, 2021-07-13:
Hi Ravi,

Thanks for the message. 

1. I can share MQTT connection across multiple tasks/ threads. This is clear.

2. I cannot believe that AWS FreeRTOS version V202007 MQTT functions are Thread safe. Can you please provide concrete evidence in the form of documentation that MQTT functions are Thread safe. Can you please confirm that I need not worry about any synchronization mechanism while using MQTT functions across multiple threads/tasks.

3. Did you ever test this using MQTT library functions across the multiple threads simultaneously with out using any  synchronization mechanism. Please confirm this.

4. Again the same question assume OTA is happening in Thread1, Thread2 is publishing shadow, Thread 3 is publishing application message simultaneously with out any synchronization mechanism between three threads. Is this possible. Please confirm.

5. If the answer is yes, please explain how it is working internally?

6. Please reply or confirm as soon as possible, based on your answer I need to change the design/implementation ASAP.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-07-14:
Hi @kumarfirmware 

The docs not explicitly mention the thread safety of the library:
https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/mqtt_design.html#mqtt_design_asyncoperation
However, library uses task pool to schedule MQTT operations in background which interacts with the network. The network interface implementation uses a [mutex](https://github.com/aws/amazon-freertos/blob/202007.00/libraries/abstractions/platform/freertos/iot_network_freertos.c#L487)  to ensure only one sender sends data. Incoming messages are received using a single [receiver task](https://github.com/aws/amazon-freertos/blob/202007.00/libraries/abstractions/platform/freertos/iot_network_freertos.c#L142). Library APIs by itself uses mutex for mutual exclusion of internal shared data structures used.

> Please reply or confirm as soon as possible, based on your answer I need to change the design/implementation ASAP.

I would recommend your design to not depend on the 202007 version of mqtt library as its currently  not actively maintained. We have released LTS versions of new [coreMQTT](https://github.com/FreeRTOS/coreMQTT ) library and thread safe version of the library using [coreMQTT Agent]( https://github.com/FreeRTOS/coreMQTT-Agent).  You can find examples of using OTA, MQTT and Shadow tasks concurrently using coreMQTT agent [here](https://github.com/FreeRTOS/coreMQTT-Agent-Demos).

kumarfirmware, 2021-07-15:
Hi Ravi,

Thanks for the message.

This is clear. Until we move to LTS which will take time to port, we need to live with 202007.

I will test this and get back to you if I see any issues.

Thanks, Regards,
Kumar.

kumarfirmware, 2021-07-24:
Hi Ravi,

Please see the screenshot attached here.

Can you tell why this error occurred? what could be the reason for this error?

Error -27648 while sending data.

Out of no where this error comes. This comes from OTA ? why does this error happen?

If you can tell me the root cause of this issue, many problems can be solved.
[image]

Thanks, Regards,
Kumar.

kumarfirmware, 2021-07-27:
Hi Ravi,

Can you please help me with the above error as soon as possible.

Thanks, Regards,
Kumar.

Raghav3107, 2021-07-27:
Hello @kumarfirmware 
I m don't know the exact solution to this. But I can give you some advice from my past experience, I faced this problem in past so that might help you. Check the `pUpdateDocument` after `snprintf`, check the desired or reported value you are adding is in the right format. When I faced this issue somehow desired value was adding the garbage value in  `pUpdateDocument`.

Hope this will help you!

kumarfirmware, 2021-07-28:
Thanks Raghav for your advice. I cross checked this and it is fine.

Also the above issue mentioned with the screenshot is different.

kumarfirmware, 2021-07-28:
Hi Ravi,

Please see the screenshot attached here.

Can you tell why this error occurred? what could be the reason for this error?

Error -27648 while sending data.

Out of no where this error comes. This comes from OTA ? why does this error happen?

If you can tell me the root cause of this issue, many problems can be solved.

This is easily reproduceable. This happens easily. Wifi ping is very stable when this error occurs. I mean wifi is very good when this happens.

OTA_error

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-07-28:
@kumarfirmware 

Error -27648 (-0x6C00) translates to mbedtls error code [MBEDTLS_ERR_SSL_INTERNAL_ERROR ](https://github.com/ARMmbed/mbedtls/blob/v2.26.0/include/mbedtls/ssl.h#L116), which is emitted from a lower layer in mbedtls. You could try printing the low level and high level error string for error code `-27648`  using the functions `mbedtls_strerror_highlevel()` and `mbedtls_strerror_lowlevel()` defined [here](https://github.com/aws/amazon-freertos/blob/202007.00/libraries/3rdparty/mbedtls_utils/mbedtls_error.h#L43).

Taking a closer look at the timing of the error, it happens 2969 ticks after successful log. Tick rate for ESP in that release version is 100 HZ so it means the error occurs roughly after 29.69 secs. What is the keep alive interval for your connection ? Is the connection getting closed due to a keep alive timeout?

kumarfirmware, 2021-07-29:
Hi Ravi,

Thanks for the message. I did not understand the following.

Keep alive interval for MQTT connection I configured as 30 seconds.

Message flow is like this.

"Error -27468 while sending data.
Failed to send PINGREQ
Network connection closed.
MQTT connection closed due to keep-alive response not received."

**These kind of strange underlying issues are causing stability/reliability issues in the production release.

We need to fix them immediately. This is absolutely not good and need to be fixed immediately. we do not have much time left to fix this. Need your immediate assistance.**

1. Is it a problem to configure Keep alive interval for MQTT connection as 30 seconds?
2. In the MQTT disconnect callback function there is no need to call this function again IotMqtt_Disconnect() right before I do a reconnect? Is there a auto reconnect function for MQTT connection?
3. When I call OTA_AgentShutdown() in MQTT disconnect callback function it goes in to infinite loop. The code snippet is as following. This seems to be a bug or something wrong here because all the time it is in "eOTA_AgentState_ShuttingDown" this state after calling OTA_AgentShutdown(0);  Why is it not able to come out of this state and it is in infinite loop? why is it not moving to "eOTA_AgentState_Stopped"?

static void mqtt_disconnect_callback(void* param, IotMqttCallbackParam_t* mqttCallbackParams)
{
    (void)param;
    OTA_State_t eState;

    // Log the reason for MQTT disconnect.
    switch (mqttCallbackParams->u.disconnectReason)
    {
        case IOT_MQTT_DISCONNECT_CALLED:
            IotLogInfo("Mqtt disconnected due to invoking diconnect function.\r\n");
            break;

        case IOT_MQTT_BAD_PACKET_RECEIVED:
            IotLogInfo("Mqtt disconnected due to invalid packet received from the network.\r\n");
            break;

        case IOT_MQTT_KEEP_ALIVE_TIMEOUT:
            IotLogInfo("Mqtt disconnected due to Keep-alive response not received.\r\n");
            break;

        default:
            IotLogInfo("Mqtt disconnected due to unknown reason.");
            break;
    }

OTA_AgentShutdown(0);

  while ((eState = OTA_GetAgentState()) != eOTA_AgentState_Stopped)
  {
      vTaskDelay(myappONE_SECOND_DELAY_IN_TICKS);
  }

}

Thanks, Regards,
Kumar.

kumarfirmware, 2021-07-30:
Hi Ravi,

Can you kindly help me with the above error as soon as possible.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-07-31:
1. Is it a problem to configure Keep alive interval for MQTT connection as 30 seconds?
It should not be,  keep alive interval can be set to any value  between 30 seconds to 1200 seconds as specified in doc [here](https://docs.aws.amazon.com/general/latest/gr/iot-core.html). From the logs above it appears error happens when it tries to send a Ping req.

To troubleshoot further please provide the following details.

1. Debug logs for MQTT and Network by setting the following log level in `iot_config.h`:
```
#define IOT_LOG_LEVEL_NETWORK                   IOT_LOG_DEBUG
#define IOT_LOG_LEVEL_MQTT                      IOT_LOG_DEBUG
```
2. The mbedtls high level and low level error string for the error code returned.

kumarfirmware, 2021-08-02:
Hi Ravi,

"You could try printing the low level and high level error string for error code -27648 using the functions mbedtls_strerror_highlevel() and mbedtls_strerror_lowlevel()"

where should I print them? which part of the code I should print them? Please let me know.

This is very high priority. We need to fix this ASAP.

Thanks, Regards,
Kumar.

kumarfirmware, 2021-08-04:
Hi Ravi,

Can you please help me with the above error as soon as possible. There is no response from you.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-08-05:
Hello,

I did a test by dumping the SSL error code string value but could not get meaningful information. The high level error string for `-27468` as returned from mbedtls is `SSL - Internal error (eg, unexpected failure in lower-level module)` and there is no low level error code for it.

With only the info logs above, it is hard to find out the exact reason for the failure, so could you **enable debug logs** for MQTT and Network libraries and send the results ?

I don't see an issue with shadow document as the last shadow update was successful from logs. Also the error code returned from OTA `603979776` is `kOTA_Err_JobParserError` which is displayed since there was no active OTA job.

Apart from turning on debug logs, you could follow troubleshooting steps from cloud side to detect device disconnections as mentioned here:
https://docs.aws.amazon.com/iot/latest/developerguide/ota-troubleshooting-fleet-disconnects.html

Server can disconnect a device for number of reasons, and the disconnection reason can be received as events for that device client ID, by subscribing to relevant topics. More details here:
https://docs.aws.amazon.com/iot/latest/developerguide/life-cycle-events.html#connect-disconnect

Hope this helps.

kumarfirmware, 2021-08-05:
Hi Ravi,

Thanks for the message.

1. I will enable debug logs for MQTT and Network libraries and send the results to you.
2. If it is a cloud side disconnection, we should not get ping failed. We should get MQTT disconnect call back with reason. This is not the case here.
3. For sure this is internal to ESP in my opinion. 
4. Also I assume this will not get fixed with out a reboot of ESP. 
5. This will be a disaster and will block production. We need to fix this ASAP.

Questions from before:-

1. How and where can I print the BLE advertisement data and see what data device is advertising in BLE? Basically I want to see exactly what this function is putting and sending the advertisement data IotBle_StartAdv( NULL )?

2.In the MQTT disconnect callback function there is no need to call this function again IotMqtt_Disconnect() right before I do a reconnect? Is there a auto reconnect function for MQTT connection?

3. When I call OTA_AgentShutdown() in MQTT disconnect callback function it goes in to infinite loop. The code snippet is as following. This seems to be a bug or something wrong here because all the time it is in "eOTA_AgentState_ShuttingDown" this state after calling OTA_AgentShutdown(0); Why is it not able to come out of this state and it is in infinite loop? why is it not moving to "eOTA_AgentState_Stopped"?
static void mqtt_disconnect_callback(void* param, IotMqttCallbackParam_t* mqttCallbackParams)
{
(void)param;
OTA_State_t eState;

// Log the reason for MQTT disconnect.
switch (mqttCallbackParams->u.disconnectReason)
{
    case IOT_MQTT_DISCONNECT_CALLED:
        IotLogInfo("Mqtt disconnected due to invoking diconnect function.\r\n");
        break;

    case IOT_MQTT_BAD_PACKET_RECEIVED:
        IotLogInfo("Mqtt disconnected due to invalid packet received from the network.\r\n");
        break;

    case IOT_MQTT_KEEP_ALIVE_TIMEOUT:
        IotLogInfo("Mqtt disconnected due to Keep-alive response not received.\r\n");
        break;

    default:
        IotLogInfo("Mqtt disconnected due to unknown reason.");
        break;
}
OTA_AgentShutdown(0);

while ((eState = OTA_GetAgentState()) != eOTA_AgentState_Stopped)
{
vTaskDelay(myappONE_SECOND_DELAY_IN_TICKS);
}

}

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-08-16:
I  downloaded  release 202007.00, and  with the provided demos, I could not reproduce this error. Here are the  debug logs from my test, where PINGREQ is successfully sent:

```83 759 [iot_thread] [DEBUG][MQTT][7590] (MQTT connection 0x3ffb5e6c) Reference count changed from 3 to 2.
84 759 [iot_thread] [INFO ][DEMO][7590] All demo topic filter subscriptions accepted.
85 3730 [iot_thread] [DEBUG][MQTT][37300] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
86 3730 [iot_thread] [DEBUG][MQTT][37300] (MQTT connection 0x3ffb5e6c) Sending PINGREQ.
87 3730 [iot_thread] [DEBUG][MQTT][37300] (MQTT connection 0x3ffb5e6c) PINGREQ sent. Scheduling check for PINGRESP in 10000 ms.
88 3730 [iot_thread] [DEBUG][MQTT][37300] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 10000 ms.
89 3733 [NetRecv] [DEBUG][NET][37330] Successfully received 1 bytes.
90 3733 [NetRecv] [DEBUG][NET][37330] Successfully received 1 bytes.
91 3734 [NetRecv] [DEBUG][MQTT][37340] (MQTT connection 0x3ffb5e6c) PINGRESP in data stream.
92 3735 [NetRecv] [DEBUG][MQTT][37350] (MQTT connection 0x3ffb5e6c) PINGRESP successfully parsed.
93 4731 [iot_thread] [DEBUG][MQTT][47310] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
94 4731 [iot_thread] [DEBUG][MQTT][47310] (MQTT connection 0x3ffb5e6c) Checking for PINGRESP.
95 4731 [iot_thread] [DEBUG][MQTT][47310] (MQTT connection 0x3ffb5e6c) PINGRESP was received.
96 4731 [iot_thread] [DEBUG][MQTT][47310] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 20000 ms.
97 6732 [iot_thread] [DEBUG][MQTT][67320] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
98 6732 [iot_thread] [DEBUG][MQTT][67320] (MQTT connection 0x3ffb5e6c) Sending PINGREQ.
99 6732 [iot_thread] [DEBUG][MQTT][67320] (MQTT connection 0x3ffb5e6c) PINGREQ sent. Scheduling check for PINGRESP in 10000 ms.
100 6732 [iot_thread] [DEBUG][MQTT][67320] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 10000 ms.
101 6735 [NetRecv] [DEBUG][NET][67350] Successfully received 1 bytes.
102 6735 [NetRecv] [DEBUG][NET][67350] Successfully received 1 bytes.
103 6736 [NetRecv] [DEBUG][MQTT][67360] (MQTT connection 0x3ffb5e6c) PINGRESP in data stream.
104 6737 [NetRecv] [DEBUG][MQTT][67370] (MQTT connection 0x3ffb5e6c) PINGRESP successfully parsed.
105 7733 [iot_thread] [DEBUG][MQTT][77330] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
106 7733 [iot_thread] [DEBUG][MQTT][77330] (MQTT connection 0x3ffb5e6c) Checking for PINGRESP.
107 7733 [iot_thread] [DEBUG][MQTT][77330] (MQTT connection 0x3ffb5e6c) PINGRESP was received.
108 7733 [iot_thread] [DEBUG][MQTT][77330] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 20000 ms.
109 9734 [iot_thread] [DEBUG][MQTT][97340] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
110 9734 [iot_thread] [DEBUG][MQTT][97340] (MQTT connection 0x3ffb5e6c) Sending PINGREQ.
111 9734 [iot_thread] [DEBUG][MQTT][97340] (MQTT connection 0x3ffb5e6c) PINGREQ sent. Scheduling check for PINGRESP in 10000 ms.
112 9734 [iot_thread] [DEBUG][MQTT][97340] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 10000 ms.
113 9737 [NetRecv] [DEBUG][NET][97370] Successfully received 1 bytes.
114 9737 [NetRecv] [DEBUG][NET][97370] Successfully received 1 bytes.
115 9738 [NetRecv] [DEBUG][MQTT][97380] (MQTT connection 0x3ffb5e6c) PINGRESP in data stream.
116 9739 [NetRecv] [DEBUG][MQTT][97390] (MQTT connection 0x3ffb5e6c) PINGRESP successfully parsed.
117 10735 [iot_thread] [DEBUG][MQTT][107350] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
118 10735 [iot_thread] [DEBUG][MQTT][107350] (MQTT connection 0x3ffb5e6c) Checking for PINGRESP.
119 10735 [iot_thread] [DEBUG][MQTT][107350] (MQTT connection 0x3ffb5e6c) PINGRESP was received.
120 10735 [iot_thread] [DEBUG][MQTT][107350] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 20000 ms.
121 12736 [iot_thread] [DEBUG][MQTT][127360] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
122 12736 [iot_thread] [DEBUG][MQTT][127360] (MQTT connection 0x3ffb5e6c) Sending PINGREQ.
123 12737 [iot_thread] [DEBUG][MQTT][127370] (MQTT connection 0x3ffb5e6c) PINGREQ sent. Scheduling check for PINGRESP in 10000 ms.
124 12737 [iot_thread] [DEBUG][MQTT][127370] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 10000 ms.
125 12739 [NetRecv] [DEBUG][NET][127390] Successfully received 1 bytes.
126 12739 [NetRecv] [DEBUG][NET][127390] Successfully received 1 bytes.
127 12740 [NetRecv] [DEBUG][MQTT][127400] (MQTT connection 0x3ffb5e6c) PINGRESP in data stream.
128 12741 [NetRecv] [DEBUG][MQTT][127410] (MQTT connection 0x3ffb5e6c) PINGRESP successfully parsed.
129 13738 [iot_thread] [DEBUG][MQTT][137380] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
130 13738 [iot_thread] [DEBUG][MQTT][137380] (MQTT connection 0x3ffb5e6c) Checking for PINGRESP.
131 13738 [iot_thread] [DEBUG][MQTT][137380] (MQTT connection 0x3ffb5e6c) PINGRESP was received.
132 13738 [iot_thread] [DEBUG][MQTT][137380] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 20000 ms.
133 15739 [iot_thread] [DEBUG][MQTT][157390] (MQTT connection 0x3ffb5e6c) Keep-alive job started.
134 15739 [iot_thread] [DEBUG][MQTT][157390] (MQTT connection 0x3ffb5e6c) Sending PINGREQ.
135 15739 [iot_thread] [DEBUG][MQTT][157390] (MQTT connection 0x3ffb5e6c) PINGREQ sent. Scheduling check for PINGRESP in 10000 ms.
136 15739 [iot_thread] [DEBUG][MQTT][157390] (MQTT connection 0x3ffb5e6c) Next keep-alive job in 10000 ms.
137 15742 [NetRecv] [DEBUG][NET][157420] Successfully received 1 bytes.
[... cut ...]
The output should have something like below:
```
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0009 len=32
0x09 0x20 0x20 0x0c 0x02 0x01 0x06 0x02 0x0a 0x03 0x05 0x12 0x20 0x00 0x40 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
D (2147) phy_init: wifi mac time delta: 23683
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x9 status=0 
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
```

ravibhagavandas, 2021-08-16:
Regarding the question:

> When I call OTA_AgentShutdown() in MQTT disconnect callback function it goes in to infinite loop. The code snippet is as following. This seems to be a bug or something wrong here because all the [...]

I will let @pvyawaha answer this question.

kumarfirmware, 2021-08-23:
Hi Ravi,

Thanks for the message. How much time you ran the test that matters a lot to reproduce the issue.

1. From my observation I always have a feeling of this two close publish messages will result in the error -27468. I mean from two different threads if I am trying to publish almost at the same time. 

**can you please confirm publishing messages from two different threads at the same time is a problem or not?**

2.  Some times even ping also throws this error -27468.

3. This -27468 is really random and it is there. Can you dig deep and see when it can occur, what makes it to occur? when it occurs why it occurs continuously? why it does not recover on reboot? 

4. Some times the strange part is shadow is published successfully but the PING throws this error -27468.

5. Another strange part is Wifi will be there but not able to establish MQTT connection, sometimes even after reboot.

We really need to get to bottom of this. How can we proceed.

Please find the MQTT disconnect call back code snippet as follows. Please let me know if u see any issue here.

static void mqtt_disconnect_callback(void* param, IotMqttCallbackParam_t* mqttCallbackParams)
{
(void)param;
OTA_State_t eState;
// Log the reason for MQTT disconnect.
switch (mqttCallbackParams->u.disconnectReason)
{
case IOT_MQTT_DISCONNECT_CALLED:
IotLogInfo("Mqtt disconnected due to invoking diconnect function.\r\n");
break;

case IOT_MQTT_BAD_PACKET_RECEIVED:
    IotLogInfo("Mqtt disconnected due to invalid packet received from the network.\r\n");
    break;

case IOT_MQTT_KEEP_ALIVE_TIMEOUT:
    IotLogInfo("Mqtt disconnected due to Keep-alive response not received.\r\n");
    break;

default:
    IotLogInfo("Mqtt disconnected due to unknown reason.");
    break;
}

// call MQTT disconnect function
  IotMqtt_Disconnect(mqttConnection, false);
}

Can you please reply as fast as possible as this is a big mess. I really need your urgent help.

Thanks,
Kumar.

ravibhagavandas, 2021-08-24:
Hi Kumar,

Looking at the implementation and [documentation](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/mqtt_function_disconnect.html) for `IotMqtt_Disconnect()` API , 

1. It's not safe to call this API from within a disconnect callback. The API modifies the mqtt connection object, so it should always be the last operation to be called after all other MQTT operations are complete. 
2. As part of closing the underlying connection, API also calls `mqtt_disconnect_callback` so you can end up in an infinite loop, if called from within the `mqtt_disconnect_callback`.
3.  Also if the network is already offline, then the API should be called with `IotMqtt_Disconnect(mqttConnection, IOT_MQTT_FLAG_CLEANUP_ONLY)` so that only cleanup of resources will be performed and  a DISCONNECT packet will not be send again.

How many tasks are you running which shares the MQTT connection ? What are the priority of each tasks and what are each tasks doing ? Are you ensuring the high priority tasks yields processing time to lower priority tasks ? Also from code snippet above it seems the disconnect operation does not wait for all other tasks before disconnecting and cleaning up the MQTT connection.

Answers to this could help reproduce the issue and debug further.

ravibhagavandas, 2021-08-30:
Hi,

I ran three tasks which subscribes, publishes and then unsubscribes by calling the MQTT library APIs. I ran this for 10 minutes but couldn't reproduce the error `-27648` as you posted. Here is an example of each task I was running:
```
void mqttTask( void * pvParameters )
{ 
    IotMqttError_t status  = IOT_MQTT_SUCCESS;
    IotMqttPublishInfo_t publishInfo = { 0 };
    const char *payload = "HELLOWORLD";
    size_t payloadLength = strlen(payload);
    const char *pTopic = "/foobar";
    size_t topicLength = strlen(pTopic);
    IotMqttSubscription_t subscriptionInfo = { 0 };

    for(;;)
    {
        publishInfo.pPayload = payload;
        publishInfo.payloadLength = payloadLength;
        publishInfo.pTopicName = pTopic;
        publishInfo.topicNameLength = topicLength;
        publishInfo.qos = IOT_MQTT_QOS_1;

        subscriptionInfo.pTopicFilter = pTopic;
        subscriptionInfo.topicFilterLength = topicLength;
        subscriptionInfo.qos = IOT_MQTT_QOS_1;
        subscriptionInfo.callback.function = logPacket;
        subscriptionInfo.callback.pCallbackContext = NULL;

        IotLogInfo("Subscribing..");

        status = IotMqtt_TimedSubscribe( mqttConnection, &subscriptionInfo, 1, 0,  2000 );
        if( status != IOT_MQTT_SUCCESS)
        {
            IotLogError("Failed to subscribe from task3");
            break;
        }
        
        IotLogInfo("Publishing..");

        status = IotMqtt_TimedPublish( mqttConnection, &publishInfo, 0,  2000 );
        if( status != IOT_MQTT_SUCCESS)
        {
            IotLogError("Failed to publish from task3");
            break;
        }

        IotLogInfo("Unsubscribing..");

        status = IotMqtt_TimedUnsubscribe( mqttConnection, &subscriptionInfo, 1, 0,  2000 );
        if( status != IOT_MQTT_SUCCESS)
        {
            IotLogError("Failed to subscribe from task3");
            break;
        }

        vTaskDelay(pdMS_TO_TICKS(50));
    }

    vTaskDelete( NULL );
}
```
Whereas from the screenshot you posted the error occurs roughly after 45 seconds. So the application code and sequence of operations in your case is different. Can you paste the entire logs from start for the screenshot above, so that I can better understand your setup ?  What is the payload size for the MQTT and Shadow messages sent or received ? Is there an OTA download in progress or queued while the issue occurs ? 

Answers to other questions:

> IotMqtt_Disconnect() API need not be called, because we already registered for mqtt_disconnect_callback(), that means already connection is disconnected. Do you agree for this?
Its recommended to call `IotMqtt_Disconnect()` even when the connection is unusable so as to free up resources and avoid a memory leak.

> IotMqtt_Disconnect() API should be called only explicitly if client wants to disconnect the connection intentionally. Do you agree for this?

Not as per my above comment.

>Even when I call IotMqtt_Disconnect() API , I get mqtt_disconnect_callback() triggered, you can test this easily. This is the evidence. Do you agree for this?

Thats true, based on my experiments, but I haven't found it in the documentation.

> Is is mandatory to call IotMqtt_Disconnect(mqttConnection, IOT_MQTT_FLAG_CLEANUP_ONLY) for reconnection, otherwise will there be a memory leak?

See my above comment. Its recommended to call this with `IOT_MQTT_FLAG_CLEANUP_ONLY` flag.

kumarfirmware, 2021-08-31:
Hi Ravi,

Thanks for the detailed explanation. 

1. 10 mins is very less time for this test. May be you can run it for more time.
2. But for me it is easily reproduceable. So something is mystery here, what is that we are not able to find out.
3. I attached the complete log here, you can analyze this very easily. 
4. What is the payload size for the MQTT and Shadow messages sent or received ? Is there an OTA download in progress or queued while the issue occurs ? - Payload size for shadow is 1500 bytes. Payload for MQTT publish is 300 bytes. OTA download is not in progress. But OTA is initialized.
5. I am using IotMqtt_Publish but you are using IotMqtt_TimedPublish. Will this be a problem? please confirm.

We do not have much time left. We need to crack this mystery. Hopefully today you will crack it.

Thanks, Regards,
Kumar.
[Error_Log.txt](https://github.com/aws/amazon-freertos/files/7082260/Error_Log.txt)

ravibhagavandas, 2021-11-01:
Hello,

I am closing this issue, since this is related to old beta version of MQTT library which is no longer maintained. Please feel free to open a new issue if you have any questions with the new released coreMQTT library.

closed 2021-11-01 by ravibhagavandas