aws/amazon-freertos #3314: If Network manager is not used to establish MQTT connection, program crashes while establishing MQTT connection.

kumarfirmware, 2021-07-11:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

**I do not want to use Network Manager at all in my code.**

**If Network manager is not used to establish MQTT connection, program crashes while establishing MQTT connection.**

The program crashes in the function IotMqtt_Connect.

Please find my code snippet as follows.

int InitializeMqtt()
{
    int status = EXIT_SUCCESS;
    IotMqttError_t mqttInitStatus = IOT_MQTT_STATUS_PENDING;

    // Custom declares.
    IotNetworkServerInfo_t xServerInfo = { 0 };
    IotNetworkCredentials_t xCredentials = AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER;
    IotNetworkInterface_t* pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;
    
    /* Set the server info. */
    xServerInfo.pHostName = clientcredentialMQTT_BROKER_ENDPOINT;
    xServerInfo.port = clientcredentialMQTT_BROKER_PORT;

    if (xServerInfo.port == 443)
    {
        xCredentials.pAlpnProtos = socketsAWS_IOT_ALPN_MQTT;
    }
    else
    {
        xCredentials.pAlpnProtos = NULL;
    }

    printf(" xServerInfo.pHostName = %s \r\n", xServerInfo.pHostName);
    printf(" xServerInfo.port = %d \r\n", xServerInfo.port);

    /* Set the server certificate for a secured connection. Other credentials
     * are set by the initializer. */
    xCredentials.pRootCa = rootCA_cert_pem;
    xCredentials.rootCaSize = rootCA_cert_pem_len;
    /* Disable SNI. */
    xCredentials.disableSni = true;
    /* ALPN is not needed. */
    xCredentials.pAlpnProtos = NULL;

    /* Strong mutual authentication to authenticate both the broker and
    * the client. */
    xCredentials.pClientCert = cert_pem;
    xCredentials.clientCertSize = cert_pem_len;
    xCredentials.pPrivateKey = privatekey_cert_pem;
    xCredentials.privateKeySize = privatekey_cert_pem_len;

    /* Set the members of the network info not set by the initializer. This
     * struct provided information on the transport layer to the MQTT connection. */
    networkInfo.createNetworkConnection = true;
    networkInfo.u.setup.pNetworkServerInfo = &xServerInfo;
    networkInfo.u.setup.pNetworkCredentialInfo = &xCredentials;
    networkInfo.pNetworkInterface = pNetworkInterface;
    networkInfo.disconnectCallback.function = prvNetworkDisconnectCallback;

#if ( IOT_MQTT_ENABLE_SERIALIZER_OVERRIDES == 1 ) && defined( IOT_DEMO_MQTT_SERIALIZER )
    networkInfo.pMqttSerializer = IOT_DEMO_MQTT_SERIALIZER;
#endif

    /* Set the members of the connection info not set by the initializer. */
    connectInfo.cleanSession = true;
    connectInfo.awsIotMqttMode = true;
    connectInfo.keepAliveSeconds = KEEPALIVE_SECONDS;
    connectInfo.clientIdentifierLength = (uint16_t)strlen(deviceID);
    connectInfo.pClientIdentifier = deviceID;

    /* Initialize the MQTT library.*/
    if (status == EXIT_SUCCESS)
    {
        mqttInitStatus = IotMqtt_Init();

        if (mqttInitStatus != IOT_MQTT_SUCCESS)
        {
            /* Failed to initialize MQTT library.*/
            status = EXIT_FAILURE;
        }
    }

    return status;    
}

int _establishMqttConnection(IotMqttConnection_t* pMqttConnection)
{
    IotMqttError_t connectStatus = IOT_MQTT_STATUS_PENDING;

    /* Establish the MQTT connection. */
        connectStatus = IotMqtt_Connect(&networkInfo,
            &connectInfo,
            otaCONN_TIMEOUT_MS,
            pMqttConnection);

        if (connectStatus != IOT_MQTT_SUCCESS)
        {
            IotLogError("MQTT CONNECT returned error %s.",
                IotMqtt_strerror(connectStatus));

            status = EXIT_FAILURE;
        }
    return status;
}

Please let me know as soon as possible.

Thanks, Regards,
Kumar.

kumarfirmware, 2021-07-12:
Hi Team,

Kindly help me in the above issue as soon as possible, because it is a crash issue.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-07-12:
From the code snippet above `xCredentials` and `xServerInfo` are declared as local variables on the function stack `InitializeMqtt()` and a pointer to their addresses is used in `_establishMqttConnection()` function  within `IotMqtt_Connect()`. If the `_establishMqttConnection` is not called from within `InitializeMqtt()` then these pointers are invalid as the stack space for `InitializeMqtt()` is already reclaimed.

Could you move the declarations of the variables to `_establishMqttConnection` as follows:

```
int _establishMqttConnection(IotMqttConnection_t* pMqttConnection)
{
IotMqttError_t connectStatus = IOT_MQTT_STATUS_PENDING;
// Custom declares.
IotNetworkServerInfo_t xServerInfo = { 0 };
IotNetworkCredentials_t xCredentials = AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER;
IotNetworkInterface_t* pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;

/* Set the server info. */
xServerInfo.pHostName = clientcredentialMQTT_BROKER_ENDPOINT;
xServerInfo.port = clientcredentialMQTT_BROKER_PORT;

if (xServerInfo.port == 443)
{
    xCredentials.pAlpnProtos = socketsAWS_IOT_ALPN_MQTT;
}
else
{
    xCredentials.pAlpnProtos = NULL;
}

printf(" xServerInfo.pHostName = %s \r\n", xServerInfo.pHostName);
printf(" xServerInfo.port = %d \r\n", xServerInfo.port);

/* Set the server certificate for a secured connection. Other credentials
 * are set by the initializer. */
xCredentials.pRootCa = ROOT_CERTIFICATE_PEM;;
xCredentials.rootCaSize = sizeof( ROOT_CERTIFICATE_PEM );
/* Disable SNI. */
xCredentials.disableSni = true;
/* ALPN is not needed. */
xCredentials.pAlpnProtos = NULL;

/* Strong mutual authentication to authenticate both the broker and
* the client. */
xCredentials.pClientCert = CLIENT_CERTIFICATE_PEM;
xCredentials.clientCertSize = sizeof( CLIENT_CERTIFICATE_PEM );;
xCredentials.pPrivateKey = CLIENT_PRIVATE_KEY_PEM;
xCredentials.privateKeySize = sizeof( CLIENT_PRIVATE_KEY_PEM );;

/* Set the members of the network info not set by the initializer. This
 * struct provided information on the transport layer to the MQTT connection. */
networkInfo.createNetworkConnection = true;
networkInfo.u.setup.pNetworkServerInfo = &xServerInfo;
networkInfo.u.setup.pNetworkCredentialInfo = &xCredentials;
networkInfo.pNetworkInterface = pNetworkInterface;
networkInfo.disconnectCallback.function = prvNetworkDisconnectCallback;
#if ( IOT_MQTT_ENABLE_SERIALIZER_OVERRIDES == 1 ) && defined( IOT_DEMO_MQTT_SERIALIZER )
networkInfo.pMqttSerializer = IOT_DEMO_MQTT_SERIALIZER;
#endif

/* Set the members of the connection info not set by the initializer. */
connectInfo.cleanSession = true;
connectInfo.awsIotMqttMode = true;
connectInfo.keepAliveSeconds = KEEPALIVE_SECONDS;
connectInfo.clientIdentifierLength = (uint16_t)strlen(deviceID);
connectInfo.pClientIdentifier = deviceID;

/* Establish the MQTT connection. */
    connectStatus = IotMqtt_Connect(&networkInfo,
        &connectInfo,
        otaCONN_TIMEOUT_MS,
        pMqttConnection);

    if (connectStatus != IOT_MQTT_SUCCESS)
    {
        IotLogError("MQTT CONNECT returned error %s.",
            IotMqtt_strerror(connectStatus));

        status = EXIT_FAILURE;
    }
return status;
}
```

kumarfirmware, 2021-07-14:
Hi Ravi,

I spent one day completely on this, it still crashes, exactly at the same place in the function IotMqtt_Connect.

Kindly help me in this as soon as possible, because it is a crash issue. This is really urgent.

Please find the code as follows

static int _establishMqttConnection(IotMqttConnection_t* pMqttConnection)
{
    int status = EXIT_SUCCESS;
    IotMqttError_t mqttInitStatus = IOT_MQTT_STATUS_PENDING;
    IotMqttError_t connectStatus = IOT_MQTT_STATUS_PENDING;

    // Custom declares.
    IotNetworkServerInfo_t xServerInfo = { 0 };
    IotNetworkCredentials_t xCredentials = AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER;
    IotNetworkInterface_t* pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;
    IotMqttConnectInfo_t connectInfo = IOT_MQTT_CONNECT_INFO_INITIALIZER;

    /* Set the server info. */
    xServerInfo.pHostName = clientcredentialMQTT_BROKER_ENDPOINT;
    xServerInfo.port = clientcredentialMQTT_BROKER_PORT;

    if (xServerInfo.port == 443)
    {
        xCredentials.pAlpnProtos = socketsAWS_IOT_ALPN_MQTT;
    }
    else
    {
        xCredentials.pAlpnProtos = NULL;
    }

    printf(" xServerInfo.pHostName = %s \r\n", xServerInfo.pHostName);
    printf(" xServerInfo.port = %d \r\n", xServerInfo.port);

    /* Set the server certificate for a secured connection. Other credentials
     * are set by the initializer. */
    xCredentials.pRootCa = ROOT_CERTIFICATE_PEM;
   xCredentials.rootCaSize = sizeof( ROOT_CERTIFICATE_PEM );
    /* Disable SNI. */
    xCredentials.disableSni = true;
    /* ALPN is not needed. */
    xCredentials.pAlpnProtos = NULL;

    /* Strong mutual authentication to authenticate both the broker and
    * the client. */
xCredentials.pClientCert = CLIENT_CERTIFICATE_PEM;
xCredentials.clientCertSize = sizeof( CLIENT_CERTIFICATE_PEM );
xCredentials.pPrivateKey = CLIENT_PRIVATE_KEY_PEM;
xCredentials.privateKeySize = sizeof( CLIENT_PRIVATE_KEY_PEM );
    /* Set the members of the network info not set by the initializer. This
     * struct provided information on the transport layer to the MQTT connection. */
    networkInfo.createNetworkConnection = true;
    networkInfo.u.setup.pNetworkServerInfo = &xServerInfo;
    networkInfo.u.setup.pNetworkCredentialInfo = &xCredentials;
    networkInfo.pNetworkInterface = pNetworkInterface;
    networkInfo.disconnectCallback.function = prvNetworkDisconnectCallback;

#if ( IOT_MQTT_ENABLE_SERIALIZER_OVERRIDES == 1 ) && defined( IOT_DEMO_MQTT_SERIALIZER )
    networkInfo.pMqttSerializer = IOT_DEMO_MQTT_SERIALIZER;
#endif

    /* Set the members of the connection info not set by the initializer. */
    connectInfo.cleanSession = true;
    connectInfo.awsIotMqttMode = true;
    connectInfo.keepAliveSeconds = otaKEEPALIVE_SECONDS;
    connectInfo.clientIdentifierLength = (uint16_t)strlen(deviceID);
    connectInfo.pClientIdentifier = deviceID;

    /* Initialize the MQTT library.*/
    if (status == EXIT_SUCCESS)
    {
        mqttInitStatus = IotMqtt_Init();

        if (mqttInitStatus != IOT_MQTT_SUCCESS)
        {
            /* Failed to initialize MQTT library.*/
            status = EXIT_FAILURE;
        }
    }

    /* Establish the MQTT connection. */
    if (status == EXIT_SUCCESS)
    {
        printf("Establish MqttConnection \r\n");

        connectStatus = IotMqtt_Connect(&networkInfo,
            &connectInfo,
            otaCONN_TIMEOUT_MS,
            pMqttConnection);

        if (connectStatus != IOT_MQTT_SUCCESS)
        {
            IotLogError("MQTT CONNECT returned error %s.",
                IotMqtt_strerror(connectStatus));

            status = EXIT_FAILURE;
        }
    }

    return status;
}

Please find the crash log as follows

Establish MqttConnection
4 295 [OTA_Update_wifi] [INFO ][MQTT][2950] MQTT library successfully initialized.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40090dda  PS      : 0x00060130  A0      : 0x8008fc51  A1      : 0x3ffe19e0
A2      : 0x3ffd14a4  A3      : 0x3ffdc948  A4      : 0x000007d0  A5      : 0x3ffc1f34
A6      : 0x3ffba500  A7      : 0x3ffbdc68  A8      : 0x00000000  A9      : 0x3ffd14ac
A10     : 0x00000000  A11     : 0x00000012  A12     : 0x00000000  A13     : 0x40148c14
A14     : 0x3ffdcf38  A15     : 0x3ffba51c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: 9d2ce420ead74a77

Backtrace: 0x40090dda:0x3ffe19e0 0x4008fc4e:0x3ffe1a00 0x4008f523:0x3ffe1a20 0x4014f59f:0x3ffe1a60 0x4014f5b9:0x3ffe1a80 0x4014ef6d:0x3ffe1aa0 0x4014efbd:0x3ffe1ac0 0x40148689:0x3ffe1af0 0x401470aa:0x3ffe1b20 0x400d2e7b:0x3ffe1b50 0x400d2ed9:0x3ffe1bd0

Thanks, Regards,
Kumar.

kumarfirmware, 2021-07-14:
Hi Ravi,

One clue, if I turn off wifi I mean if I turn off access point, program does not crash. The following is the log.

Establish MqttConnection
1058 2176 [OTA_Update_wifi] [INFO ][MQTT][21760] MQTT library successfully initialized.
1059 2176 [OTA_Update_wifi] [ERROR][NET][21760] Failed to resolve a17ekevi1yo7w4-ats.iot.us-east-1.amazonaws.com.
1060 2176 [OTA_Update_wifi] [ERROR][MQTT][21760] Failed to establish new MQTT connection, error NETWORK ERROR.
1061 2176 [OTA_Update_wifi] [ERROR][DEMO][21760] MQTT CONNECT returned error NETWORK ERROR.
1062 2176 [OTA_Update_wifi] [ERROR][DEMO][21760] ERROR:  MQTT_AGENT_Connect() Failed.

This means the culprit is this guy.

  IotNetworkInterface_t* pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;
  networkInfo.pNetworkInterface = pNetworkInterface;

I guess this information should be more than sufficient for you to solve the problem immediately.

Please let me know how to proceed.

Thanks, Regards,
Kumar.

ravibhagavandas, 2021-07-14:
Hi @kumarfirmware 

There are other steps after a successful TCP connection is made - creating a TLS handshake with the broker,  sending an MQTT CONNECT message and receiving back a CONNACK as acknowledgement, before which the function returns success.
It can fail at any of these steps. Could you run this through the debugger and check at which step tis crashing by stepping in through it ? More details on debugging on ESP32 can be found [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_debugging).
Also few other things you could try:

1. Enable debug logs for MQTT API and Network Interface by editing the [iot_config.h](https://github.com/aws/amazon-freertos/blob/202007.00/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L49) file as follows:
```
#define IOT_LOG_LEVEL_NETWORK                   IOT_LOG_DEBUG
#define IOT_LOG_LEVEL_MQTT                      IOT_LOG_DEBUG
```
2. Create a TLS connection prior to calling `IotMqtt_Connect()` to check if its succeeding TLS handshake. You can create a TLS connection  as follows:
```
void * pNetworkConnection = NULL;
IotNetworkError_t xNetworkStatus;
xNetworkStatus = pNetworkInterface->create( &xServerInfo, &xCredentials, &pNetworkConnection);
configASSERT(xNetworkStatus == IOT_NETWORK_SUCCESS);

//Pass the connection directly to MQTT library.
networkInfo.createNetworkConnection = false;
networkInfo.u.pNetworkConnection = pNetworkConnection;
networkInfo.pNetworkInterface = pNetworkInterface;
networkInfo.disconnectCallback.function = prvNetworkDisconnectCallback;
```
3. Did you make sure network info structure is initialized properly as:

```
IotMqttNetworkInfo_t networkInfo = IOT_MQTT_NETWORK_INFO_INITIALIZER;
```

4. Did you check if the stack size for the task is adequate for TLS connection ?

kumarfirmware, 2021-07-24:
Hi Ravi,

This is fixed. Please close the issue.

closed 2021-07-24 by kumarfirmware