aws/amazon-freertos #3323: [Feature Request] Update aws demos to be compatible with ESP-IDF's newest toolchains

GKFernandes, 2021-07-14:
Hello everyone,

**Is your feature request related to an issue you've encountered? If so, please elaborate.**
I'm having some trouble compiling the aws demos (in this specific case, the AWS OTA demo) using the most recent toolchains ( I have tested esp-2020r3-8.4.0 and esp-2021r1).

When I follow the AWS tutorials and go to vendors/espressif/esp-idf and run export.bat, I can see that an older toolchain is added to PATH, like the screenshot below;

[image]

So the added toolchain is for ESP-IDF v3.3 - 1.22.0-80-g6c4433a5-5.2.0.

The demos run just fine when I try to use the `bundled` toolchain that comes when I clone the repo, problem starts happening when I try to use the latest toolchain, it seems to not be working so far.

Whenever I run the cmake command to generate build files for the AWS OTA demo with the latest toolchain, which by the way is:

**cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B build -GNinja**

I get the following:

C:\Users\IOT\Desktop\ota\amazon-freertos>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B build -GNinja
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/IOT/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Users/IOT/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- Submodule update
Skipping submodule '../../../libraries/abstractions/backoff_algorithm/'
Skipping submodule '../../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT-Agent/source/dependency/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT-Agent/source/dependency/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT-Agent/source/dependency/coreMQTT/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/ota_for_aws/'
Skipping submodule '../../../libraries/ota_for_aws/source/dependency/coreJSON/'
Skipping submodule '../../../libraries/ota_for_aws/source/dependency/coreJSON/'
Skipping submodule '../../../libraries/ota_for_aws/source/dependency/coreJSON/'
WARNING: IDF_PATH environment variable is not cleared.
    If CMake is generating an error, consider clearing the IDF_PATH environment
    variable, and generating a clean build. This message can be ignored if
    CMake was successful.
CMake Error at vendors/espressif/boards/esp32/CMakeLists.txt:38 (include):
  include could not find load file:

    C:/Users/IOT/Desktop/ota/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/idf.cmake
Call Stack (most recent call first):
  CMakeLists.txt:70 (include)

CMake Error at vendors/espressif/boards/esp32/CMakeLists.txt:39 (include):
  include could not find load file:

    C:/Users/IOT/Desktop/ota/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/project_utils.cmake
Call Stack (most recent call first):
  CMakeLists.txt:70 (include)

CMake Error at vendors/espressif/boards/esp32/CMakeLists.txt:582 (idf_build_component):
  Unknown CMake command "idf_build_component".
Call Stack (most recent call first):
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "C:/Users/IOT/Desktop/ota/amazon-freertos/build/CMakeFiles/CMakeOutput.log".

So, the new feature would be to make the aws demos compatible with the newest toolchains of ESP-IDF.

shubhamkulkarni97, 2021-07-15:
Hi @GKFernandes,

Version of ESP32 toolchain you use is based on ESP-IDF version. If you are using ESP-IDF v3.3, `xtensa-esp32-elf-gcc v5.2.0` is the only supported toolchain.

If you want to use `esp-2020r3-8.4.0` , you need to use newer version of ESP-IDF. In this case, you can use master branch of Amazon FreeRTOS which supports ESP-IDF v4.2.

You can run install.bat and export.bat scripts to install and select required toolchain.

GKFernandes, 2021-07-15:
Hello @shubhamkulkarni97 ,

Thank you for you reply, I have ellaborated better on the tests yesterday, let me update the information:

I have indeed tried to use both ESP-IDF versions, v3.3 and v4.2, if you look at the amazon-freertos repo in the 202012.00-branch you'll see that when running export.bat inside ../vendors/espressif/esp-idf the v3.3 toolchain is added to PATH.

I have then cloned the same repo (git clone -b main https://github.com/aws/amazon-freertos.git) and I can see that the toolchain was not updated, it still uses the same toolchain as it did for ESP-IDF v3.3.

There was even this note on AWS's getting started guide:

[image]

After trying to compile the demos using a newer toolchain downloaded from Espressif's github, the same error has shown up.

It seems there is a brand new release that just came out, I will try to compile and post my results here.

Thank you,

muneebahmed10, 2021-07-16:
Hi @GKFernandes,

> I have just cloned the new release that came out 202107 and it works like a charm! When I execute the export.bat, I can see that in fact the newest ESP-IDF toolchain is being added to PATH.

Given that you have reached a resolution, I will close this issue. Please open a new issue if you have any further questions.

closed 2021-07-16 by muneebahmed10