aws/amazon-freertos #3330: [General] releases don't include submodule code

koooge, 2021-07-15:
**Briefly summarize the issue being raised**
Please describe the topic requiring attention in this issue, as well some background on its current status.

Hi there,
releases do not include submodules' code like freertos_kernel and libraries/. Is this intended?
https://github.com/aws/amazon-freertos/releases/tag/202012.00

**Describe the desired outcome**
I hope releases include FreeRTOS code. It would make amazon-freertos easier to use

**System information**
- Hardware board: [ ... ]
- IDE used: [ ... ]
- Operating System: [ Windows | Linux | MacOS ]
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [ mqtt_demo_mutual_auth | shadow_demo | custom application | etc... ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code/Steps to reproduce the current status"`.

**References**
Provide any screenshots, logs, etc. that may assist in providing a better understanding of the highlighted issue.

**Code/Steps to reproduce the current status**
If applicable, provide concrete instructions or the relevant code snippet(s) for reproducing the current status. Code should be wrapped in the ``cpp`` tag in order to be displayed clearly. For example:
```cpp
printf("Hello World")
```
**Additional context**
Add any remaining information not mentioned in the previous responses.

muneebahmed10, 2021-07-15:
Hi @koooge,

I just downloaded the `freertos-aws-reference-integrations.tar.gz` archive and it contained the submodule contents, including the FreeRTOS kernel and libraries. Please make sure that you are downloading the uploaded `freertos-aws-reference-integrations.tar.gz` asset and not the source code archives auto generated by GitHub.

koooge, 2021-07-15:
Aha, I had downloaded zip. Thanks!

closed 2021-07-15 by koooge