aws/amazon-freertos #3333: Could not subscribe coreMQTT demo

Git-shouvik, 2021-07-19:
**Describe the bug**
Trying out demos/coreMQTT , I have generated the build files, build the project and flashed and then when i use monitor using idf.py monitor. 
I get this error: 
6 59 [iot_thread] [ERROR][DEMO][590] Failed to initialize all the networks configured for the device.

7 59 [iot_thread] [INFO ][INIT][590] SDK cleanup done.

8 60 [iot_thread] [ERROR][DEMO][600] Failed to initialize the demo. exiting...

9 60 [iot_thread] [INFO ][DEMO][600] -------DEMO FINISHED-------

**System information**
- Hardware board: [ ESP-32 ]
- IDE used: [ ubuntu terminal]
- Operating System: [Linux ]
- Code version: (202107.00-3-g19c657cf0)
- Project/Demo: [ demos/coreMQTT]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
while I subscribe from AWS IoT console, it should print the output after i run idf.py monitor
[image]

muneebahmed10, 2021-07-20:
Hi @Git-shouvik,

From your logs, I see the message
```
Warning: the client certificate should be updated. Please see https://aws.amazon.com/freertos/getting-started/.
```

In addition to the above URL, I would recommend reviewing https://docs.aws.amazon.com/iot/latest/developerguide/create-iot-resources.html and ensuring you have set up your certificates and policies correctly

yourslab, 2021-07-25:
Hi @Git-shouvik, were you able to resolve the problem after reviewing your certificates and policies?

Git-shouvik, 2021-07-27:
ya, the issue was with the policies.
I resolved it by deleting and recreating with python SetupAWS.py setup.
Thanks.

closed 2021-07-27 by Git-shouvik