aws/amazon-freertos #3334: As I mentioned before, you need to update `SHADOW_DESIRED_JSON` and `SHADOW_REPORTED_JSON` JSON documents sent from the device:

SGOSAI602, 2021-07-22:
As I mentioned before, you need to update `SHADOW_DESIRED_JSON` and `SHADOW_REPORTED_JSON` JSON documents sent from the device:
* https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L100
* https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L125

Here are the relevant pieces from our sample to show you how to update the `SHADOW_DESIRED_JSON`:

```c
#define SHADOW_DESIRED_JSON     \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"rssi\":%03d"               \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"

#define EXPECTED_DESIRED_JSON_SIZE    ( sizeof( SHADOW_DESIRED_JSON ) - 1 )

static char pUpdateDocument[ EXPECTED_DESIRED_JSON_SIZE + 1 ] = { 0 };

status = snprintf( pUpdateDocument,
                   EXPECTED_DESIRED_JSON_SIZE + 1,
                   SHADOW_DESIRED_JSON,
                   ( int ) rssi,
                   ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );
```

Thanks.

_Originally posted by @aggarg in https://github.com/aws/amazon-freertos/issues/2471#issuecomment-707222742_

SGOSAI602, 2021-07-22:
Hello,
 how to update the SHADOW_DESIRED_JSON: if my sensor data value is 1234 what and where the changes in code.
we changed in below function but geeting shadow rejected error.

`#define SHADOW_DESIRED_JSON     \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"rssi\":%04d"               \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"
`
I want shadow update like below.

`{
 "desired": {
   "wifi": {
     "rssi near": 1234
   }
 }
}`

thanks

aggarg, 2021-07-22:
I am not sure I completely understand your question completely - are you asking how to write a snprintf to generate a JSON document or anything else? 

Can you also dump the generated JSON document (i.e. the content of pUpdateDocument buffer after the snprintf call) that is getting rejected?

yourslab, 2021-07-22:
@SGOSAI602 Are you sure about using `%04d`? What if RSSI is 4? You will get
```{
 "desired": {
   "wifi": {
     "rssi ": 0004
   }
 }
}```
0004 is not a valid JSON value, and that is probably why the shadow update is being rejected.

SGOSAI602, 2021-07-23:
> I am not sure I understand your question completely - are you asking how to write a snprintf to generate a JSON document or anything else?
> 
> Can you also dump the generated JSON document (i.e. the content of pUpdateDocument buffer after the snprintf call) that is getting rejected?

Hi  aggarg ,

We have a sensor, we have sent its data to the cloud with the help of I2C,
I am getting the value of my sensor in 32 bytes, like "3422" so how to send 32 bytes of data to AWS Shadow?
If we send 16 bytes data to cloud like "55" it works well and data is updated on AWS shadow but when send 32 bytes data it is not working, my question is this How to send 32 bytes of data to AWS Shadow?

 ```
`#define SHADOW_DESIRED_JSON     \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"rssi\":%03d"               \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"`

```

SGOSAI602, 2021-07-23:
> @SGOSAI602 Are you sure about using `%04d`? What if RSSI is 4? You will get
> 
> ```
> {
>  "desired": {
>    "wifi": {
>      "rssi ": 0004
>    }
>  }
> }
> ```
> 
> 0004 is not a valid JSON value, and that is probably why the shadow update is being rejected.

Hi yourslab,
 We are not sure about %04. We have a sensor, and its out value is constantly changing, like 16 bytes or 32 bytes.  
If we send 16 byte data using '%03' its work well.

yourslab, 2021-07-23:
Do you mean 32 bits for an int? If that's the case, you know that the maximum value of a signed int is 2147483647, which is well beyond 4 digits. If you wish to just get the first 4 digits, use modulo operator: `rssi % 10000`

You should probably also just use `%d` instead of `%04d` so that you don't have extra leading zeros.

aggarg, 2021-07-23:
Please answer the following questions to help us help you better:

* What are the maximum and minimum values your sensor can return?
* Please add a log statement to dump the **generated JSON document** when it is rejected and share that. 

Thanks.

SGOSAI602, 2021-07-23:
Sorry it's my mistake ,yes you are right it's 32 bits .

SGOSAI602, 2021-07-26:
> Please answer the following questions to help us help you better:
> 
> * What are the maximum and minimum values your sensor can return?
> * Please add a log statement to dump the **generated JSON document** when it is rejected and share that.
> 
> Thanks.
maximum and minimum values your sensor can return?

sensor returns minimum value is '0' and maximum value is '9999' .
below attached aws_IoT_demo_shadow.txt file.

[aws_IoT_demo_shadow.txt](https://github.com/aws/amazon-freertos/files/6876093/aws_IoT_demo_shadow.txt)

```
`2 12 [iot_thread] [INFO ][DEMO][120] ---------STARTING DEMO---------

I (460) BTDM_INIT: BT controller compile version [0fff744]
3 12 [iot_thread] [INFO ][INIT][120] SDK successfully initialized.
I (470) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (570) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
I (880) wifi:wifi driver task: 3ffda1a0, prio:23, stack:3584, core=0
I (880) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (880) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (900) wifi:wifi firmware version: 1afdae0
I (900) wifi:config NVS flash: enabled
I (900) wifi:config nano formating: disabled
I (900) wifi:Init dynamic tx buffer num: 32
I (910) wifi:Init data frame dynamic rx buffer num: 32
I (910) wifi:Init management frame dynamic rx buffer num: 32
I (920) wifi:Init management short buffer num: 32
I (920) wifi:Init static rx buffer size: 1600
I (920) wifi:Init static rx buffer num: 10
I (930) wifi:Init dynamic rx buffer num: 32
I (940) wifi:mode : sta (30:ae:a4:9c:fc:1c)
I (940) WIFI: SYSTEM_EVENT_STA_START
I (3600) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (4580) wifi:state: init -> auth (b0)
I (4590) wifi:state: auth -> assoc (0)
I (5590) wifi:state: assoc -> init (400)
I (5590) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (5590) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 4
I (6840) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (6840) wifi:state: init -> auth (b0)
I (6840) wifi:state: auth -> assoc (0)
I (6850) wifi:state: assoc -> run (10)
I (6870) wifi:connected with oppo, aid = 1, channel 11, BW20, bssid = 86:da:da:35:f7:bc
I (6870) wifi:security type: 3, phy: bgn, rssi: -19
I (6870) wifi:pm start, type: 1

I (6870) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (6960) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (7430) event: sta ip: 192.168.43.62, mask: 255.255.255.0, gw: 192.168.43.1
I (7430) WIFI: SYSTEM_EVENT_STA_GOT_IP
4 709 [iot_thread] [INFO ][DEMO][7090] Successfully initialized the demo. Network type for the demo: 1
5 709 [iot_thread] [INFO ][MQTT][7090] MQTT library successfully initialized.
6 709 [iot_thread] [INFO ][Shadow][7090] Shadow library successfully initialized.
7 709 [iot_thread] [INFO ][DEMO][7090] Shadow Thing Name is aumraj_aws602 (length 13).
E (8940) PKCS11: failed nvs get file size 4354 0
8 1328 [iot_thread] [INFO ][MQTT][13280] Establishing new MQTT connection.
9 1328 [iot_thread] [INFO ][MQTT][13280] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABL.
10 1328 [iot_thread] [INFO ][MQTT][13280] (MQTT connection 0x3ffdf67c, CONNECT operation 0x3ffdfa9c) Waiting for operation completion.
11 1359 [iot_thread] [INFO ][MQTT][13590] (MQTT connection 0x3ffdf67c, CONNECT operation 0x3ffdfa9c) Wait complete with result SUCCESS.
12 1359 [iot_thread] [INFO ][MQTT][13590] New MQTT connection 0x3ffb3e20 established.
13 1359 [iot_thread] [INFO ][Shadow][13590] (aumraj_aws602) Modifying Shadow DELTA callback.
14 1359 [iot_thread] [INFO ][Shadow][13590] (aumraj_aws602) Adding new DELTA callback.
15 1359 [iot_thread] [INFO ][MQTT][13590] (MQTT connection 0x3ffdf67c) SUBSCRIBE operation scheduled.
16 1359 [iot_thread] [INFO ][MQTT][13590] (MQTT connection 0x3ffdf67c, SUBSCRIBE operation 0x3ffe86cc) Waiting for operation completion.
17 1389 [iot_thread] [INFO ][MQTT][13890] (MQTT connection 0x3ffdf67c, SUBSCRIBE operation 0x3ffe86cc) Wait complete with result SUCCESS.
18 1389 [iot_thread] [INFO ][Shadow][13890] (aumraj_aws602) Shadow DELTA callback operation complete with result SUCCESS.
[... 27 lines cut ...]
46 1626 [iot_thread] [INFO ][MQTT][16260] (MQTT connection 0x3ffdf67c) Connection disconnected.
47 1628 [iot_thread] [INFO ][MQTT][16280] (MQTT connection 0x3ffdf67c) Network connection closed.
I (16670) wifi:state: run -> init (0)
I (16670) wifi:pm stop, total sleep time: 5983278 us / 9802361 us

I (16670) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (16680) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (16690) wifi:wifi deinit internal: wifi not stop
48 1633 [iot_thread] [INFO ][MQTT][16330] (MQTT connection 0x3ffdf67c) Network connection destroyed.
I (16700) wifi:flush txq
I (16700) wifi:stop sw txq
I (16700) wifi:lmac stop hw txq
49 1633 [iot_thread] [INFO ][Shadow][16330] Shadow library cleanup done.
50 1633 [iot_thread] [INFO ][MQTT][16330] MQTT library cleanup done.
I (16720) wifi:Deinit lldesc rx mblock:10
51 1633 [iot_thread] [INFO ][DEMO][16330] memory_metrics::freertos_heap::before::bytes::198288
52 1633 [iot_thread] [INFO ][DEMO][16330] memory_metrics::freertos_heap::after::bytes::81628
53 1633 [iot_thread] [INFO ][DEMO][16330] memory_metrics::demo_task_stack::before::bytes::5848
54 1633 [iot_thread] [INFO ][DEMO][16330] memory_metrics::demo_task_stack::after::bytes::1896
55 1633 [iot_thread] [ERROR][DEMO][16330] Error running demo.
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
56 1645 [iot_thread] [INFO ][INIT][16450] SDK cleanup done.
57 1645 [iot_thread] [INFO ][DEMO][16450] -------DEMO FINISHED-------`
```

SGOSAI602, 2021-07-26:
> Do you mean 32 bits for an int? If that's the case, you know that the maximum value of a signed int is 2147483647, which is well beyond 4 digits. If you wish to just get the first 4 digits, use [...]

we tried to used '%d' instead of `%04d` but getting shadow rejected and Error running demo.

yourslab, 2021-07-26:
@SGOSAI602 Please also log the JSON document so that we can see if it is formatted correctly

SGOSAI602, 2021-07-26:
> @SGOSAI602 Please also log the JSON document so that we can see if it is formatted correctly

when we tried to used '%d' instead of %04d but getting bellow error.
```

2 12 [iot_thread] [INFO ][DEMO][120] ---------STARTING DEMO---------

I (460) BTDM_INIT: BT controller compile version [0fff744]
3 12 [iot_thread] [INFO ][INIT][120] SDK successfully initialized.
I (470) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (570) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
I (880) wifi:wifi driver task: 3ffda1a0, prio:23, stack:3584, core=0
I (880) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (880) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (900) wifi:wifi firmware version: 1afdae0
I (900) wifi:config NVS flash: enabled
I (900) wifi:config nano formating: disabled
I (900) wifi:Init dynamic tx buffer num: 32
I (900) wifi:Init data frame dynamic rx buffer num: 32
I (910) wifi:Init management frame dynamic rx buffer num: 32
I (920) wifi:Init management short buffer num: 32
I (920) wifi:Init static rx buffer size: 1600
I (920) wifi:Init static rx buffer num: 10
I (930) wifi:Init dynamic rx buffer num: 32
I (940) wifi:mode : sta (30:ae:a4:9c:fc:1c)
I (940) WIFI: SYSTEM_EVENT_STA_START
I (4080) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 201
I (8210) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 201
I (11650) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (12640) wifi:state: init -> auth (b0)
I (12650) wifi:state: auth -> assoc (0)
I (12670) wifi:state: assoc -> run (10)
I (12690) wifi:connected with oppo, aid = 1, channel 6, BW20, bssid = 86:da:da:35:f7:bc
I (12690) wifi:security type: 3, phy: bgn, rssi: -35
I (12690) wifi:pm start, type: 1

I (12700) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (12760) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (13390) event: sta ip: 192.168.43.62, mask: 255.255.255.0, gw: 192.168.43.1
I (13390) WIFI: SYSTEM_EVENT_STA_GOT_IP
4 1305 [iot_thread] [INFO ][DEMO][13050] Successfully initialized the demo. Network type for the demo: 1
5 1305 [iot_thread] [INFO ][MQTT][13050] MQTT library successfully initialized.
6 1305 [iot_thread] [INFO ][Shadow][13050] Shadow library successfully initialized.
7 1305 [iot_thread] [INFO ][DEMO][13050] Shadow Thing Name is aumraj_aws602 (length 13).
E (14020) PKCS11: failed nvs get file size 4354 0
8 1765 [iot_thread] [INFO ][MQTT][17650] Establishing new MQTT connection.
9 1765 [iot_thread] [INFO ][MQTT][17650] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABL.
10 1765 [iot_thread] [INFO ][MQTT][17650] (MQTT connection 0x3ffdf8f4, CONNECT operation 0x3ffdf690) Waiting for operation completion.
11 1806 [iot_thread] [INFO ][MQTT][18060] (MQTT connection 0x3ffdf8f4, CONNECT operation 0x3ffdf690) Wait complete with result SUCCESS.
12 1806 [iot_thread] [INFO ][MQTT][18060] New MQTT connection 0x3ffb3e20 established.
13 1806 [iot_thread] [INFO ][Shadow][18060] (aumraj_aws602) Modifying Shadow DELTA callback.
14 1806 [iot_thread] [INFO ][Shadow][18060] (aumraj_aws602) Adding new DELTA callback.
15 1806 [iot_thread] [INFO ][MQTT][18060] (MQTT connection 0x3ffdf8f4) SUBSCRIBE operation scheduled.
16 1806 [iot_thread] [INFO ][MQTT][18060] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffea8e0) Waiting for operation completion.
17 1837 [iot_thread] [INFO ][MQTT][18370] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffea8e0) Wait complete with result SUCCESS.
18 1837 [iot_thread] [INFO ][Shadow][18370] (aumraj_aws602) Shadow DELTA callback operation complete with result SUCCESS.
19 1837 [iot_thread] [INFO ][Shadow][18370] (aumraj_aws602) Modifying Shadow UPDATED callback.
20 1837 [iot_thread] [INFO ][Shadow][18370] (aumraj_aws602) Adding new UPDATED callback.
21 1837 [iot_thread] [INFO ][MQTT][18370] (MQTT connection 0x3ffdf8f4) SUBSCRIBE operation scheduled.
22 1837 [iot_thread] [INFO ][MQTT][18370] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffea8e0) Waiting for operation completion.
23 1868 [iot_thread] [INFO ][MQTT][18680] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffea8e0) Wait complete with result SUCCESS.
24 1868 [iot_thread] [INFO ][Shadow][18680] (aumraj_aws602) Shadow UPDATED callback operation complete with result SUCCESS.
25 1868 [iot_thread] [INFO ][MQTT][18680] (MQTT connection 0x3ffdf8f4) SUBSCRIBE operation scheduled.
26 1868 [iot_thread] [INFO ][MQTT][18680] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffea8e0) Waiting for operation completion.
27 1898 [iot_thread] [INFO ][MQTT][18980] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffea8e0) Wait complete with result SUCCESS.
28 1898 [iot_thread] [INFO ][MQTT][18980] (MQTT connection 0x3ffdf8f4) SUBSCRIBE operation scheduled.
29 1898 [iot_thread] [INFO ][MQTT][18980] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffeaab8) Waiting for operation completion.
30 1929 [iot_thread] [INFO ][MQTT][19290] (MQTT connection 0x3ffdf8f4, SUBSCRIBE operation 0x3ffeaab8) Wait complete with result SUCCESS.
31 1929 [iot_thread] [INFO ][MQTT][19290] (MQTT connection 0x3ffdf8f4) MQTT PUBLISH operation queued.
[... 13 lines cut ...]
[... cut ...]
51 2060 [iot_thread] [INFO ][DEMO][20600] memory_metrics::freertos_heap::after::bytes::81620
52 2060 [iot_thread] [INFO ][DEMO][20600] memory_metrics::demo_task_stack::before::bytes::5848
53 2060 [iot_thread] [INFO ][DEMO][20600] memory_metrics::demo_task_stack::after::bytes::1944
54 2060 [iot_thread] [ERROR][DEMO][20600] Error running demo.
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
55 2072 [iot_thread] [INFO ][INIT][20720] SDK cleanup done.
56 2072 [iot_thread] [INFO ][DEMO][20720] -------DEMO FINISHED-------
```

aggarg, 2021-07-26:
As @yourslab mentioned, you need to add a log to dump the generated JSON document. Is it possible for you to share that?

SGOSAI602, 2021-07-27:
> As @yourslab mentioned, you need to add a log to dump the generated JSON document. Is it possible for you to share that?

yes it is possible, but we don't understand which log you need, you need project file? Or the creation of the build file which after complications?

yourslab, 2021-07-27:
In other words, please update the demo so that you print the JSON document that is used to update the Shadow when it fails.

SGOSAI602, 2021-07-27:
we shared my demo file please check .

[shadow.zip](https://github.com/aws/amazon-freertos/files/6883684/shadow.zip)

Raghav3107, 2021-07-27:
Hello @SGOSAI602 
From your shadow file, I don't understand why you are trying to send the two separate desired shadows. 
```
#define SHADOW_DESIRED_ADC_JSON \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"sensor_data\":%2d"       \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"

#define SHADOW_DESIRED_I2C_JSON \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"light_data\":%03d"       \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"
```
Instead, you can do 
```
#define SHADOW_DESIRED_ADC_JSON \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"sensor_data\":%2d,"      \
    "\"light_data\":%03d"       \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"
```
example to add other parameters ->
```
#define SHADOW_DESIRED_I2C_JSON   \
    "{"                           \
    "\"state\":{"                 \
    "\"desired\":{"               \
    "\"string Variable\":\"%s\"," \
    "\"int Variable\":%d,"        \
    "\"string Variable\":\"%s\"," \
    "\"int64_t Variable\":%lld"   \
    "}"                           \
    "},"                          \
    "\"clientToken\":\"%06lu\""   \
    "}"
```
And I saw in your code, you are sending the `int` data. You don't need to add %04 or %02 unless want to send only 4 or 2 digit data.

> 41 2052 [iot_thread] [ERROR][DEMO][20520] Failed to generate desired state document for Shadow update 1 of 20.
 This error is coming from 
```
 if ((size_t)status != EXPECTED_DESIRED_ADC_JSON_SIZE)
            {
                IotLogError("Failed to generate desired state document for Shadow update"
                            " %d of %d.",
                            i, AWS_IOT_DEMO_SHADOW_UPDATE_COUNT);

                status = EXIT_FAILURE;
                break;
            }
```
So, Try to increase the` EXPECTED_DESIRED_ADC_JSON_SIZE` value to 500. The size will not be that much. But for testing, increase it to 500.

Hope this will help you!

aggarg, 2021-07-27:
The error you are getting is from the following code block:

```c
status = snprintf( pUpdateDocument,
                           EXPECTED_DESIRED_I2C_JSON_SIZE + 1,
                           SHADOW_DESIRED_I2C_JSON,
                           ( int ) bh1750,
                           ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );

        /* Check for errors from snprintf. The expected value is the length of
         * the desired JSON document less the format specifier for the state. */
        if( ( size_t ) status != EXPECTED_DESIRED_I2C_JSON_SIZE )
        {
            IotLogError( "Failed to generate desired state document for Shadow update"
                         " %d of %d.", i, AWS_IOT_DEMO_SHADOW_UPDATE_COUNT );

            status = EXIT_FAILURE;
            break;
        }
```
Reading this code tells you that the error log means that `snprintf` did not return the expected value. 

I am showing you the changes you need to make for ADC json and I hope you will be able to make similar changes for I2C json. 

```c
#define SHADOW_DESIRED_ADC_JSON \
    "{"                         \
    "\"state\":{"               \
    "\"desired\":{"             \
    "\"sensor_data\":%d"        \
    "}"                         \
    "},"                        \
    "\"clientToken\":\"%06lu\"" \
    "}"

#define EXPECTED_DESIRED_ADC_JSON_SIZE    ( 500 )

/* In function _sendShadowUpdates. */

    static char pUpdateDocument[ EXPECTED_DESIRED_ADC_JSON_SIZE ] = { 0 };

   /* Set the common members of the Shadow update document info. */
    updateDocument.pThingName = pThingName;
    updateDocument.thingNameLength = thingNameLength;
    updateDocument.u.update.pUpdateDocument = pUpdateDocument;

  .
  .
  .

         status = snprintf( pUpdateDocument,
                           EXPECTED_DESIRED_ADC_JSON_SIZE,
                           SHADOW_DESIRED_ADC_JSON,
                           ( int ) TempratureLM35,
                           ( long unsigned ) ( IotClock_GetTimeMs() % 1000000 ) );

        if( ( status > 0 ) && ( status < EXPECTED_DESIRED_ADC_JSON_SIZE ) )
        {
            updateDocument.u.update.updateDocumentLength = status;
            status = EXIT_SUCCESS;
        }
        else
        {
            IotLogError( "Failed to generate desired state document for Shadow update"
                         " %d of %d.", i, AWS_IOT_DEMO_SHADOW_UPDATE_COUNT );

            status = EXIT_FAILURE;
            break;
        }

```

Thanks.

SGOSAI602, 2021-07-29:
thank you Gaurav-Aggarwal, for support and valuable reply.

closed 2021-07-29 by SGOSAI602