aws/amazon-freertos #3336: [BUG] fail to build code downloaded from AWS console

knap1930, 2021-07-23:
**Describe the bug**
Code downloaded from AWS console fails to build, when I clone amazon-freertos repo, it works fine.

**System information**
- Hardware board: ESP32-DevKitC
- IDE used: none
- Operating System: MacOS
- Code version: 202107.00
- Project/Demo: OTA over HTTP Demo

**Expected behavior**
success

**Screenshots or console output**
```
Executing action: all (aliases: build)
Running ninja in directory /Users/derek/esp/FreeRTOS/build
Executing "ninja all"...
[0/2] Re-checking globbed directories...
[1/2] Re-running CMake...
WARNING: IDF_PATH environment variable is not cleared.
    If CMake is generating an error, consider clearing the IDF_PATH environment
    variable, and generating a clean build. This message can be ignored if
    CMake was successful.
-- Component directory /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/unity does not contain a CMakeLists.txt file. No component will be added
fatal: not a git repository (or any of the parent directories): .git
-- Checking Python dependencies...
Python requirements from /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/requirements.txt are satisfied.
-- Project sdkconfig file /Users/derek/esp/FreeRTOS/build/sdkconfig
Loading defaults file /Users/derek/esp/FreeRTOS/build/sdkconfig.defaults...
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:21 CONFIG_NIMBLE_ENABLED was replaced with CONFIG_BT_NIMBLE_ENABLED
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:22 CONFIG_NIMBLE_MAX_CONNECTIONS was replaced with CONFIG_BT_NIMBLE_MAX_CONNECTIONS
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:23 CONFIG_NIMBLE_MAX_BONDS was replaced with CONFIG_BT_NIMBLE_MAX_BONDS
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:24 CONFIG_NIMBLE_MAX_CCCDS was replaced with CONFIG_BT_NIMBLE_MAX_CCCDS
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:25 CONFIG_NIMBLE_L2CAP_COC_MAX_NUM was replaced with CONFIG_BT_NIMBLE_L2CAP_COC_MAX_NUM
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:26 CONFIG_NIMBLE_PINNED_TO_CORE was replaced with CONFIG_BT_NIMBLE_PINNED_TO_CORE
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:27 CONFIG_NIMBLE_ROLE_CENTRAL was replaced with CONFIG_BT_NIMBLE_ROLE_CENTRAL
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:28 CONFIG_NIMBLE_ROLE_PERIPHERAL was replaced with CONFIG_BT_NIMBLE_ROLE_PERIPHERAL
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:29 CONFIG_NIMBLE_ROLE_BROADCASTER was replaced with CONFIG_BT_NIMBLE_ROLE_BROADCASTER
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:30 CONFIG_NIMBLE_ROLE_OBSERVER was replaced with CONFIG_BT_NIMBLE_ROLE_OBSERVER
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:31 CONFIG_NIMBLE_NVS_PERSIST was replaced with CONFIG_BT_NIMBLE_NVS_PERSIST
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:32 CONFIG_NIMBLE_SM_LEGACY was replaced with CONFIG_BT_NIMBLE_SM_LEGACY
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:33 CONFIG_NIMBLE_SM_SC was replaced with CONFIG_BT_NIMBLE_SM_SC
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:34 CONFIG_NIMBLE_DEBUG was replaced with CONFIG_BT_NIMBLE_DEBUG
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:35 CONFIG_NIMBLE_SVC_GAP_DEVICE_NAME was replaced with CONFIG_BT_NIMBLE_SVC_GAP_DEVICE_NAME
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:36 CONFIG_NIMBLE_GAP_DEVICE_NAME_MAX_LEN was replaced with CONFIG_BT_NIMBLE_GAP_DEVICE_NAME_MAX_LEN
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:37 CONFIG_NIMBLE_ATT_PREFERRED_MTU was replaced with CONFIG_BT_NIMBLE_ATT_PREFERRED_MTU
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:38 CONFIG_NIMBLE_SVC_GAP_APPEARANCE was replaced with CONFIG_BT_NIMBLE_SVC_GAP_APPEARANCE
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:39 CONFIG_NIMBLE_ACL_BUF_COUNT was replaced with CONFIG_BT_NIMBLE_ACL_BUF_COUNT
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:40 CONFIG_NIMBLE_ACL_BUF_SIZE was replaced with CONFIG_BT_NIMBLE_ACL_BUF_SIZE
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:41 CONFIG_NIMBLE_HCI_EVT_BUF_SIZE was replaced with CONFIG_BT_NIMBLE_HCI_EVT_BUF_SIZE
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:42 CONFIG_NIMBLE_HCI_EVT_HI_BUF_COUNT was replaced with CONFIG_BT_NIMBLE_HCI_EVT_HI_BUF_COUNT
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:43 CONFIG_NIMBLE_HCI_EVT_LO_BUF_COUNT was replaced with CONFIG_BT_NIMBLE_HCI_EVT_LO_BUF_COUNT
/Users/derek/esp/FreeRTOS/build/sdkconfig.defaults:44 CONFIG_NIMBLE_MESH was replaced with CONFIG_BT_NIMBLE_MESH
/var/folders/m1/ybr42xhs5g51l5dv7m19ggj00000gn/T/confgen_tmpbot061da:20 line was updated to CONFIG_BT_BLUEDROID_ENABLED=n
/var/folders/m1/ybr42xhs5g51l5dv7m19ggj00000gn/T/confgen_tmpbot061da:34 line was updated to CONFIG_BT_NIMBLE_DEBUG=n
/var/folders/m1/ybr42xhs5g51l5dv7m19ggj00000gn/T/confgen_tmpbot061da:44 line was updated to CONFIG_BT_NIMBLE_MESH=n
-- App "amazon-freertos" version: 1
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/derek/esp/FreeRTOS/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /Users/derek/esp/FreeRTOS/vendors/espressif/esp-idf/components/esp32/ld/esp32.peripherals.ld
=========================Resolving dependencies==========================
                          All modules available
[... 123 lines cut ...]
  Enabled by user:         ble, ble_hal, ble_wifi_provisioning, common_io, core_mqtt_demo_
                           dependencies, logging, pkcs11, pkcs11_helpers, pkcs11_
                           implementation, pkcs11_utils, platform, secure_sockets, wifi
  Enabled by dependency:   backoff_algorithm, common, core_mqtt, crypto, demo_base,
                           dev_mode_key_provisioning, freertos, kernel, pkcs11_mbedtls,
                           tls, transport_interface_secure_sockets, utils
  3rdparty dependencies:   mbedtls, pkcs11, tinycbor
  Available demos:         demo_core_mqtt, demo_gatt_server, demo_mqtt_ble,
                           demo_numeric_comparison, demo_tcp, demo_wifi_provisioning
  Available tests:
=========================================================================

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/derek/esp/FreeRTOS/build
```

**Steps to reproduce bug**
download code from AWS console
extract zip file
vendors/espressif/esp-idf/install.sh [this gives me error, and I need to add execute permissions]
source vendors/espressif/esp-idf/export.sh
edit tools/aws_config_quick_start/configure.json
python SetupAWS.py setup
idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build

yourslab, 2021-07-23:
Thanks for pointing this out @knap1930. I was able to reproduce the problem on my own Mac, and we will be investigating further to resolve the issue. For now, I would suggest using the demo by cloning the Github repository.

yourslab, 2021-12-10:
Hi @knap1930, has cloning the repository and build through that clone solved the issue for you?

yourslab, 2021-12-13:
I am closing this issue but please feel free to re-open if the problem arises again

closed 2021-12-13 by yourslab