aws/amazon-freertos #3338: [BUG] changing demo's does not work

knap1930, 2021-07-23:
**Describe the bug**
When I change `#define CONFIG_CORE_MQTT_MUTUAL_AUTH_DEMO_ENABLED` to `#define CONFIG_OTA_HTTP_UPDATE_DEMO_ENABLED` in `vendors/vendor/boards/board/aws_demos/config_files/aws_demo_config.h` like mentioned in your docs... absolutely nothing happens! 

**System information**
- Hardware board: [ ... ]
- IDE used: [ ... ]
- Operating System: [ Windows | Linux | MacOS ]
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [ mqtt_demo_mutual_auth | shadow_demo | custom application | etc... ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
The selected demo (in this case OTA HTTP demo runs)

My guess is that it looks like there is a second file where this defined vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h

knap1930, 2021-07-23:
I can confirm changing the demo in both places
`vendors/vendor/boards/board/aws_demos/config_files/aws_demo_config.h`
AND
`vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h`
works... I feel like I was eating crazy pills for a while here..

knap1930, 2021-07-23:
so I guess I see where I went wrong, and maybe this isn't a bug...

[image]

I guess the red parts are suppose to be variables, and I needed to replace them with my specific vendor and board... but the fact `vendors/vendor/boards/board/aws_demos/config_files/aws_demo_config.h` exists is maddening.. maybe this is used for a template when creating a new board?

closed 2021-07-23 by knap1930

mleister97, 2021-09-27:
Facing the same issue. Thanks for your input!