aws/amazon-freertos #3339: [General] Size analysis of the binaries of  esp32 project.

Devyami-Apcon, 2021-07-25:
**Briefly summarize the issue being raised**
I have 

**Describe the desired outcome**
Identify the intended result of further discussion and eventual resolution of this issue.

**System information**
- Hardware board: [ ... ]
- IDE used: [ ... ]
- Operating System: [ Windows | Linux | MacOS ]
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [ mqtt_demo_mutual_auth | shadow_demo | custom application | etc... ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code/Steps to reproduce the current status"`.

**References**
Provide any screenshots, logs, etc. that may assist in providing a better understanding of the highlighted issue.

**Code/Steps to reproduce the current status**
If applicable, provide concrete instructions or the relevant code snippet(s) for reproducing the current status. Code should be wrapped in the ``cpp`` tag in order to be displayed clearly. For example:
```cpp
printf("Hello World")
```
**Additional context**
Add any remaining information not mentioned in the previous responses.

yourslab, 2021-07-25:
Tagging @shubhamkulkarni97 who may be able to answer your question better.

Also, have you tried building with `idf.py` instead? If AFR subdirectory is `freertos`, you can call:
```sh
freertos/vendors/espressif/esp-idf/tools/idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build```

freertos/vendors/espressif/esp-idf/tools/idf.py size
```

shubhamkulkarni97, 2021-07-26:
Hi @Devyami-Apcon @yourslab, idf.py size command is not supported in amazon-freertos.

Instead you can run this command:
```
./freertos/vendors/espressif/esp-idf/tools/idf_size.py <path-to-map-file>
```

Devyami-Apcon, 2021-07-26:
@shubhamkulkarni97 

Thanks, this worked. 
Just a doubt though, my bin file and map file have different names. For example my bin file is afr_demo.bin and my map file is aws_demos.map 

Would I still be tracking my customised project or would this be tracking predefined aws demos?

shubhamkulkarni97, 2021-07-26:
@Devyami-Apcon It should track for customised project.
Thanks for pointing out that the map file is not generated with correct name. I will create a PR to fix the name of generated map file.

Devyami-Apcon, 2021-07-26:
Thanks @shubhamkulkarni97 @yourslab for your responses. Closing the issue.

closed 2021-07-26 by Devyami-Apcon