aws/amazon-freertos #3343: aws_application_version.h not avaliable in 202107.00-3-g19c657cf0

Git-shouvik, 2021-07-28:
**Briefly summarize the issue being raised**
There is no such file present in **amazon-freertos/demos/include**, as mentioned  [https://docs.aws.amazon.com/freertos/latest/userguide/dg-ota-update-firmware.html](url).
**Describe the desired outcome**
Perform ota update test by changing the APP_VERSION_BUILD token value in aws_application_version.h .

**System information**
- Hardware board: [ esp32 devkitc ]
- IDE used: [  ubuntu terminal ]
- Operating System: [ Linux ]
- Code version:  202107.00-3-g19c657cf0
- Project/Demo: [ ota ]

Raghav3107, 2021-07-30:
Hello @Git-shouvik 

Check the `ota_demo_config.h` files at `.../aws_demos/config_files` directory.

alfred2g, 2021-08-16:
closing the Issue due to no response.
please reopen with new info.

closed 2021-08-16 by alfred2g