aws/amazon-freertos #3344: [General] idf.py size, idf.py size-components failed when using amazon-freertos as external project.

Raghav3107, 2021-07-30:
I m using the amazon-freetos as an external repository. When I m using the `idf.py menuconfig`, `idf.py clean` all this command is working, but `idf.py size, size-components, and size-files` are not working.

**System information**
- Hardware board: [ esp-devkit ]
- IDE used: [ ... ]
- Operating System: [ Linux ]
- Code version: amazon-freertos 202107 esp-idf v4.2.1

Here is the link of repo [link ](https://github.com/Raghav3107/amazon-freertos-custom-v202107).

Thanks!

yourslab, 2021-08-01:
Hi @Raghav3107. This was answered in a previous issue:
https://github.com/aws/amazon-freertos/issues/3339#issuecomment-886339547

To quote, you can use the following script instead:
```
./vendors/espressif/esp-idf/tools/idf_size.py
```

I'm not entirely sure if `size-components` and `size-files` are supported yet, but maybe @shubhamkulkarni97 will be able to assist you more.

shubhamkulkarni97, 2021-08-04:
Hi @Raghav3107 @yourslab,

* For size-components, you can use `./vendors/espressif/esp-idf/tools/idf_size.py --archives <map-file>`
* For size-files, you can use `./vendors/espressif/esp-idf/tools/idf_size.py --files <map-file>`

Thanks,
Shubham

closed 2021-08-07 by Raghav3107