aws/amazon-freertos #3347: [General] Not able to build AWS OTA Demo for ESP32

pratyushjagaty, 2021-08-04:
**Briefly summarize the issue being raised**
When I try to run the following command: 
```
idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build
```

I get the following error:

```
/Users/jagaty/esp/esp-idf/components/newlib/platform_include/time.h:28:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_settime(clockid_t clock_id, const struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/Users/jagaty/esp/esp-idf/components/newlib/platform_include/time.h:29:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_gettime(clockid_t clock_id, struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/Users/jagaty/esp/esp-idf/components/newlib/platform_include/time.h:30:18: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_getres(clockid_t clock_id, struct timespec *res);
                  ^~~~~~~~~
                  _lock_t
In file included from /Users/jagaty/Documents/Experiments/OTA_Test/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform.h:68,
                 from /Users/jagaty/Documents/Experiments/OTA_Test/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c:18:
/Users/jagaty/Documents/Experiments/OTA_Test/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform_time.h:78:9: error: unknown type name 'time_t'
 typedef time_t mbedtls_time_t;
```

**Describe the desired outcome**
Ideally I should be able to able to build this and then flash onto my device.

**System information**
- Hardware board: [https://shop.m5stack.com/products/esp32-psram-timer-camera-x-ov3660](M5 Timer X Camera with ESP32)
- IDE used: VS Code
- Operating System: MacOS 
- Code version: latest - downloaded from AWS Console
- Project/Demo: OTA

**References**
[error.txt](https://github.com/aws/amazon-freertos/files/6930506/error.txt)

**Code/Steps to reproduce the current status**
- Download FreeRTOS with OTA demo enabled from AWS console onto the MAC.
- [ESP32 specific setup](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html)
- Complete the [prerequisites](https://docs.aws.amazon.com/freertos/latest/userguide/ota-prereqs.html)
- Run `idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build` from the **FreeRTOS** directory.

**Additional context**
[Similar issue](https://github.com/aws/amazon-freertos/issues/2959#issuecomment-769577330)
But I cannot find any solution mentioned to the issue.

Would appreciate any help!

AniruddhaKanhere, 2021-08-04:
Hello @pratyushjagaty!
Thank you for reporting this error. We are looking into this and will get back to you.

pratyushjagaty, 2021-08-05:
IDF Version:
```
idf.py --version
ESP-IDF v4.2.2-197-gf4a2c5b52
```

Executed `fullclean` with the result:
```
idf.py fullclean
Executing action: fullclean
Done
```

Finally run: `idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build`

The errors I face are the same. The exact stack trace is in `errors.txt`

AniruddhaKanhere, 2021-08-05:
Thank you for those details @pratyushjagaty.
Would you mind printing out the environment variables as well (you can type `env` to print all environment variables)? I think that `fullclean` does not remove the environment variables - this is what I got from the ESP website:
_`idf.py fullclean` will delete the entire “build” directory contents. This includes all CMake configuration output. The next time the project is built, CMake will configure it from scratch. Note that this option recursively deletes all files in the build directory, so use with care. Project configuration is not deleted._

You should delete/clear the (old) environment variables from the list before starting anew.

Once you clean the environment variables, you can add them back to your terminal's path with this command (as copied from the getting started guide): `source vendors/espressif/esp-idf/export.sh`

pratyushjagaty, 2021-08-06:
Env variables after a clean and restart
```
TERM_SESSION_ID=w0t0p0:BA42DD65-D486-42F5-93AF-3DC6B03CA3A7
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.n0CM5K3R1J/Listeners
LC_TERMINAL_VERSION=3.4.8
COLORFGBG=15;0
ITERM_PROFILE=Default
XPC_FLAGS=0x0
PWD=/Users/jagaty/Desktop/FreeRTOS_Testing/FreeRTOS
SHELL=/bin/zsh
__CFBundleIdentifier=com.googlecode.iterm2
LC_CTYPE=UTF-8
TERM_PROGRAM_VERSION=3.4.8
TERM_PROGRAM=iTerm.app
PATH=/apollo/env/SDETools/bin:/Users/jagaty/.toolbox/bin:/Library/Frameworks/Python.framework/Versions/3.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/jagaty/Library/Python/2.7/bin:/Library/Apple/usr/bin
LC_TERMINAL=iTerm2
COLORTERM=truecolor
COMMAND_MODE=unix2003
TERM=xterm-256color
HOME=/Users/jagaty
TMPDIR=/var/folders/xt/qc1497916t14xxc0b2rbhqwsq_xh4j/T/
USER=jagaty
XPC_SERVICE_NAME=0
LOGNAME=jagaty
ITERM_SESSION_ID=w0t0p0:BA42DD65-D486-42F5-93AF-3DC6B03CA3A7
__CF_USER_TEXT_ENCODING=0x0:0:0
SHLVL=1
OLDPWD=/Users/jagaty
JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_231.jdk/Contents/Home
P9K_TTY=old
_P9K_TTY=/dev/ttys000
ZSH=/Users/jagaty/.oh-my-zsh
PAGER=less
LESS=-R
LSCOLORS=Gxfxcxdxbxegedabagacad
P9K_SSH=0
_=/usr/bin/env
```

Post this I run: `. $HOME/esp/esp-idf/export.sh`

And the env variables are :
```
TERM_SESSION_ID=w0t0p0:BA42DD65-D486-42F5-93AF-3DC6B03CA3A7
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.n0CM5K3R1J/Listeners
LC_TERMINAL_VERSION=3.4.8
COLORFGBG=15;0
ITERM_PROFILE=Default
XPC_FLAGS=0x0
PWD=/Users/jagaty/Desktop/FreeRTOS_Testing/FreeRTOS
SHELL=/bin/zsh
__CFBundleIdentifier=com.googlecode.iterm2
LC_CTYPE=UTF-8
TERM_PROGRAM_VERSION=3.4.8
TERM_PROGRAM=iTerm.app
PATH=/Users/jagaty/esp/esp-idf/components/esptool_py/esptool:/Users/jagaty/esp/esp-idf/components/espcoredump:/Users/jagaty/esp/esp-idf/components/partition_table:/Users/jagaty/esp/esp-idf/components/app_update:/Users/jagaty/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/Users/jagaty/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/Users/jagaty/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/jagaty/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/jagaty/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin:/Users/jagaty/.espressif/python_env/idf4.2_py3.7_env/bin:/Users/jagaty/esp/esp-idf/tools:/apollo/env/SDETools/bin:/Users/jagaty/.toolbox/bin:/Library/Frameworks/Python.framework/Versions/3.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/jagaty/Library/Python/2.7/bin:/Library/Apple/usr/bin
LC_TERMINAL=iTerm2
COLORTERM=truecolor
COMMAND_MODE=unix2003
TERM=xterm-256color
HOME=/Users/jagaty
TMPDIR=/var/folders/xt/qc1497916t14xxc0b2rbhqwsq_xh4j/T/
USER=jagaty
XPC_SERVICE_NAME=0
LOGNAME=jagaty
ITERM_SESSION_ID=w0t0p0:BA42DD65-D486-42F5-93AF-3DC6B03CA3A7
__CF_USER_TEXT_ENCODING=0x0:0:0
SHLVL=1
OLDPWD=/Users/jagaty
JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_231.jdk/Contents/Home
P9K_TTY=old
_P9K_TTY=/dev/ttys000
ZSH=/Users/jagaty/.oh-my-zsh
PAGER=less
LESS=-R
LSCOLORS=Gxfxcxdxbxegedabagacad
P9K_SSH=0
IDF_PATH=/Users/jagaty/esp/esp-idf
IDF_TOOLS_EXPORT_CMD=/Users/jagaty/esp/esp-idf/export.sh
IDF_TOOLS_INSTALL_CMD=/Users/jagaty/esp/esp-idf/install.sh
OPENOCD_SCRIPTS=/Users/jagaty/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts
IDF_PYTHON_ENV_PATH=/Users/jagaty/.espressif/python_env/idf4.2_py3.7_env
_=/usr/bin/env
```

And finally I have run: 
`idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build`

I get the same error. (Linked in `errors.txt` above)

AniruddhaKanhere, 2021-08-08:
Hello @pratyushjagaty, I am sorry to hear that. We are trying to debug the issue you are having.
Also, while we debug the issue, it might be fruitful to parallelly ask this question on the ESP forum as they have more knowledge about the tool.

pratyushjagaty, 2021-08-09:
Yeah, I have gone ahead and done it. It's pending approval from a moderator. Will keep you posted.

ravibhagavandas, 2021-08-09:
Hello @pratyushjagaty,

Taking a look at the environment variables it seems the `IDF_PATH`, `IDF_TOOLS_EXPORT_CMD` etc.. are set to an esp idf path outside of amazon freertos. This could be an issue as the code in this repo uses a customized version of esp-idf which is in `vendors/espressif/esp-idf`. 
As mentioned in doc [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf42), you need to only download the ESP tool chain externally but use the IDF from within this repo under `vendors/espressif/esp-idf`. You can do so by first opening a new session to clear all current env variables and executing `vendors/espressif/esp-idf/install.sh` and `source vendors/espressif/esp-idf/export.sh`.

pratyushjagaty, 2021-08-10:
This solved the issue. Had to do what you said and do a `chmod` on the `idf.py` inside the tools directory. I am closing this now! 

Thanks!

closed 2021-08-10 by pratyushjagaty