aws/amazon-freertos #3351: [General] How to disable all Demos?

dFohlen, 2021-08-06:
**Briefly summarize the issue being raised**

We are in the process of building our own project using AWS IoT Core and AWS FreeRTOS. We have configured the AWS FreeRTOS device software for download in IoT Core -> Software -> FreeRTOS device software. At least one demo has to be selected here. Then we set up our project according to the instructions: [Using FreeRTOS in your own CMake project for ESP32](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project ).

Using these instructions, I have deactivated all demos in the config file: [Configuring the demos](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-next-steps.html#configuring-demos)

But our project still includes lot of demo dependencies:
```
=========================Resolving dependencies==========================
                          All modules available                          

====================Configuration for FreeRTOS====================
  Version:                 202107.00
  Git version:             Unknown

Target microcontroller:
  vendor:                  Espressif
  board:                   ESP32-DevKitC
  description:             Development board produced by Espressif that comes in two 
                           variants either with ESP-WROOM-32 or ESP32-WROVER module
  family:                  ESP32
  data ram size:           520KB
  program memory size:     4MB

Host platform:
  OS:                      Darwin-20.5.0
  Toolchain:               xtensa-esp32
  Toolchain path:          /User/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.
                           0/xtensa-esp32-elf
  CMake generator:         Ninja

FreeRTOS modules:
  Modules to build:        backoff_algorithm, ble, ble_hal, ble_wifi_provisioning, 
                           common, common_io, core_http, core_http_demo_dependencies, 
                           core_json, core_mqtt, core_mqtt_agent, core_mqtt_agent_demo_
                           dependencies, core_mqtt_demo_dependencies, crypto, 
                           dev_mode_key_provisioning, device_shadow, device_shadow_demo_
                           dependencies, http_demo_helpers, kernel, logging, 
                           mqtt_agent_interface, mqtt_demo_helpers, mqtt_subscription_
                           manager, ota, ota_demo_dependencies, ota_demo_version, pkcs11, 
                           pkcs11_helpers, pkcs11_implementation, pkcs11_utils, platform, 
                           secure_sockets, tls, transport_interface_secure_sockets, wifi
  Enabled by user:         ble, ble_hal, ble_wifi_provisioning, common_io, core_http_demo_
                           dependencies, core_json, core_mqtt_agent_demo_dependencies, 
                           core_mqtt_demo_dependencies, device_shadow, device_shadow_demo_
                           dependencies, logging, ota_demo_dependencies, pkcs11, 
                           pkcs11_helpers, pkcs11_implementation, pkcs11_utils, platform, 
                           secure_sockets, wifi
  Enabled by dependency:   backoff_algorithm, common, core_http, core_mqtt, 
                           core_mqtt_agent, crypto, demo_base, dev_mode_key_provisioning, 
                           freertos, http_demo_helpers, kernel, mqtt_agent_interface, 
                           mqtt_demo_helpers, mqtt_subscription_manager, ota, 
                           ota_demo_version, pkcs11_mbedtls, tls, transport_interface_
                           secure_sockets, utils
  3rdparty dependencies:   mbedtls, pkcs11, tinycbor
  Available demos:         demo_core_http, demo_core_mqtt, demo_core_mqtt_agent, 
                           demo_device_shadow, demo_gatt_server, demo_mqtt_ble, 
                           demo_numeric_comparison, demo_ota_ble, demo_ota_core_http, 
                           demo_ota_core_mqtt, demo_shadow_ble, demo_tcp, demo_wifi_
                           provisioning
  Available tests:         
=========================================================================

-- Configuring done
-- Generating done
```

This is what we actually want to use:
```
target_link_libraries(my_project PRIVATE
        idf::json
        idf::wpa_supplicant
        AFR::wifi
        AFR::ota
        AFR::core_mqtt_agent
        AFR::mqtt_agent_interface
        AFR::transport_interface_secure_sockets
        AFR::core_json
        AFR::device_shadow
        AFR::dev_mode_key_provisioning)
```        

**Describe the desired outcome**

How do I get rid of all the dependencies through the demos? The demos make it really difficult to keep your own code separate without creating any unwanted dependencies.

**System information**
- Hardware board: ESP32-DevKitC
- IDE used: Jetbrains CLion
- Operating System: macOS Big Sur Version 11.4 (20F71)
- Code version: 202107.00

yanjos-dev, 2021-12-06:
Hello @dFohlen, I apologize for the delay. The list of libraries that you referenced is used to generate a list of the libraries that might be used. It’s a list of libraries that have the required dependencies fulfilled. I don’t believe that means the libraries are linked with the application by default. Can you confirm if this is still an issue and whether there are unintended libraries being linked with your application?

abhidixi11, 2021-12-29:
Hello @dFohlen I'm closing this issue. Please feel free to open a new issue if you still have questions regarding this matter. Thanks.

closed 2021-12-29 by abhidixi11