aws/amazon-freertos #3356: `.iram0.text' will not fit in region `iram0_0_seg' error when enabling the external ram.

Raghav3107, 2021-08-15:
Hello @shubhamkulkarni97 

One new issue I m facing when I enabling the external ram in esp32.

**Error:**
```
l  -L /home/horsemann/Desktop/WorkSpace/universalRemote_v202107/freertos/vendors/espressif/esp-idf/components/esp_wifi/lib/esp32  -u esp_app_desc  freertos/libraries/3rdparty/libafr_3rdparty_mbedtls.a  freertos/afr_kernel.a && :
/home/horsemann/Desktop/WorkSpace/esp_toolchain/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: afr_demo section `.iram0.text' will not fit in region `iram0_0_seg'
/home/horsemann/Desktop/WorkSpace/esp_toolchain/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/horsemann/Desktop/WorkSpace/esp_toolchain/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 956 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
Changes which I did.
I added the pre-built library and linked with the target using below line and I added mulitple components in custom repo.
Also, I enable the rtc gpio in menuconfig.
```
# Declaring the precompiled BSEC lib as imported global so that it doesn't need to be recompiled
add_library(bme680lib STATIC IMPORTED GLOBAL)
set_target_properties(bme680lib PROPERTIES IMPORTED_LOCATION "${PROJECT_SOURCE_DIR}/components/bme680/algo/normal_version/bin/esp/esp32/libalgobsec.a")

# For some reason, math library from newlib isn't implicitly imported so that needs to be manually
# done for now. You might be able to get rid of the hack at some point
add_library(mathlib STATIC IMPORTED GLOBAL)
set_target_properties(mathlib PROPERTIES IMPORTED_LOCATION "${PROJECT_SOURCE_DIR}/freertos/vendors/espressif/esp-idf/components/newlib/lib/libm.a")
```
 When I m compile without enabling the external ram that time I not facing any  issue.
After enabling the external ram,  I m getting above error.

Sdkconfig:
```
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_TYPE_ESPPSRAM16=
CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
CONFIG_SPIRAM_TYPE_ESPPSRAM64=
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_SPEED_80M=
CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_USE_MEMMAP=
CONFIG_SPIRAM_USE_CAPS_ALLOC=
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
```
```
#
# RTCIO configuration
#
CONFIG_RTCIO_SUPPORT_RTC_GPIO_DESC=y
```

shubhamkulkarni97, 2021-08-16:
Hi @Raghav3107,

This error states that IRAM memory region of ESP32 has overflowed due to excess code being added to this region.

To fix this issue, you can disable following configs in menuconfig:
* CONFIG_ESP32_WIFI_IRAM_OPT
* CONFIG_ESP32_WIFI_RX_IRAM_OPT

Disabling these configs will place some WiFi specific functions in flash instead of IRAM. This will reduce IRAM usage at the cost of reduced WiFi performance.

You can find documentation for these configs [here](https://docs.espressif.com/projects/esp-idf/en/v4.2.2/esp32/api-reference/kconfig.html#config-esp32-wifi-iram-opt).

Hope this helps!

Thanks,
Shubham

Raghav3107, 2021-08-16:
Hello @shubhamkulkarni97 
I will try this.
But I have some question:

I check the components size. And you  can see in the log, program  uses the 0.8% of IRAM then how should the IRAM memory region overflow?

```
horsemann@Raghav:~/Desktop/WorkSpace/universalRemote_v202107$ ./freertos/vendors/espressif/esp-idf/tools/idf_size.py --archives ./build/afr_demo.map
Total sizes:
 DRAM .data size:   16688 bytes
 DRAM .bss  size:   61408 bytes
Used static DRAM:   78096 bytes (  46484 available, 62.7% used)
Used static IRAM:    1027 bytes ( 130045 available, 0.8% used)
      Flash code:  983603 bytes
    Flash rodata:  252032 bytes
Total image size:~1253350 bytes (.bin may be padded larger)

```

shubhamkulkarni97, 2021-08-16:
@Raghav3107 There is some issue while running the command due to which you are getting less usage in the log.

I can see 98.6% IRAM usage on default aws_demos application with external ram enabled.
```
Total sizes:
 DRAM .data size:   14856 bytes
 DRAM .bss  size:   36704 bytes
Used static DRAM:   51560 bytes (  73020 available, 41.4% used)
Used static IRAM:  129267 bytes (   1805 available, 98.6% used)
      Flash code:  726095 bytes
    Flash rodata:  168136 bytes
Total image size:~1038354 bytes (.bin may be padded larger)
```

> And If it overflowing, then after disabling the external ram option why its not throwing same error?
> Because after disabling external ram in menuconfig, i m not getting any error.

This behavior is expected because when you enable external RAM some more code is added to IRAM region. This is mainly required to handle data in external ram.

closed 2021-08-20 by Raghav3107