aws/amazon-freertos #3358: MQTT disconnection after some time and up on reconnect for hours it does not connect, it connects only after reboot.

kumarfirmware, 2021-08-16:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

ESP32 is connected to the corporate network Wi-Fi. Very High speed network. No chance of unstable network.
Also ESP does not loose ping at all. Ping is very stable.

**Device is able to have successful MQTT connection to the AWS cloud for almost a day and after that ping response failed and MQTT disconnection happened.**

**After the MQTT disconnection , device is not able to reconnect successfully even after trying to do MQTT connect for hours together.**

Only after reboot of the device, MQTT connection is established successfully.

This is horrible and shame and same issues from two years and no efficient solutions or quick fixes. 

I am asking same questions again back and forth but no confident solution.

How do we even use this SDK in production? How do we even deploy this in mass production? 

This is shame , random MQTT disconnection and does not connect back even after 100 attempts of trying to connect back.

only reboot solves.

Please let me know how do I fix this as soon as possible?

Thanks,
Kumar.

kumarfirmware, 2021-08-19:
Hi Team,

Can you please help me on this as soon as possible. This is an issue faced in the site. This needs to be fixed immediately.

I am clueless with this.

I need your immediate assistance in this.

Thanks
Kumar.

pvyawaha, 2021-08-19:
Hello @kumarfirmware ,

Can you please share the logs when this happens ? Also are you using the exponential backoff and reconnection method for reconnection attempts?

kumarfirmware, 2021-08-19:
Hi pvyawaha,

Thanks for the message. I am not using exponential backoff and reconnection method for reconnection attempts.

I am just using 10 seconds gap ad reconnecting.

MQTT connect, if it fails, wait 10 seconds, connect again. like this continuously I am doing.

Please let me know if this is a problem.

Thanks.
Kumar.

kumarfirmware, 2021-09-01:
Hi Muneeb,

Thanks for the message.

As you already know I am using AWS FreeRTOS version 202007 and ESP32.

Do you think the above issue related to secure sockets in not fixed in the version 202007.

can you please give me the steps how should I update the fix in the version 202007.

Can you please let me know here with a code snippet how to close the MQTT connection gracefully?

Thanks, Regards,
Kumar.

muneebahmed10, 2021-11-21:
Closing this as this issue is being discussed in #3359.

closed 2021-11-21 by muneebahmed10