aws/amazon-freertos #3359: If any MQTT publish message fails with the  Error -27648 while sending data,  it continuously fails, recovers only on reboot

kumarfirmware, 2021-08-16:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

If any MQTT publish message fails with the following error, it will not recover and it continuously fails, recovers only on reboot.

[iot_thread] [ERROR][NET][117930] Error -27648 while sending data.

This occurs very randomly. 

In my opinion this error is observed happening generally when two close publish or one publish and one subscribe happens almost together or close or back to back.

Can we do publish from multiple threads with out worrying about thread safety? I strongly suspect thread safety here.

How can we use this SDK or how can we trust this SDK, if this error keeps coming now and then and creates instability in the system.

Please let me know the fix for this as soon as possible.

Again this is the same issue from 2 years.

There is single MQTT connection for all the three threads, there is unique  client ID.

Thanks,
Kumar.

[image]

kumarfirmware, 2021-08-19:
Hi Team,

Can you please help me on this as soon as possible. This is an issue faced in the site. This needs to be fixed immediately.

I am clueless with this.

I need your immediate assistance in this.

Thanks
Kumar.

pvyawaha, 2021-08-19:
Hello @kumarfirmware ,

If you are using coreMQTT APIs from multiple threads and sharing connection then thread safety must be handled in the application.

Alternatively you can use the coreMQTT Agent library that adds thread safety to the coreMQTT library. Please refer https://freertos.org/mqtt-agent/index.html

kumarfirmware, 2021-08-19:
Hi  Pvyawaha,

Thanks for the message. I am using AWS FreeRTOS version 202007 and ESP32. There is no coreMQTT and coreMQTT Agent library in this version.

Please let me know how to proceed in detail. please give me clear detailed information with respect to version 202007.

Thanks
Kumar.

muneebahmed10, 2021-09-01:
Hi @kumarfirmware,

From your logs, it looks like the publish completes successfully, and then on line 54 the timestamp jumps from 1659 to 4628, where it fails to send a PINGREQ packet as part of the keep alive job. Is the connection intended to be left inactive long enough for the keep alive job to kick in?

Also, there have been issues in the past with this version of the MQTT library where the [network receive task fails](https://github.com/aws/amazon-freertos/issues/2356#issuecomment-694628494) and the application isn't notified until the keep alive job fails, resulting in similar log messages to what you are seeing. Can you enable debug logs to try to narrow down where the issue occurs between the send of the PUBLISH and the keep alive attempt?

Additionally, as the MQTT library that was part of 202007 was removed, we recommend that you upgrade to the latest release if you are able.

kumarfirmware, 2021-09-01:
Hi Muneeb,

Thanks for the message. I do not close the MQTT connection and there is no need to close the MQTT connection.  MQTT PING is a continuous process, it will go on.

The worry is sometimes this Error -27648 does not go off hours together even after reboot, it continuously keep giving this Error -27648.

I am using AWS FreeRTOS version 202007 and ESP32. As you know it takes long time to move to AWS FreeRTOS version 202107.

Meanwhile production sites are struggling with this issue with AWS FreeRTOS version 202007.

How do I replace MQTT library seamlessly in the AWS FreeRTOS version 202007?

Please help me with this immediately with detailed steps.

Any immediate help in this is highly appreciated.

Thanks, Regards,
Kumar.

muneebahmed10, 2021-09-02:
Hi @kumarfirmware,

Can you enable debug logging? As I mentioned earlier, it's possible the connection goes down even before the ping request attempt is made, and more logs could help narrow down when the error occurs.

If you aren't able to upgrade to the latest release due to the MQTT API changing, are you able to use the MQTT library from previous tags, like 202012.00 or [20210526_Archive](https://github.com/aws/amazon-freertos/tree/20210526_Archive), which was the last tag before the old MQTT API was removed? I ask since both of those tags contain the coreMQTT library, and also have a compatibility layer to interface with the old MQTT API that you are using in 202007.00. I think using coreMQTT along with the compatibility layer might help you since you would still be able to use the same API you currently are.

muneebahmed10, 2021-09-02:
The MQTT library in 202007 and 202012 is not the same. 202012 uses coreMQTT, but only contains a compatibility layer to be able to use the APIs in 202007. Internally, it will still use coreMQTT, so you will need to include the coreMQTT library as well. If 202012 works for you, then I would recommend next using 20210526_Archive, since it contains bug fixes for memory leaks that were in 202012.

That tag would have had testing for demo builds and runs when each commit was merged, but it does not have extensive testing for every build combination as our release tags do.

kumarfirmware, 2021-09-02:
Hi Muneeb,

Thanks for the message. I started using 20210526_Archive. Everything looks good, until now.

But I guess shadows are handled a bit differently here in this version compared to 202007, so I need to adopt my code here.

So I assume I should do device shadows also using normal MQTT Publish and subscribe to match the compatibility for MQTT API used in AWS V202007.

So I cannot use the shadow library here right?

Please confirm this as soon as possible, so that it will help me to move forward.

Thanks, Regards,
Kumar.

muneebahmed10, 2021-09-02:
Hi,

I'm not sure why you wouldn't be able to use the Shadow library in that version. Looking at the [commit history](https://github.com/aws/amazon-freertos/commits/20210526_Archive/libraries/c_sdk/aws/shadow), there weren't that many changes to the old shadow files in that tag. Since that version of the shadow library used the old MQTT APIs, it will just be using the MQTT compatibility layer instead, so the APIs would be the same. Of course, the recommendation is to use the latest [Device Shadow library](https://github.com/aws/device-shadow-for-aws-iot-embedded-sdk/tree/main) but the old API is still present in the 20210526 tag.

I'm not sure what you mean by master/release for 20210526_Archive. `20210526_Archive` is a git tag that points to the same commit that the `main` branch pointed to on 2021-05-26. The current `main` and `release` branches point to newer commits that do not have the MQTT compatibility layer.

Release branches are tested the most. Our recommendation is still to use the latest release version 202107.00, but if that is not possible then you can use 202012 or 20210526_Archive. It's up to you whether you are willing to use 20210526_Archive in production; if you only want to use release tags, then you should know that 20210526 contains bug fixes for memory leaks that were present in 202012, so you should copy over the changes that were made to the MQTT compatibility layer

kumarfirmware, 2021-09-03:
Hi Muneeb,

Thanks for the message. Thanks for the clarification.

I am getting the following error when I am using old shadows in 20210526_Archive.

../vendors/espressif/boards/esp32/aws_demos/application_code/comms.c:40:10: fatal error: aws_iot_shadow.h: No such file or directory
 #include "aws_iot_shadow.h"

         ^~~~~~~~~~~~~~~~~~
compilation terminated.

Can you please help me with this as soon as possible.

Thanks, Regards,
Kumar.

kumarfirmware, 2021-09-03:
Hi Muneeb,

Can you please help me with the above error as soon as possible. 

I guess shadow library is not building and I am not able to use it in my application code.

Thanks, Regards,
Kumar.

kumarfirmware, 2021-09-03:
Hi Muneeb,

Can you please help me with the above error as soon as possible.

Thanks, Regards,
Kumar.

muneebahmed10, 2021-09-03:
From the error, it looks like you need to include [this file](https://github.com/aws/amazon-freertos/blob/20210526_Archive/libraries/c_sdk/aws/shadow/include/aws_iot_shadow.h). You should add the directory that file is in to your include path before compiling.

kumarfirmware, 2021-09-03:
Hi Muneeb,

In 202007 it works automatically. I am not sure how to add this to the make file in 20210526_Archive.

Can you please help me with this?

Thanks, Regards,
Kumar.

muneebahmed10, 2021-09-03:
What build system/compiler are you using? For gcc compilers, you can add include paths by using the `-I` option

kumarfirmware, 2021-09-03:
Hi Muneeb,

I am using gcc compiler.

See in the build folder under CMakeFiles I see the following files. Here it clearly shows shadow library is not compiled at all.

I am not using anything on my own. I am just using the existing make files.

[image]

Thanks, Regards,
Kumar.

Thanks, Regards,
Kumar.

muneebahmed10, 2021-09-04:
If you are using that version of shadow and cmake, then you should add a dependency on `AFR::shadow` in your `CMakeLists.txt`. For example, [this demo](https://github.com/aws/amazon-freertos/blob/3403475d47e17f0e7c1ab49a27d256ee1de2c517/demos/device_shadow_for_aws/CMakeLists.txt#L30) has a dependency on `AFR::device_shadow` since it uses the latest Shadow library. You would use `AFR::shadow` since you are using the older one.

kumarfirmware, 2021-09-04:
Hi Muneeb,

I tried this, but still it is not solving the issue.

We should some how call this AFR::shadow in CMakeLists.txt in vendors\espressif\boards\esp32 , because we are using shadows in this folder vendors\espressif\boards\esp32\aws_demos\application_code.

If you can crack this some how it will really help. I guess we are very close.

Thanks, Regards,
Kumar.

kumarfirmware, 2021-09-09:
Hi Muneeb,

I am able to build this successfully now.

I will test and see if I can reproduce the [iot_thread] [ERROR][NET][117930] Error -27648 while sending data.

I will let you know how the test goes.

Thanks,
Kumar.

dipen-1337lab, 2021-09-10:
Namaste @kumarfirmware,

> [iot_thread] [ERROR][NET][117930] Error -27648 while sending data.

One of the many possible reasons for the above error is the possibility of the duplication of `clientcredentialIOT_THING_NAME` on multiple devices/ clients.

We had encountered this error wherein two clients - one being our `ESP32 device` and, the other being a `client node running a Python script`, were both using the same `device identifier`. As a consequence, whenever the Python script published a MQTT Topic, the ESP32 device would start showing the `Error -27648 while sending data.` on the console. This happens, even if the published MQTT topic is not subscribed by the ESP32 device.

**Reason for such behavior:** According to the [MQTT spec](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc385349242), multiple connections to the same endpoint must have unique client IDs, otherwise it's likely that the first connection will suddenly be dropped by the server, without notification, after the second connection has been made. 

Requesting you to go through this https://github.com/aws/amazon-freertos/issues/2916#issuecomment-761175166, which is part of the [AFR & ESP32: NETWORK ERROR encountered, for aperiodic MQTT Publish operation](https://github.com/aws/amazon-freertos/issues/2916) issue, I had raised.

Additionally please note, the FreeRTOS version we are using for development is V202002.00, which is older than yours.

Hope this helps!

Thanks | Regards,
Dipen

kumarfirmware, 2021-09-12:
Hi Dipen,

Thanks for the message. 

I am pretty sure I am using unique clientcredentialIOT_THING_NAME. 

I am not doing any multiple MQTT connections with same clientcredentialIOT_THING_NAME.

I am still facing the issue [iot_thread] [ERROR][NET][117930] Error -27648 while sending data. some times.

Thanks, 
Kumar.

kumarfirmware, 2021-09-13:
Hi Muneeb,

As you know I am using AWS 20210526_Archive, as this has a compatibility layer for MQTT API used in AWS V202007.

I do not see any problem with the publish.  But I have a problem with the subscribe.

Whenever the cloud is sending a message to ESP32, I am getting the following error.

Length of the message from the cloud is around 5,600 bytes.

Please help me with the following error as soon as possible.

[image]

Thanks,
Kumar.

muneebahmed10, 2021-10-15:
Hi @kumarfirmware,

We apologize for taking this long to get back to you. The compatibility layer is thread safe. You can use it to publish from multiple tasks including OTA simultaneously.

Thanks

muneebahmed10, 2021-10-20:
Hi @kumarfirmware,

Looking at some of your logs,
```
384 388558 [normal_operatio] TLS_Connect fail (0x7d3, NULL)
385 388558 [normal_operatio] [ERROR][NET][3885580] Failed to establish new connection. Socket status: -1.
386 388559 [normal_operatio] [ERROR][MQTT][3885590] Failed to establish new MQTT connection, error NETWORK ERROR.
387 388559 [normal_operatio] [ERROR][DEMO][3885590] MQTT CONNECT returned error NETWORK ERROR.
Start establishing MQTT connection 
388 389569 [normal_operatio] [INFO ][MQTT][3895690] MQTT library successfully initialized.
abort() was called at PC 0x4016b883 on core 0
```

Some comments:
* You should not attempt a new MQTT connection if the TLS connection fails
* You should not call `IotMqtt_Init()` a second time without calling `IotMqtt_Cleanup()`

It looks like you are not yet able to reconnect. You should keep retrying the TLS connection before proceeding further with the MQTT connection.

pvyawaha, 2022-01-03:
Hello @kumarfirmware ,

Closing this issue now, please reopen if you have more questions on the same topic or create a new one for other topics.

closed 2022-01-03 by pvyawaha