aws/amazon-freertos #3360: [General] Multiple coreMQTT Agent Subscription fails

Prahar08modi, 2021-08-17:
**Briefly summarize the issue being raised**
Firstly, I'm using freertos in my own CMake project on ESP32 following the link [https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project](url) 

My Directory structure is mentioned below:

[image]

Background Knowledge:
I tested coreMQTT_Agent demo code and it worked successfully. Kudos to the team for the great work :1st_place_medal: . I want to level up the code by creating generic publish and subscribe functions based on coreMQTT_Agent library. This is the same as creating mqtt_demo_helpers (already provided under demos/common directory which is based on coreMQTT library) but for the coreMQTT_Agent library. 

So, basically, I want to create a coreMQTT_pegasus component (which handles MQTT connection with AWS IoT Cloud Broker and spawns an MQTT Agent Task) based on coreMQTT_Agent library and pubsub component with generic publish and subscribe functions which use coreMQTT_pegasus component to successfully publish and subscribe to any topic. I want to use the pubsub component in my application anywhere I want to publish a message or subscribe to a topic.

I successfully tested creating a connection to AWS IoT Cloud Broker and custom payload Publish to any topic. But, I'm facing issues while creating a generic subscribe function. Also, before mentioning the issue, I want to clear my understanding about the subscribe process handled by coreMQTT_Agent. 

My understanding of the coreMQTT_Agent's subscribe process:
1. Whenever a subscribe request is acknowledged back by AWS IoT Broker, we add the topic, topic length, callback, and callback context associated with that subscribe request to a global subscription list.
2. Also, coreMQTT_Agent has its own IncomingPublishCallback for any incoming publishes targeted to coreMQTT_Agent's MQTT Connection
3. This IncomingPublishCallback calls SubscriptionManager_HandleIncomingPublishes to check the global subscription list that the incoming message is from the topic we already subscribed to.
4. Now, if the match is found, the associated/corresponding subscription callback is called otherwise it says that an unsolicited publish was received.

The issue is as follows
Whenever I call my generic subscribeTopic function, I receive an acknowledgment from AWS IoT Cloud Broker and the subscribe request parameters such as topic, topic length, callback, and context are added to a Global Subscription List's 0th element by calling SubscriptionManager_AddSubscription. Now when I call subscribeTopic again, I receive acknowledgment and the subscribe request parameters are again added to Global Subscription List's 1st element but it also updates the 0th element's topic name to 1st element's topic name. Only the Global Subscription List's topic name is updated. Other parameters remain the same.

Calling subscribeTopic multiple times, I noticed that the Global Subscription List's topic name is updated to the latest subscribe request's topic name for every element in the list.

I am not able to find who is updating the Global Subscription List other than SubscriptionManager_AddSubscription function.

For getting to the same page ASAP, I am sharing the gist for coreMQTT_pegasus and pubsub components

**Describe the desired outcome**
Global Subscription List should be able to add multiple topics without updating the previously subscribed topic names. 
Ultimate Goal: To be able to subscribe to any topic using my generic subscribeTopic function from pubsub component.

**System information**
- Hardware board: esp32_wrover_kit
- IDE used: None
- Operating System: [ Linux ]
- Code version: my_AFR - 154a6aa5559e455e3b9d1f08c4e038d662f5a627
- Project/Demo: [ Custom Application ]

**Code/Steps to reproduce the current status**
coreMQTT_pegasus component's files : [https://gist.github.com/Prahar08modi/45eb5158fc0a810bf6d75a06817f3dec](url)
pubsub component's files: [https://gist.github.com/Prahar08modi/2d94d335be6344f20f5f173a05dbea72](url)

pvyawaha, 2021-08-19:
Hello @Prahar08modi ,

Thank you for sharing information about the issue you are facing. I am trying to reproduce it and check where the subscription table is updated.

Prahar08modi, 2021-08-20:
Hello @pvyawaha ,

Thanks for your reply. 

In my knowledge, the subscription table is updated by SubscriptionManager_AddSubscription function in my_AFR/demos/common/mqtt_subscription_manager.c

So, as suggested by you, I checked the subscription table by adding some printfs whenever I add a new subscription. You can find printf updated code in the latest revision of pubsub and mqtt_subscription_manager gist 

Also, I have added a new gist for mqtt_subscription_manager files.
mqtt_subscription_manager files: [https://gist.github.com/Prahar08modi/6decc94552b8d0dd993918139c5d2a01](url)

**Test Explanation**:
1. Wifi Connection Establishes
2. Connects to AWS IoT Cloud Broker using coreMQTT_pegasus component
3. Subscribe request is sent for test/topic32
4. After a successful first subscribe request, we wait for 10s and another subscribe request for test/topic425 is sent
5. After both subscriptions are successful, 
    a sample message 
        {
          "message": "Hello from AWS IoT console"
        }
    is published to test/topic32 and test/topic425

I'm sharing the log of the code: [log.txt](https://github.com/aws/amazon-freertos/files/7019541/log.txt)

**Observations**:

When there are no subscriptions in the subscription table, we get xAvailableIndex as 0. After 1st subscription, the topic is added at the 0th index in the subscription table as seen from the below image. Till now everything is fine. But just before sending another subscribe request, the 0th index of the subscription list got updated to the new topic name. The only topic name is updated, the topic length at the 0th index is still the same. Then again, in SubscriptionManager_AddSubscription function, new xAvailableIndex is found which is 1. So, the new subscribe request parameters are added to the 1st element of the subscription table. But, the 0th element's topic name still remains updated.
As a result of this, only the callback function associated with the latest subscribe request is called and for other subscribed topics, I get "Unsolicited publish was received". 

**For reference**:

[image]

When an incoming message is received:
 
[image]

[image]

The subscription table is updated by some part of the code. I can only find SubscriptionManager_AddSubscription updating it. I want to ask whether coreMQTT Agent can update it?

Thanks in advance.

Prahar08modi, 2021-08-25:
Hello @pvyawaha,

Were you able to reproduce the issue?