aws/amazon-freertos #3361: [BUG] system gets crash when tls try to reconnect to broker

Raghav3107, 2021-08-18:
**Describe the bug**
After getting the keep live timout error system to reconnect to  broker and at first attempt it connect but did not receive any data over network then its try to connect again. Now this time getting crash and throws the load prohibition error.
**System information**
- Hardware board: [ esp32 ]
- IDE used: [ .esp-idf ]
- Operating System: [ Linux  ]
- Code version: (v202107 )

**Screenshots or console output**

**First Scenario:**
```
136 33655 [MQTT Agent] [core_mqtt.c:886] [INFO] [MQTT] Packet received. ReceivedBytes=0.

137 40018 [MQTT Agent] [core_mqtt.c:2192] [ERROR] [MQTT] Exiting process loop due to failure: ErrorStatus=MQTTKeepAliveTimeout

138 40018 [MQTT Agent] [core_mqtt_agent.c:1023] [ERROR] [MQTT] MQTT operation failed with status MQTTKeepAliveTimeout

139 40018 [MQTT Agent] [mqtt_agent_task.c:877] [INFO] [MQTTDemo] Reconnected TCP. 

140 40018 [MQTT Agent] [mqtt_agent_task.c:786] [INFO] [MQTTDemo] Disconnecting TLS connection.

141 40018 [MQTT Agent] [mqtt_agent_task.c:763] [INFO] [MQTTDemo] Creating a TLS connection to a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com:8883.

142 40331 [MQTT Agent] [core_mqtt.c:886] [INFO] [MQTT] Packet received. ReceivedBytes=2.

143 40331 [MQTT Agent] [core_mqtt_serializer.c:970] [INFO] [MQTT] CONNACK session present bit not set.

144 40331 [MQTT Agent] [core_mqtt_serializer.c:912] [INFO] [MQTT] Connection accepted.

145 40331 [MQTT Agent] [core_mqtt.c:1563] [INFO] [MQTT] Received MQTT CONNACK successfully from broker.

146 40331 [MQTT Agent] [core_mqtt.c:1829] [INFO] [MQTT] MQTT connection established with the broker.

147 40331 [MQTT Agent] [mqtt_agent_task.c:573] [INFO] [MQTTDemo] Session present: 0

148 40331 [MQTT Agent] [mqtt_agent_task.c:619] [INFO] [MQTTDemo] Resubscribe to the topic $aws/things/+/jobs/# will be attempted.

153 40331 [MQTT Agent] [mqtt_agent_task.c:619] [INFO] [MQTTDemo] Resubscribe to the topic $aws/things/FW_A8032A6C7ABC/jobs/notify-next will be attempted.

154 40331 [MQTT Agent] [mqtt_agent_task.c:619] [INFO] [MQTTDemo] Resubscribe to the topic $aws/things/FW_A8032A6C7ABC/jobs/notify-next will be attempted.

155 40331 [MQTT Agent] [mqtt_agent_task.c:619] [INFO] [MQTTDemo] Resubscribe to the topic $aws/things/FW_A8032A6C7ABC/shadow/update/delta will be attempted.

156 40331 [MQTT Agent] [mqtt_agent_task.c:619] [INFO] [MQTTDemo] Resubscribe to the topic $aws/things/FW_A8032A6C7ABC/shadow/update/accepted will be attempted.

157 40331 [MQTT Agent] [mqtt_agent_task.c:619] [INFO] [MQTTDemo] Resubscribe to the topic $aws/things/FW_A8032A6C7ABC/shadow/update/rejected will be attempted.

158 40331 [MQTT Agent] [mqtt_agent_task.c:895] [INFO] [MQTTDemo] Reconnected to MQTT broker

159 40733 [MQTT Agent] [transport_secure_sockets.c:208] [ERROR] [Transport_Secure_Sockets] Failed to receive data over network. bytesReceived=-30848

160 40733 [MQTT Agent] [core_mqtt_serializer.c:2396] [ERROR] [MQTT] A single byte was not read from the transport: transportStatus=-30848.

161 40733 [MQTT Agent] [core_mqtt.c:1320] [ERROR] [MQTT] Receiving incoming packet length failed. Status=MQTTRecvFailed

162 40733 [MQTT Agent] [core_mqtt.c:2192] [ERROR] [MQTT] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed

163 40733 [MQTT Agent] [core_mqtt_agent.c:1023] [ERROR] [MQTT] MQTT operation failed with status MQTTRecvFailed

164 40733 [MQTT Agent] [mqtt_agent_task.c:877] [INFO] [MQTTDemo] Reconnected TCP. 

165 40733 [MQTT Agent] [mqtt_agent_task.c:786] [INFO] [MQTTDemo] Disconnecting TLS connection.

166 40733 [MQTT Agent] [mqtt_agent_task.c:763] [INFO] [MQTTDemo] Creating a TLS connection to a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com:8883.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d47c8  PS      : 0x00060830  A0      : 0x800e8829  A1      : 0x3fff7190  
[... 127 lines cut ...]

0x400d38a7: prvMQTTConnect at /home/horsemann/Desktop/WorkSpace/universalRemote_v202107/build/../src/mqtt_agent_task.c:577
 (inlined by) prvConnectToMQTTBroker at /home/horsemann/Desktop/WorkSpace/universalRemote_v202107/build/../src/mqtt_agent_task.c:948

0x400d3c4f: prvMQTTAgentTask at /home/horsemann/Desktop/WorkSpace/universalRemote_v202107/build/../src/mqtt_agent_task.c:884

```

I checked the Line no where its crash
```
        for( lIndex = 0; lIndex < pxSubscribeArgs->numSubscriptions; lIndex++ ) //  crash at this line 
        {
            /* This demo doesn't attempt to resubscribe in the event that a SUBACK failed. */
            if( pxReturnInfo->pSubackCodes[ lIndex ] == MQTTSubAckFailure )
            {
                LogError( ( "Failed to resubscribe to topic %.*s.",
                            pxSubscribeArgs->pSubscribeInfo[ lIndex ].topicFilterLength,
                            pxSubscribeArgs->pSubscribeInfo[ lIndex ].pTopicFilter ) );
                /* Remove subscription callback for unsubscribe. */
                removeSubscription( xGlobalSubscriptionList,
                                    pxSubscribeArgs->pSubscribeInfo[ lIndex ].pTopicFilter,
                                    pxSubscribeArgs->pSubscribeInfo[ lIndex ].topicFilterLength );
            }
        }
```

pvyawaha, 2021-08-19:
Hello @Raghav3107,

Thank you for reporting this , can you please share more information on how can I reproduce this ? Are you connecting to a hotspot and disabling data on it so that it goes in to Mqtt receive failed ?

Raghav3107, 2021-08-20:
hello @pvyawaha 

> Are you connecting to a hotspot and disabling data on it so that it goes in to Mqtt receive failed ?

I m using the hotspot but my data is not disable. You can see in log it make connection with broker after  that it not receive the packet and shows error `Status=MQTTRecvFailed`.

> can you please share more information on how can I reproduce this

Connect with  hotspot and first and after connecting to broker after some time disable the hotspot and on it again. 
As per program it should after connecting to hotspot again tls try to reconnnect and that time you may face this issue.

Hope this can generate same error.

Thanks

muneebahmed10, 2021-09-01:
Hi @Raghav3107,

The error sounds like memory is trying to be accessed out of scope with the dereference of `pxSubscribeArgs` in the condition of the `for` loop. That variable points to `xSubArgs` declared [here](https://github.com/aws/amazon-freertos/blob/cb0973379b8e014f53bdacdf548b11ccaa93e4d4/demos/coreMQTT_Agent/mqtt_agent_task.c#L616-L617), which was declared as `static`. Can you make sure that variable is still `static` in your code, and then also try moving the declaration of the two variables `xSubArgs` and `xSubInfo` outside of the function, and make them global variables?

Raghav3107, 2021-09-02:
Helllo @muneebahmed10 

> Can you make sure that variable is still `static` in your code,

Yeah! variable is the still  static, here you can see 

```
static MQTTStatus_t prvHandleResubscribe( void )
{
    MQTTStatus_t xResult = MQTTBadParameter;
    uint32_t ulIndex = 0U;
    uint16_t usNumSubscriptions = 0U;

    /* These variables need to stay in scope until command completes. */
    static MQTTAgentSubscribeArgs_t xSubArgs = { 0 };
    static MQTTSubscribeInfo_t xSubInfo[ SUBSCRIPTION_MANAGER_MAX_SUBSCRIPTIONS ] = { 0 };
    static MQTTAgentCommandInfo_t xCommandParams = { 0 };

    /* Loop through each subscription in the subscription list and add a subscribe
     * command to the command queue. */
    for( ulIndex = 0U; ulIndex < SUBSCRIPTION_MANAGER_MAX_SUBSCRIPTIONS; ulIndex++ )
    {
        /* Check if there is a subscription in the subscription list. This demo
         * doesn't check for duplicate subscriptions. */
        if( xGlobalSubscriptionList[ ulIndex ].usFilterStringLength != 0 )
        {
            xSubInfo[ usNumSubscriptions ].pTopicFilter = xGlobalSubscriptionList[ ulIndex ].pcSubscriptionFilterString;
            xSubInfo[ usNumSubscriptions ].topicFilterLength = xGlobalSubscriptionList[ ulIndex ].usFilterStringLength;

            /* QoS1 is used for all the subscriptions in this demo. */
            xSubInfo[ usNumSubscriptions ].qos = MQTTQoS1;

            LogInfo( ( "Resubscribe to the topic %.*s will be attempted.",
                       xSubInfo[ usNumSubscriptions ].topicFilterLength,
                       xSubInfo[ usNumSubscriptions ].pTopicFilter ) );

            usNumSubscriptions++;
        }
    }

    if( usNumSubscriptions > 0U )
    {
        xSubArgs.pSubscribeInfo = xSubInfo;
        xSubArgs.numSubscriptions = usNumSubscriptions;

        /* The block time can be 0 as the command loop is not running at this point. */
        xCommandParams.blockTimeMs = 0U;
        xCommandParams.cmdCompleteCallback = prvSubscriptionCommandCallback;
        xCommandParams.pCmdCompleteCallbackContext = ( void * ) &xSubArgs;

        /* Enqueue subscribe to the command queue. These commands will be processed only
         * when command loop starts. */
        xResult = MQTTAgent_Subscribe( &xGlobalMqttAgentContext, &xSubArgs, &xCommandParams );
    }
    else
    {
        /* Mark the resubscribe as success if there is nothing to be subscribed. */
        xResult = MQTTSuccess;
    }

    if( xResult != MQTTSuccess )
    {
        LogError( ( "Failed to enqueue the MQTT subscribe command. xResult=%s.",
                    MQTT_Status_strerror( xResult ) ) );
    }

    return xResult;
}
```

```

static void prvSubscriptionCommandCallback( MQTTAgentCommandContext_t * pxCommandContext,
                                            MQTTAgentReturnInfo_t * pxReturnInfo )
{
    size_t lIndex = 0;
    MQTTAgentSubscribeArgs_t * pxSubscribeArgs = ( MQTTAgentSubscribeArgs_t * ) pxCommandContext;

    /* If the return code is success, no further action is required as all the topic filters
     * are already part of the subscription list. */
    if( pxReturnInfo->returnCode != MQTTSuccess )
    {
        /* Check through each of the suback codes and determine if there are any failures. */
        for( lIndex = 0; lIndex < pxSubscribeArgs->numSubscriptions; lIndex++ )
        {
            /* This demo doesn't attempt to resubscribe in the event that a SUBACK failed. */
            if( pxReturnInfo->pSubackCodes[ lIndex ] == MQTTSubAckFailure )
            {
                LogError( ( "Failed to resubscribe to topic %.*s.",
                            pxSubscribeArgs->pSubscribeInfo[ lIndex ].topicFilterLength,
                            pxSubscribeArgs->pSubscribeInfo[ lIndex ].pTopicFilter ) );
                /* Remove subscription callback for unsubscribe. */
                removeSubscription( xGlobalSubscriptionList,
                                    pxSubscribeArgs->pSubscribeInfo[ lIndex ].pTopicFilter,
                                    pxSubscribeArgs->pSubscribeInfo[ lIndex ].topicFilterLength );
            }
        }
    }
}
```
Should I make this variable global ?
```
    static MQTTAgentSubscribeArgs_t xSubArgs = { 0 };
    static MQTTSubscribeInfo_t xSubInfo[ SUBSCRIPTION_MANAGER_MAX_SUBSCRIPTIONS ] = { 0 };
```

thanks for the update.!!

muneebahmed10, 2021-09-02:
Yes, please try making those variables global

Raghav3107, 2021-09-03:
Hello @muneebahmed10 
I tried by making the global variables but this issue did not resolved. I m uploading the log in text file. Might you will get better idea.
[Resubscribe_failed_log.txt](https://github.com/aws/amazon-freertos/files/7105512/Resubscribe_failed_log.txt)

Also system  crash at same point, I m upload the mqtt_agent file
[mqtt_agent_task.txt](https://github.com/aws/amazon-freertos/files/7105519/mqtt_agent_task.txt)

Raghav3107, 2021-10-08:
Hello @muneebahmed10   
Can you help on this?