aws/amazon-freertos #3362: [BUG] BLE WiFi Provisioning BSSID

AndreaPetrelli, 2021-08-19:
**Describe the bug**

Adding a new network with BLE provisioning fail with error:
Decode add network cbor message, status = 519

**System information**
- Hardware board: ESP32

The function that read the BSSID from message

```cpp
status = prvGetByteStringValueFromMap( &map, IOT_BLE_WIFI_PROV_BSSID_KEY, pAddNetworkReq->info.network.ucBSSID, &length );
```

has a incorrect length, the correct code should be

```cpp
length = sizeof( pAddNetworkReq->info.network.ucBSSID );
status = prvGetByteStringValueFromMap( &map, IOT_BLE_WIFI_PROV_BSSID_KEY, pAddNetworkReq->info.network.ucBSSID, &length );
```

pvyawaha, 2021-08-19:
Thank you for reporting this , we will raise a pull request to fix this and share update here.

ravibhagavandas, 2021-08-19:
Thanks for pointing this out, I have raised a PR to fix the same: https://github.com/aws/amazon-freertos/pull/3363

closed 2021-08-20 by ravibhagavandas