aws/amazon-freertos #3364: CoreMQTT mutual DEMO: connect OK.   Receive ERROR: MQTTRecvFailed

eelaix, 2021-08-20:
**Describe the bug**
CoreMQTT mutual auth DEMO: Code NEVER CHANGED
connect OK.   Receive ERROR: MQTTRecvFailed
democonfigNETWORK_BUFFER_SIZE  tested on 1024U. 2048U 10240U
The debug message in this post is based on 10240U

**System information**
- Hardware board: [ STM32H7 ]
- IDE used: [ AC6 ]
- Operating System: [ MacOS ]
- Code version: (amazon freertos 202107)
- Project/Demo: [ mqtt_demo_mutual_auth ]
- STM32H7 RMII with ETH and freertos_plus_tcp

**Expected behavior**

**Screenshots or console output**

1 3 [IP-task] PHY ID 7C0F0

2 53 [IP-task] xPhyReset: phyBMCR_RESET 0 ready

3 103 [IP-task] +TCP: advertise: 01E1 config 3100

4 103 [IP-task] prvEthernetUpdateConfig: LS mask 00 Force 1

7 1603 [IP-task] Autonego ready: 00000004: full duplex 100 mbit high status

8 1603 [EMAC-task] Network buffers: 56 lowest 56

9 1603 [IP-task] Link Status is high

10 3114 [EMAC-task] Network buffers: 56 lowest 55

11 3116 [IP-task] vDHCPProcess: offer c0a80164ip

12 3122 [IP-task] vDHCPProcess: offer c0a80164ip

13 4139 [iot_thread] [INFO ][DEMO][4139] ---------STARTING DEMO---------

14 4147 [iot_thread] [INFO ][INIT][4147] SDK successfully initialized.

15 4154 [iot_thread] [INFO ][DEMO][4154] Successfully initialized the demo. Network type for the demo: 4

16 4163 [iot_thread] [INFO] Creating a TLS connection to xxxxxxxxxxxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn:8883.
17 4175 [EMAC-task] Heap: current 236128 lowest 236128

18 4520 [EMAC-task] Heap: current 212328 lowest 212328

19 4625 [EMAC-task] Network buffers: 54 lowest 54

20 4625 [EMAC-task] RX descriptors 7/8

21 4625 [EMAC-task] Network buffers: 53 lowest 53

22 7292 [EMAC-task] Heap: current 192600 lowest 190720

23 9928 [iot_thread] [INFO] Creating an MQTT connection to xxxxxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
24 10106 [iot_thread] [INFO] Packet received. ReceivedBytes=2.
25 10112 [iot_thread] [INFO] CONNACK session present bit not set.
26 10127 [iot_thread] [INFO] Connection accepted.
27 10132 [iot_thread] [INFO] Received MQTT CONNACK successfully from broker.
28 10147 [iot_thread] [INFO] MQTT connection established with the broker.
29 10163 [iot_thread] [INFO] An MQTT connection is established with xxxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
30 10183 [iot_thread] [INFO] Attempt to subscribe to the MQTT topic 340351303333/example/topic.
31 10202 [iot_thread] [INFO] SUBSCRIBE sent for topic 340351303333/example/topic to broker.
32 10292 [iot_thread] [ERROR] Failed to receive data over network. bytesReceived=-30848
33 10309 [iot_thread] [ERROR] A single byte was not read from the transport: transportStatus=-30848.
34 10327 [iot_thread] [ERROR] Receiving incoming packet length failed. Status=MQTTRecvFailed
35 10345 [iot_thread] [ERROR] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed
36 10363 [iot_thread] [ERROR] MQTT_ProcessLoop failed to receive packet: Packet type=90, LoopDuration=1400, Status=MQTTRecvFailed
37 10383 [iot_thread] [ERROR] Failed to close connection: SOCKETS_Shutdown call failed. -95
38 10401 [iot_thread] [ERROR] SecureSocketsTransport_Disconnect() failed to close the network connection. StatusCode=4.
39 10421 [iot_thread] [INFO] Demo failed at iteration 1.
40 10435 [iot_thread] [INFO] Short delay before starting the next iteration.... 
41 15444 [iot_thread] [INFO] Creating a TLS connection to xxxxxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn:8883.
42 20624 [iot_thread] [INFO] Creating an MQTT connection to xxxxxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
43 20790 [iot_thread] [INFO] Packet received. ReceivedBytes=2.
44 20796 [iot_thread] [INFO] CONNACK session present bit not set.
45 20811 [iot_thread] [INFO] Connection accepted.
46 20825 [iot_thread] [INFO] Received MQTT CONNACK successfully from broker.
47 20832 [iot_thread] [INFO] MQTT connection established with the broker.
48 20848 [iot_thread] [INFO] An MQTT connection is established with xxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
49 20867 [iot_thread] [INFO] Attempt to subscribe to the MQTT topic 340351303333/example/topic.
50 20886 [iot_thread] [INFO] SUBSCRIBE sent for topic 340351303333/example/topic to broker.
51 20979 [iot_thread] [ERROR] Failed to receive data over network. bytesReceived=-30848
52 20996 [iot_thread] [ERROR] A single byte was not read from the transport: transportStatus=-30848.
53 21005 [iot_thread] [ERROR] Receiving incoming packet length failed. Status=MQTTRecvFailed
54 21013 [iot_thread] [ERROR] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed
55 21022 [iot_thread] [ERROR] MQTT_ProcessLoop failed to receive packet: Packet type=90, LoopDuration=1400, Status=MQTTRecvFailed
56 21034 [iot_thread] [INFO] Demo failed at iteration 2.
57 21039 [iot_thread] [INFO] Short delay before starting the next iteration.... 
58 26056 [iot_thread] [INFO] Creating a TLS connection to xxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn:8883.
59 31496 [iot_thread] [INFO] Creating an MQTT connection to xxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
60 31661 [iot_thread] [INFO] Packet received. ReceivedBytes=2.
61 31667 [iot_thread] [INFO] CONNACK session present bit not set.
62 31682 [iot_thread] [INFO] Connection accepted.
63 31687 [iot_thread] [INFO] Received MQTT CONNACK successfully from broker.
64 31702 [iot_thread] [INFO] MQTT connection established with the broker.
65 31718 [iot_thread] [INFO] An MQTT connection is established with xxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
66 31738 [iot_thread] [INFO] Attempt to subscribe to the MQTT topic 340351303333/example/topic.
67 31756 [iot_thread] [INFO] SUBSCRIBE sent for topic 340351303333/example/topic to broker.
68 31863 [iot_thread] [ERROR] Failed to receive data over network. bytesReceived=-30848
69 31880 [iot_thread] [ERROR] A single byte was not read from the transport: transportStatus=-30848.
70 31889 [iot_thread] [ERROR] Receiving incoming packet length failed. Status=MQTTRecvFailed
71 31906 [iot_thread] [ERROR] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed
72 31924 [iot_thread] [ERROR] MQTT_ProcessLoop failed to receive packet: Packet type=90, LoopDuration=1400, Status=MQTTRecvFailed
73 31954 [iot_thread] [ERROR] Failed to close connection: SOCKETS_Shutdown call failed. -95
74 31972 [iot_thread] [ERROR] SecureSocketsTransport_Disconnect() failed to close the network connection. StatusCode=4.
75 31991 [iot_thread] [INFO] Demo failed at iteration 3.
76 31997 [iot_thread] [INFO] Short delay before starting the next iteration.... 
77 37013 [iot_thread] [INFO] Demo run failed with 3 failed loops out of total 3 loops. RequiredSuccessCounts=2.
78 37023 [iot_thread] [INFO ][DEMO][37023] memory_metrics::freertos_heap::before::bytes::258528

79 37032 [iot_thread] [INFO ][DEMO][37032] memory_metrics::freertos_heap::after::bytes::186032

80 37050 [iot_thread] [INFO ][DEMO][37050] memory_metrics::demo_task_stack::before::bytes::10084

81 37068 [iot_thread] [INFO ][DEMO][37068] memory_metrics::demo_task_stack::after::bytes::6812

82 38082 [iot_thread] [ERROR][DEMO][38082] Error running demo.

83 38088 [iot_thread] [INFO ][INIT][38088] SDK cleanup done.

84 38103 [iot_thread] [INFO ][DEMO][38103] -------DEMO FINISHED-------

eelaix, 2021-08-20:
#define IOT_LOG_LEVEL_MQTT                      IOT_LOG_DEBUG
Logs IN DEBUG:

21 14409 [iot_thread] [INFO] Creating an MQTT connection to xxxxxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
22 14428 [iot_thread] [DEBUG] Encoded size for length 80 is 1 bytes.
23 14435 [iot_thread] [DEBUG] CONNECT packet remaining length=80 and packet size=82.
24 14451 [iot_thread] [DEBUG] CONNECT packet size is 82 and remaining length is 80.
25 14468 [iot_thread] [DEBUG] Encoded size for length 80 is 1 bytes.
26 14483 [iot_thread] [DEBUG] Length of serialized CONNECT packet is 82.
27 14491 [iot_thread] [DEBUG] BytesSent=82, BytesRemaining=0
28 14497 [iot_thread] [DEBUG] Successfully sent packet at time 10386.
29 14503 [iot_thread] [DEBUG] Sent 82 bytes of CONNECT packet.
30 14651 [iot_thread] [DEBUG] Encoded size for length 2 is 1 bytes.
31 14657 [iot_thread] [DEBUG] BytesReceived=2, BytesRemaining=0, TotalBytesReceived=2.
32 14674 [iot_thread] [INFO] Packet received. ReceivedBytes=2.
33 14689 [iot_thread] [INFO] CONNACK session present bit not set.
34 14695 [iot_thread] [INFO] Connection accepted.
35 14708 [iot_thread] [INFO] Received MQTT CONNACK successfully from broker.
36 14724 [iot_thread] [INFO] MQTT connection established with the broker.
37 14731 [iot_thread] [INFO] An MQTT connection is established with xxxxxxxxx.ats.iot.cn-north-1.amazonaws.com.cn.
38 14759 [iot_thread] [INFO] Attempt to subscribe to the MQTT topic 340351303333/example/topic.
39 14777 [iot_thread] [DEBUG] Encoded size for length 31 is 1 bytes.
40 14783 [iot_thread] [DEBUG] Subscription packet remaining length=31 and packet size=33.
41 14800 [iot_thread] [DEBUG] SUBSCRIBE packet size is 33 and remaining length is 31.
42 14817 [iot_thread] [DEBUG] Encoded size for length 31 is 1 bytes.
43 14832 [iot_thread] [DEBUG] Length of serialized SUBSCRIBE packet is 33.
44 14840 [iot_thread] [DEBUG] BytesSent=33, BytesRemaining=0
45 14855 [iot_thread] [DEBUG] Successfully sent packet at time 10735.
46 14870 [iot_thread] [DEBUG] Sent 33 bytes of SUBSCRIBE packet.
47 14876 [iot_thread] [INFO] SUBSCRIBE sent for topic 340351303333/example/topic to broker.
48 14945 [iot_thread] [ERROR] Failed to receive data over network. bytesReceived=-30848
49 14962 [iot_thread] [ERROR] A single byte was not read from the transport: transportStatus=-30848.
50 14980 [iot_thread] [ERROR] Receiving incoming packet length failed. Status=MQTTRecvFailed
51 14988 [iot_thread] [ERROR] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed
52 15006 [iot_thread] [ERROR] MQTT_ProcessLoop failed to receive packet: Packet type=90, LoopDuration=1400, Status=MQTTRecvFailed
53 15036 [iot_thread] [ERROR] Failed to close connection: SOCKETS_Shutdown call failed. -95
54 15053 [iot_thread] [ERROR] SecureSocketsTransport_Disconnect() failed to close the network connection. StatusCode=4.
55 15073 [iot_thread] [INFO] Demo failed at iteration 1.
56 15078 [iot_thread] [INFO] Short delay before starting the next iteration....

eelaix, 2021-08-20:
$FREERTOS/libraries/3rdparty/mbedtls_config/aws_mbedtls_config.h
Line 2221:
#define MBEDTLS_DEBUG_C

The Logs :

426 67856 [iot_thread] mbedTLS: |3| 0x2400da7c: handshake message: msglen = 4858, type = 11, hslen = 4858
427 67876 [iot_thread] mbedTLS: |2| 0x2400da7c: <= read record
428 67894 [iot_thread] mbedTLS: |3| 0x2400da7c: peer certificate #1:
429 67901 [iot_thread] mbedTLS: |3| 0x2400da7c: cert. version     : 3
430 67916 [iot_thread] mbedTLS: |3| 0x2400da7c: serial number     : 0E:7C:32:7E:14:13:78:63:D2:F4:17:03:34:3E:9B:C6
431 67936 [iot_thread] mbedTLS: |3| 0x2400da7c: issuer name       : C=US, O=Amazon, OU=Server CA 1B, CN=Amazon
432 67955 [iot_thread] mbedTLS: |3| 0x2400da7c: subject name      : CN=*.ats.iot.cn-north-1.amazonaws.com.cn
433 67974 [iot_thread] mbedTLS: |3| 0x2400da7c: issued  on        : 2021-03-10 00:00:00
434 67991 [iot_thread] mbedTLS: |3| 0x2400da7c: expires on        : 2022-02-18 23:59:59
435 68007 [iot_thread] mbedTLS: |3| 0x2400da7c: signed using      : RSA with SHA-256
436 68024 [iot_thread] mbedTLS: |3| 0x2400da7c: RSA key size      : 2048 bits
437 68040 [iot_thread] mbedTLS: |3| 0x2400da7c: basic constraints : CA=false
438 68047 [iot_thread] mbedTLS: |3| 0x2400da7c: subject alt name  : *.ats.iot.cn-north-1.amazonaws.com.cn
439 68075 [iot_thread] mbedTLS: |3| 0x2400da7c: key usage         : Digital Signature, Key Encipherment
440 68093 [iot_thread] mbedTLS: |3| 0x2400da7c: ext key usage     : TLS Web Server Authentication, TLS Web Client Authentication
441 68114 [iot_thread] mbedTLS: |3| 0x2400da7c: value of 'crt->rsa.N' (2048 bits) is:
442 68130 [iot_thread] mbedTLS: |3| 0x2400da7c:  b0 87 33 30 a8 30 bc 56 36 0d 2d da a4 70 7e cc
457 68389 [iot_thread] mbedTLS: |3| 0x2400da7c:  66 be c7 1b 5a d8 e0 9a 4c d6 64 a7 c7 1f 63 99
458 68406 [iot_thread] mbedTLS: |3| 0x2400da7c: value of 'crt->rsa.E' (17 bits) is:
459 68423 [iot_thread] mbedTLS: |3| 0x2400da7c:  01 00 01
460 68428 [iot_thread] mbedTLS: |3| 0x2400da7c: peer certificate #2:
461 68444 [iot_thread] mbedTLS: |3| 0x2400da7c: cert. version     : 3
462 68451 [iot_thread] mbedTLS: |3| 0x2400da7c: serial number     : 06:7F:94:57:85:87:E8:AC:77:DE:B2:53:32:5B:BC:99:8B:56:0D
463 68462 [iot_thread] mbedTLS: |3| 0x2400da7c: issuer name       : C=US, O=Amazon, CN=Amazon Root CA 1
464 68471 [iot_thread] mbedTLS: |3| 0x2400da7c: subject name      : C=US, O=Amazon, OU=Server CA 1B, CN=Amazon
465 68482 [iot_thread] mbedTLS: |3| 0x2400da7c: issued  on        : 2015-10-22 00:00:00
466 68499 [iot_thread] mbedTLS: |3| 0x2400da7c: expires on        : 2025-10-19 00:00:00
467 68516 [iot_thread] mbedTLS: |3| 0x2400da7c: signed using      : RSA with SHA-256
468 68524 [iot_thread] mbedTLS: |3| 0x2400da7c: RSA key size      : 2048 bits
469 68540 [iot_thread] mbedTLS: |3| 0x2400da7c: basic constraints : CA=true, max_pathlen=0
470 68557 [iot_thread] mbedTLS: |3| 0x2400da7c: key usage         : Digital Signature, Key Cert Sign, CRL Sign
471 68576 [iot_thread] mbedTLS: |3| 0x2400da7c: value of 'crt->rsa.N' (2048 bits) is:
472 68593 [iot_thread] mbedTLS: |3| 0x2400da7c:  c2 4e 16 67 dd ce bc 6a c8 37 5a ec 3a 30 b0 1d
487 68859 [iot_thread] mbedTLS: |3| 0x2400da7c:  d3 88 8b 4a 65 8c 4a d4 c3 2e 4c 9b 55 f4 86 e5
488 68877 [iot_thread] mbedTLS: |3| 0x2400da7c: value of 'crt->rsa.E' (17 bits) is:
489 68885 [iot_thread] mbedTLS: |3| 0x2400da7c:  01 00 01
490 68899 [iot_thread] mbedTLS: |3| 0x2400da7c: peer certificate #3:
491 68915 [iot_thread] mbedTLS: |3| 0x2400da7c: cert. version     : 3
492 68921 [iot_thread] mbedTLS: |3| 0x2400da7c: serial number     : 06:7F:94:4A:2A:27:CD:F3:FA:C2:AE:2B:01:F9:08:EE:B9:C4:C6
493 68951 [iot_thread] mbedTLS: |3| 0x2400da7c: issuer name       : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root 494 68974 [iot_thread] mbedTLS: |3| 0x2400da7c: subject name      : C=US, O=Amazon, CN=Amazon Root CA 1
495 68992 [iot_thread] mbedTLS: |3| 0x2400da7c: issued  on        : 2015-05-25 12:00:00
496 69009 [iot_thread] mbedTLS: |3| 0x2400da7c: expires on        : 2037-12-31 01:00:00
497 69026 [iot_thread] mbedTLS: |3| 0x2400da7c: signed using      : RSA with SHA-256
498 69042 [iot_thread] mbedTLS: |3| 0x2400da7c: RSA key size      : 2048 bits
499 69059 [iot_thread] mbedTLS: |3| 0x2400da7c: basic constraints : CA=true
500 69066 [iot_thread] mbedTLS: |3| 0x2400da7c: key usage         : Digital Signature, Key Cert Sign, CRL Sign
501 69084 [iot_thread] mbedTLS: |3| 0x2400da7c: value of 'crt->rsa.N' (2048 bits) is:
502 69101 [iot_thread] mbedTLS: |3| 0x2400da7c:  b2 78 80 71 ca 78 d5 e3 71 af 47 80 50 74 7d 6e
517 69333 [iot_thread] mbedTLS: |3| 0x2400da7c:  80 c4 80 4c 3e 3b 24 26 8e 04 ae 6c 9a c8 aa 0d
518 69351 [iot_thread] mbedTLS: |3| 0x2400da7c: value of 'crt->rsa.E' (17 bits) is:
519 69358 [iot_thread] mbedTLS: |3| 0x2400da7c:  01 00 01
520 69373 [iot_thread] mbedTLS: |3| 0x2400da7c: peer certificate #4:
521 69388 [iot_thread] mbedTLS: |3| 0x2400da7c: cert. version     : 3
522 69395 [iot_thread] mbedTLS: |3| 0x2400da7c: serial number     : A7:0E:4A:4C:34:82:B7:7F
523 69412 [iot_thread] mbedTLS: |3| 0x2400da7c: issuer name       : C=US, O=Starfield Technologies, Inc., OU=Starfield Class 2 Certification Authority
524 69443 [iot_thread] mbedTLS: |3| 0x2400da7c: subject name      : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root 525 69466 [iot_thread] mbedTLS: |3| 0x2400da7c: issued  on        : 2009-09-02 00:00:00
[... cut ...]
1231 84541 [iot_thread] [ERROR] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed
1232 84549 [iot_thread] [ERROR] MQTT_ProcessLoop failed to receive packet: Packet type=90, LoopDuration=1400, Status=MQTTRecvFailed
1233 84571 [iot_thread] [ERROR] Failed to close connection: SOCKETS_Shutdown call failed. -95
1234 84588 [iot_thread] [ERROR] SecureSocketsTransport_Disconnect() failed to close the network connection. StatusCode=4.
1235 84608 [iot_thread] [INFO] Demo failed at iteration 1.
1236 84623 [iot_thread] [INFO] Short delay before starting the next iteration....

muneebahmed10, 2021-09-01:
HI @eelaix,

In your logs, it looks like the subscribe acknowledgment is received and the callback invoked around line 1177. Then, roughly 90 seconds later, line 1210 shows that the server sends a close-notify since 1.5 times the 60 second keep alive interval has elapsed. It looks like mbedtls is getting blocked on a receive call after the SUBACK is received.

What hardware/network stack are you using?

eelaix, 2021-09-02:
> HI @eelaix,
>
> In your logs, it looks like the subscribe acknowledgment is received and the callback invoked around line 1177. Then, roughly 90 seconds later, line 1210 shows that the server sends a [...]

Hi, thanks for your reply.

My hardware is based on STM32H750 MCU , and I configured both with RMII eth and UART LTE module,

when I using RMII ETH everything runs ok,

but when I change to UART LTE (software is X-CUBE-CELLULAR 6.0.0) shows this.

NOW, I am trying to run under FreeRTOS-Cellular API and MQTT-AGENT,
I am  trying this solution to avoid this problem.

eelaix, 2021-09-04:
> HI @eelaix,
>
> In your logs, it looks like the subscribe acknowledgment is received and the callback invoked around line 1177. Then, roughly 90 seconds later, line 1210 shows that the server sends a [...]

Hi, @muneebahmed10 
When I integration CELLULAR API WITH FreeRTOS 2021.07,
I have found the SecureSocket abstraction library (202007, Branches->feature/cellular) does not work,
is there any solution ?

here is the log of secure socket library output: (blocked at the end)

9 19286 [iot_thread] [INFO ][xDEMO][19286] SIM card is READY

19 24475 [iot_thread] [INFO ][xDEMO][24475] State Machine Change oldState = REGISTERED, newState = CONNECTED ,trigger = PDN_ACTIVE

21 28451 [iot_thread] [INFO ][xDEMO][28451] Successfully initialized the demo. Network type for the demo: 8

22 28451 [iot_thread] [DEBUG][xnIOT][28451] Creating command queue.

23 28451 [iot_thread] [INFO ][xnIOT][28451] Creating a TLS connection to xxxx.ats.iot.cn-north-1.amazonaws.com.cn:8883.

24 28451 [iot_thread] [DEBUG][SeSkt][28451] Created CELLULAR Socket 0x2400e7c8.

25 29171 [iot_thread] [DEBUG][SeSkt][29170] Ip address 52.81.xxx.xx port 8883

26 29250 [iot_thread] [DEBUG][SeSkt][29250] Socket open callback on Socket 0x2400e7c8 604000680 5.

27 30789 [iot_thread] [DEBUG][SeSkt][30789] prvNetworkSend expect 168 write 168

28 30912 [iot_thread] [DEBUG][SeSkt][30912] Data ready on Socket 0x2400e7c8

29 30936 [iot_thread] [DEBUG][SeSkt][30936] prvNetworkRecv expect 5 read 5

30 30960 [iot_thread] [DEBUG][SeSkt][30960] prvNetworkRecv expect 91 read 91

31 30980 [iot_thread] [DEBUG][SeSkt][30979] prvNetworkRecv expect 5 read 5

closed 2021-09-24 by eelaix