aws/amazon-freertos #3368: [BUG] Failed to execute state transition handler

amiotk, 2021-08-24:
I'm running the latest AFR and have noticed the following error coming from ota.c just after connecting to AWS MQTT broker:
```
139 62664 [OTA Agent] [ota.c:2831] [ERROR] [OTA] Failed to execute state transition handler: Handler returned error: OtaErr_t=OtaErrJobParserErrorr
```

Although the error occurs every time the device connects to the broker, it doesn't seem to cause any damage, as the MQTT and OTA agent still happily run. The root cause seems to be the MQTT message received by the device:
```
{
"clientToken": ":thingname",
"timestamp":   1629735402
}
```

This message document doesn't contain some parameters required causing `verifyRequiredParamsExtracted()` to display the following traces: 
```
129 62650 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=execution
130 62651 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=execution.jobId
131 62653 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=execution.jobDocument
132 62654 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=execution.jobDocument.afr_ota
133 62655 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=execution.jobDocument.afr_ota.protocols
134 62657 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=execution.jobDocument.afr_ota.files
135 62659 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=filesize
136 62660 [OTA Agent] [ota.c:1738] [INFO] [OTA] Failed job document content check: Required job document parameter was not extracted: parameter=fileid
137 62662 [OTA Agent] [ota.c:1823] [INFO] [OTA] Failed to parse JSON document as AFR_OTA job: DocParseErr_t=7
138 62663 [OTA Agent] [ota.c:2240] [INFO] [OTA] No active job available in received job document: OtaJobParseErr_t=OtaJobParseErrNoActiveJobs
139 62664 [OTA Agent] [ota.c:2831] [ERROR] [OTA] Failed to execute state transition handler: Handler returned error: OtaErr_t=OtaErrJobParserError
140 62666 [OTA Agent] [ota.c:2839] [INFO] [OTA] Current State=[WaitingForJob], Event=[ReceivedJobDocument], New state=[CreatingFile]
```

Is it an expected behaviour? Any idea on how to fix it?

**System information**
- Hardware board: ESP32-Devkit-C
- Operating System: Windows
- Code version: 202107.00-11-g3af657034
- Project/Demo: ota_demo_core_mqtt

pvyawaha, 2021-08-24:
Hello @amiotk ,

The log shows that no active jobs were available to process and the state machine does not transition to requesting file blocks ( download ). This can be ignored and we will look into improving the logs.

amiotk, 2021-08-25:
Hello @pvyawaha 

Thanks for the comment. In such case, I will disable this log message.

closed 2021-08-25 by amiotk