aws/amazon-freertos #3376: [General] <How will I connect coreMQTT demo to tensorflow lite project?>

aditi230, 2021-08-31:
**Briefly summarize the issue being raised**
I have a running tensorflow lite project ,now I am willing to see that output on AWS Test Client through MQTT .
How should I do that?

**System information**
- Hardware board: [ESP 32 DevKitC]
- IDE used: [ Terminal]
- Operating System: [ Linux  ]
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [ coreMQTT]

pvyawaha, 2021-09-01:
Hello @aditi230 ,

You can submodule the coreMQTT library in your application https://github.com/FreeRTOS/coreMQTT and have the required transport interface based on mbedtls/lwip. You can refer to the transport interface details here https://freertos.org/network-interface.html#examples. If you are looking for thread safety for MQTT operations then you can use the coreMQTT Agent library which is thread safe https://github.com/FreeRTOS/coreMQTT-Agent.

aditi230, 2021-09-14:
Okay Thanks

closed 2021-09-16 by pvyawaha