aws/amazon-freertos #3378: [General] Docker build with WSL2

laszlopataki, 2021-09-02:
**Briefly summarize the issue being raised**
I am about to create a docker image to build my Makefile project that consumes AFR. 
I receive the following error message because of the line ending difference. 
```
/project/freertos/vendors/espressif/esp-idf/install.sh: line 2: $'\r': command not found
```

**Describe the desired outcome**
I am able to build a Makefile project that consumes AFR with a Docker image.
```
cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE="freertos/tools/cmake/toolchains/xtensa-esp32.cmake" -GNinja
```
Just like I can build an ESP-IDF project with `espressif/idf:release-v4.2` image from the Docker hub. 

**System information**
- Hardware board: ESP32
- IDE used: WSL2
- Operating System:  Windows
- Code version: 20210526_Archive-139-gb7151fe03

**Code/Steps to reproduce the current status**
Following the Getting Started guide I am consuming AFR in my Makefile project. 

**Questions**
Do you have any best practices? 
I am wondering if `*.sh text eol=lf` would be beneficial in the `.gitattributes`?

mingyue86010, 2021-09-06:
It looks your install.sh line ending character is messed up. It most likly your file is either ever opened or stored in windows environments and has a windows/DOS style of ending character (CR LF). You need convert if to Unix style ending character then I beleve the error will gone. Try the tool dos2unix. Here are some more details: https://stackoverflow.com/questions/11616835/r-command-not-found-bashrc-bash-profile/32912867

laszlopataki, 2021-09-07:
Hi  @mingyue86010, 

Thank you for your answer. 
I have windows 10 and my aim is to be able to build with docker. 

The target is the same as with the ESP-IDF [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html), but doing it in the AFR way explained [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html) at the "Using FreeRTOS in your own CMake project for ESP32" section. 

The problem is that if I use AFR in the explained way in the Getting Started Guide, then the included ESP-IDF under `/freertos/vendors/espressif/esp-idf` is part of the project. So the line ending in the project build files and in the executables won't be correct. 

Ideally, we have a docker image that is able to build an AFR project, similarly to the Espressif docker build solution: 
```
docker run --rm -v $PWD:/project -w /project aws/afr-idf:release-v4.0 cmake -S . -B build-directory -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja 
```

Do you have any docker image that is able to build a project which consumes AFR?

mingyue86010, 2021-09-08:
Hi @laszlopataki ,

Currently we do not provide a docker image to build the AFR project.

I get the code from 20210526_Archive can it look the install.sh has no problem running on Unix envirnoment. Regarding of your issue, it probably the git auto line ending conversion at checkout cause your problem. See https://docs.github.com/en/get-started/getting-started-with-git/configuring-git-to-handle-line-endings . Our project has a similar [.gitattributes](https://github.com/aws/amazon-freertos/blob/main/.gitattributes) file, which set the "text=auto". So when you checkout the code by git in Windows environment, it will convert the file with your native line ending (CR LF). However when you build the code in Unix environment which by your docker.

So you can just simply set "text=false" to prevent the conversion. More details can be found here:https://stackoverflow.com/questions/21822650/disable-git-eol-conversions

Hopfully this can resolve your problem.

Regards,
Ming

laszlopataki, 2021-09-28:
Hi, 

Thank you for your answer. 

I just wanted to share a dockerfile that I could use to build an AFR project. 
It is based on the docker image in the IDF project. 
I have tested it on Windows, but it should work on Linux as well. 

As a result of this, you are able to build with the same docker image on your local Windows/Linux PC and on the build server. 

```
docker build --rm --no-cache -t "john/afr-builder:v1" .

docker run --rm -v $PWD:/project -v /project/freertos -w /project john/aft-builder:v1 cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE="freertos/tools/cmake/toolchains/xtensa-esp32.cmake" -GNinja

docker run --rm -v $PWD:/project -v /project/freertos -w /project john/afr-builder:v1 idf.py build
```

```
FROM ubuntu:18.04

ARG DEBIAN_FRONTEND=noninteractive

# We need libpython2.7 due to GDB tools
RUN apt-get update && apt-get install -y \ 
    apt-utils \
    bison \
    ca-certificates \
    ccache \
    check \
    curl \
    flex \
    git \
    gperf \
    lcov \
    libncurses-dev \
    libusb-1.0-0-dev \
    make \
    ninja-build \
    libpython2.7 \
    python3 \
    python3-pip \
    unzip \
    wget \
    xz-utils \
    zip \
   && apt-get autoremove -y \
   && rm -rf /var/lib/apt/lists/* \
   && update-alternatives --install /usr/bin/python python /usr/bin/python3 10

RUN python -m pip install --upgrade pip virtualenv

# To build the image for a branch or a tag of IDF, pass --build-arg IDF_CLONE_BRANCH_OR_TAG=name.
# To build the image with a specific commit ID of IDF, pass --build-arg IDF_CHECKOUT_REF=commit-id.
# It is possibe to combine both, e.g.:
#   IDF_CLONE_BRANCH_OR_TAG=release/vX.Y
#   IDF_CHECKOUT_REF=<some commit on release/vX.Y branch>.

ARG AFR_CLONE_URL=https://github.com/aws/amazon-freertos.git
ARG AFR_CLONE_BRANCH_OR_TAG=release
ARG AFR_CHECKOUT_REF=
ARG AFR_PATH=/project/freertos

ARG IDF_CLONE_URL=https://github.com/espressif/esp-afr-sdk.git
ARG IDF_CLONE_BRANCH_OR_TAG=release/v4.2
ARG IDF_CHECKOUT_REF=
ARG IDF_INSTALL_PATH=/opt/esp/idf

ENV IDF_TOOLS_PATH=/opt/esp

RUN echo AFR_CHECKOUT_REF=$AFR_CHECKOUT_REF AFR_CLONE_BRANCH_OR_TAG=$AFR_CLONE_BRANCH_OR_TAG && \
    git clone --recursive \
      ${AFR_CLONE_BRANCH_OR_TAG:+-b $AFR_CLONE_BRANCH_OR_TAG} \
      $AFR_CLONE_URL $AFR_PATH && \
    if [ -n "$AFR_CHECKOUT_REF" ]; then \
      cd $AFR_PATH && \
      git checkout $AFR_CHECKOUT_REF && \
      git submodule update --init --recursive; \
    fi

RUN echo IDF_CHECKOUT_REF=$IDF_CHECKOUT_REF IDF_CLONE_BRANCH_OR_TAG=$IDF_CLONE_BRANCH_OR_TAG && \
    git clone --recursive \
      ${IDF_CLONE_BRANCH_OR_TAG:+-b $IDF_CLONE_BRANCH_OR_TAG} \
      $IDF_CLONE_URL $IDF_INSTALL_PATH && \
    if [ -n "$IDF_CHECKOUT_REF" ]; then \
      cd $IDF_INSTALL_PATH && \
      git checkout $IDF_CHECKOUT_REF && \
      git submodule update --init --recursive; \
    fi

# Install all the required tools, plus CMake. 
# Set IDF_PATH to the installer IDF for the time of install
ENV IDF_PATH=/opt/esp/idf
RUN $IDF_INSTALL_PATH/tools/idf_tools.py --non-interactive install required \
  && $IDF_INSTALL_PATH/tools/idf_tools.py --non-interactive install cmake \
  && $IDF_INSTALL_PATH/tools/idf_tools.py --non-interactive install-python-env 

# Set IDF_PATH to the build esp-afr-sdk
ENV IDF_PATH=$AFR_PATH/vendors/espressif/esp-idf

# Clean up 
RUN rm -rf $IDF_INSTALL_PATH \
 && rm -rf $IDF_TOOLS_PATH/dist

# Ccache is installed, enable it by default
ENV IDF_CCACHE_ENABLE=1

COPY entrypoint.sh /opt/esp/entrypoint.sh

ENTRYPOINT [ "/opt/esp/entrypoint.sh" ]
CMD [ "/bin/bash" ]
```

Regards,
Laszlo

closed 2021-09-28 by laszlopataki