aws/amazon-freertos #3379: [SecureSocket For CELLULAR] VERSION2020.07 can not work in 2021.07

eelaix, 2021-09-03:
Hi, friends.
When I integration CELLULAR API WITH  FreeRTOS 2021.07,
I have found the SecureSocket abstraction library does not work,
anybody can help me? thank you.

here is the log of secure socket library output: (blocked at the end)

9 19286 [iot_thread] [INFO ][xDEMO][19286] SIM card is READY

19 24475 [iot_thread] [INFO ][xDEMO][24475] State Machine Change oldState = REGISTERED, newState = CONNECTED ,trigger = PDN_ACTIVE

21 28451 [iot_thread] [INFO ][xDEMO][28451] Successfully initialized the demo. Network type for the demo: 8

22 28451 [iot_thread] [DEBUG][xnIOT][28451] Creating command queue.

23 28451 [iot_thread] [INFO ][xnIOT][28451] Creating a TLS connection to xxxx.ats.iot.cn-north-1.amazonaws.com.cn:8883.

24 28451 [iot_thread] [DEBUG][SeSkt][28451] Created CELLULAR Socket 0x2400e7c8.

25 29171 [iot_thread] [DEBUG][SeSkt][29170] Ip address 52.81.xxx.xx port 8883

26 29250 [iot_thread] [DEBUG][SeSkt][29250] Socket open callback on Socket 0x2400e7c8 604000680 5.

27 30789 [iot_thread] [DEBUG][SeSkt][30789] prvNetworkSend expect 168 write 168

28 30912 [iot_thread] [DEBUG][SeSkt][30912] Data ready on Socket 0x2400e7c8

29 30936 [iot_thread] [DEBUG][SeSkt][30936] prvNetworkRecv expect 5 read 5

30 30960 [iot_thread] [DEBUG][SeSkt][30960] prvNetworkRecv expect 91 read 91

31 30980 [iot_thread] [DEBUG][SeSkt][30979] prvNetworkRecv expect 5 read 5

eelaix, 2021-09-06:
=======================
Debug Message When "#define MBEDTLS_DEBUG_C" enabled at:
amazon-freertos/libraries/3rdparty/mbedtls_config/aws_mbedtls_config.h
source code line = 2221
=======================

1 0 [TmrSvc] vApplicationIPNetworkEventHook

4 1306 [iot_thread] [INFO ][xDEMO][1306] ---------STARTING DEMO---------

5 1308 [iot_thread] [INFO ][INIT][1308] SDK successfully initialized.

6 1608 [iot_thread] [ERROR][COMhw][1608] LTE MODEM Setup BaudRate=115200, MCUPOWER:1, PWRKEY:0

10 19278 [iot_thread] [INFO ][xDEMO][19278] SIM card is READY

20 23430 [iot_thread] [INFO ][xDEMO][23430] State Machine Change oldState = REGISTERED, newState = CONNECTED ,trigger = PDN_ACTIVE

21 27342 [iot_thread] [ERROR][xDEMO][27342] LTE.IMEI:865016050643683,CCID:89860621290004660000,sign:18,IP:10.135.96.xxx,NET:46001

22 27342 [iot_thread] [INFO ][xDEMO][27342] Successfully initialized the demo. Network type for the demo: 8

23 27342 [iot_thread] [DEBUG][xnIOT][27342] Creating command queue.

24 27342 [iot_thread] [INFO ][xnIOT][27342] Creating a TLS connection to x.ats.iot.cn-north-1.amazonaws.com.cn:8883.

25 27344 [iot_thread] [DEBUG][SeSkt][27344] Created CELLULAR Socket 0x24001170.

26 27873 [iot_thread] [DEBUG][SeSkt][27873] Ip address 52.80.33.xxx port 8883

27 27957 [iot_thread] [DEBUG][SeSkt][27957] Socket open callback on Socket 0x24001170 603984128 5.

28 29413 [iot_thread] mbedTLS: |2| 0x2400d5dc: => handshake

29 29413 [iot_thread] mbedTLS: |2| 0x2400d5dc: client state: 0

30 29413 [iot_thread] mbedTLS: |2| 0x2400d5dc: => flush output

31 29413 [iot_thread] mbedTLS: |2| 0x2400d5dc: <= flush output

32 29413 [iot_thread] mbedTLS: |2| 0x2400d5dc: client state: 1

33 29413 [iot_thread] mbedTLS: |2| 0x2400d5dc: => flush output

34 29413 [iot_thread] mbedTLS: |2| 0x2400d5dc: <= flush output

35 29424 [iot_thread] mbedTLS: |2| 0x2400d5dc: => write client hello

36 29424 [iot_thread] mbedTLS: |3| 0x2400d5dc: client hello, max version: [3:3]

37 29424 [iot_thread] mbedTLS: |3| 0x2400d5dc: dumping 'client hello, random bytes' (32 bytes)

38 29424 [iot_thread] mbedTLS: |3| 0x2400d5dc: 0000:  69 d1 32 e0 f8 2e 43 a3 fe bf 97 b5 c2 9f 07 9e  i.2...C.........

39 29426 [iot_thread] mbedTLS: |3| 0x2400d5dc: 0010:  1e de 54 6f 39 fa b2 8f 2d f6 37 8c 4d bb 22 8a  ..To9...-.7.M.".

40 29426 [iot_thread] mbedTLS: |3| 0x2400d5dc: clie

mingyue86010, 2021-09-06:
Hi @eelaix , so you are trying to integrate the code here https://github.com/aws/amazon-freertos/tree/feature/cellular/libraries/abstractions/cellular into the amazon-freeROTS 2021.07? Let me check with my team and find someone can answer your question on the cellular integration.

eelaix, 2021-09-07:
> Hi @eelaix , so you are trying to integrate the code here https://github.com/aws/amazon-freertos/tree/feature/cellular/libraries/abstractions/cellular into the amazon-freeROTS 2021.07? Let me check with my team and find someone can answer your question on the cellular integration.

yes, I am using these files:

**[CELLULAR API]**
https://github.com/aws/amazon-freertos/tree/feature/cellular/libraries/abstractions/cellular/include

**[COMMON 3GPP API]**
https://github.com/aws/amazon-freertos/tree/feature/cellular/vendors/aws/modules/3gpp/ports/cellular

**[SPECIAL DRIVER]**
https://github.com/aws/amazon-freertos/tree/feature/cellular/vendors/quectel/modules/bg96/ports/cellular

**[SecureSocket Library]**
https://github.com/aws/amazon-freertos/tree/feature/cellular/libraries/abstractions/secure_sockets/cellular

when I integrate these codes(2020.07)  into amazon-freeRTOS 2021.07
everything worked fine except secure socket library.

chinglee-iot, 2021-09-08:
Hi eelaix,

Thank you for your interest in feature cellular.
Further development of cellular project is moved to the following repositories.
**https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Demo
https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Library**
We suggest you reference these repositories for new updates.

The supported boards and cellular modules combinations in feature/cellular are listed below.
1. windows PC + quectel bg96
2. windows PC + sierra wireless hl7802
3. windows PC + ublox sara-r4
4. st stm32l475_discovery + quectel bg96
5. sierra wireless sensorhub + hl7802

From your first and second log, it looks like the log stop in mbedtls handshaking.
I suggest you use windows PC + your modem for comparision with your hardware device.

Please reference the https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_windows.html to setup windows device simulator.

To enable cellular modem with windows device simulator, apply following steps.
1. Add the following entry in cmake-gui
    BOARD_HAS_CELLULAR : to enable board cellular module setup
        Name : BOARD_HAS_CELLULAR
        Type : BOOL
        Value : SELECT
        Description : (Optional)
    SECURE_SOCKETS_CELLULAR : to use secure sockets cellular
        Name : SECURE_SOCKETS_CELLULAR
        Type : BOOL
        Value : SELECT
        Description : (Optional)
2. Select your cellular module in vendors/pc/boards/windows/CMakeLists.txt
Change this line if your modem is not quectel bg96.
include("${AFR_VENDORS_DIR}/**quectel**/modules/**bg96**/CMakeLists.txt")

3. Select the democonfigNETWORK_TYPES in vendors/pc/boards/windows/aws_demos/config_files/aws_demo_config.h
```
-#define democonfigNETWORK_TYPES                        ( AWSIOT_NETWORK_TYPE_ETH )
+#define democonfigNETWORK_TYPES                        ( AWSIOT_NETWORK_TYPE_CELLULAR )
```

4. Select the supported and enable network in vendors/pc/boards/windows/aws_demos/config_files/aws_iot_network_config.h
```
-#define configSUPPORTED_NETWORKS    ( AWSIOT_NETWORK_TYPE_ETH )
+#define configSUPPORTED_NETWORKS    ( AWSIOT_NETWORK_TYPE_ETH | AWSIOT_NETWORK_TYPE_CELLULAR )

-#define configENABLED_NETWORKS      ( AWSIOT_NETWORK_TYPE_ETH )
+#define configENABLED_NETWORKS      ( AWSIOT_NETWORK_TYPE_ETH | AWSIOT_NETWORK_TYPE_CELLULAR )
```

5. Select the com port of your cellular modem on PC in vendors/pc/boards/windows/aws_demos/config_files/aws_cellular_config.h
```
+#define cellularconfigCOMM_INTERFACE_PORT                   "<Your COM port on PC for example COM50>"
```
6. Setup cellular APN and other config in demos/include/aws_cellular_demo.h
```
#define configCELLULAR_APN                   "<your APN>"
```

7. (optional) In case you don't use ethernet on your PC, you can apply the following patch to skip ethernet setup.
```
diff --git a/vendors/pc/boards/windows/aws_demos/application_code/main.c b/vendors/pc/boards/windows/aws_demos/application_code/main.c
index ca734c25a..d6d344112 100644
--- a/vendors/pc/boards/windows/aws_demos/application_code/main.c
+++ b/vendors/pc/boards/windows/aws_demos/application_code/main.c
@@ -117,8 +117,19 @@ int main( void )
      * vApplicationIPNetworkEventHook() below).  The address values passed in here
      * are used if ipconfigUSE_DHCP is set to 0, or if ipconfigUSE_DHCP is set to 1
      * but a DHCP server cannot be contacted. */
-       FreeRTOS_printf(("FreeRTOS_IPInit\n"));
-       vApplicationIPInit();
+       // FreeRTOS_printf(("FreeRTOS_IPInit\n"));
+       // vApplicationIPInit();
+
+    /* A simple example to demonstrate key and certificate provisioning in
+     * microcontroller flash using PKCS#11 interface. This should be replaced
+     * by production ready key provisioning mechanism. */
+    vDevModeKeyProvisioning( );
+
+    /* Initialize AWS system libraries */
+    SYSTEM_Init();
+
+    /* Start the demo tasks. */
+    DEMO_RUNNER_RunDemos();

     /* Start the RTOS scheduler. */
     FreeRTOS_printf( ( "vTaskStartScheduler\n" ) );
```

The second suggestion is to enable the cellular debug log.
Apply the following patch to enable cellular debug log.
```
diff --git a/vendors/aws/modules/3gpp/ports/cellular/include/private/cellular_internal.h b/vendors/aws/modules/3gpp/ports/cellular/include/private/cellular_internal.h
index bd5587ce7..abe8a3305 100644
--- a/vendors/aws/modules/3gpp/ports/cellular/include/private/cellular_internal.h
+++ b/vendors/aws/modules/3gpp/ports/cellular/include/private/cellular_internal.h
@@ -28,12 +28,12 @@

 /* Configure logs for HAL functions. */
 #ifdef IOT_LOG_LEVEL_CELLULAR
-    #define LIBRARY_LOG_LEVEL        IOT_LOG_LEVEL_CELLULAR
+    #define LIBRARY_LOG_LEVEL        IOT_LOG_DEBUG
 #else
     #ifdef IOT_LOG_LEVEL_GLOBAL
-        #define LIBRARY_LOG_LEVEL    IOT_LOG_LEVEL_GLOBAL
+        #define LIBRARY_LOG_LEVEL    IOT_LOG_DEBUG
     #else
-        #define LIBRARY_LOG_LEVEL    IOT_LOG_INFO
+        #define LIBRARY_LOG_LEVEL    IOT_LOG_DEBUG
     #endif
 #endif

```

eelaix, 2021-09-09:
@chinglee-iot 

Thank you for your reply,
I will follow your advice and try again.

eelaix, 2021-09-13:
@chinglee-iot 
YES, everything worked perfect!  THANK YOU!

closed 2021-09-13 by eelaix