aws/amazon-freertos #3380: How to send data every second from AWS FreeRTOS Greengrass Discovery 202007 to GreenGrass Core (Raspberry Pi)

Yoimer, 2021-09-03:
Hi @pvyawaha  and everyone at AWS. Hope you're doing great.

Currently I am using the `202007.00` AWS FreeRTOS Discovery version. It works just fine.

However, the number of messages sent to AWS through the Raspberry pi has a definition here

[ggdDEMO_MAX_MQTT_MESSAGES  ](https://github.com/Yoimer/freertos_greengrass_202007/blob/de0022a388104d230311f470b8852125d76cd031/FreeRTOS/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c#L63).

I want to send data constantly, every second is fine. This could be changed later.

What would be the best way achieve this?

Could I just make an endless loop here? [ulMessageCounter](https://github.com/Yoimer/freertos_greengrass_202007/blob/de0022a388104d230311f470b8852125d76cd031/FreeRTOS/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c#L117) or would this break things?

Looking forward to seeing your comments.

Thanks in advance.

Yoimer, 2021-09-06:
Hello everyone. Are there any comments regarding this?. Thanks in advance.

mingyue86010, 2021-09-06:
Hi @Yoimer, thank you for your patient. I will forward you question in our team to the person who is expert on the Greengrass. They probably will get you back tomorrow after the vacation.

Yoimer, 2021-09-07:
@mingyue86010 Thanks a lot for you reply. Looking forward to reading your comments.

Yoimer, 2021-09-08:
Hello @mingyue86010 hope you're doing great. I was wondering if you have any comments about this request. Looking forward to hearing from you. Respectfully. Yoimer

muneebahmed10, 2021-09-08:
Hi @Yoimer,

You can change anything you want in that file, since it's just a demo. If you're going to make an infinite loop of publishes, though, you may want to add error handling code in case there's a disconnect or error with any of the publishes.

Additionally, if you're able to use a later version of the code, we would recommend trying to update, as the demo in 202007.00 uses the older MQTT library instead of the current coreMQTT library that is in the current release.

Yoimer, 2021-09-08:
@muneebahmed10 Thanks for responding. Does the AWS Greengrass on the latest version `202107.00` send data constantly like as an infinite loop?
Looking forward to hearing your comments.
Thanks in advance

muneebahmed10, 2021-09-09:
Hi @Yoimer,

As I mentioned in the earlier post, the demos are all example code, and you can change any part of them. If you want an infinite loop then you can just write an infinite loop.

Yoimer, 2021-09-09:
@muneebahmed10 Thanks for replying. I will try using the infinite loop and let you know. We already have a fully developed solution using AWS FreeRTOS `202007` for ESP32. Your team helped us a lot to achieve that.

Now we want to create a second solution and increment our clients. That's why we are testing now with AWS FreeRTOS GreenGrass Discovery.

dan4thewin, 2021-09-28:
Since it looks like the question was answered, I'll close this issue.  If this needs more attention, please open a new issue.  Thank you.

closed 2021-09-28 by dan4thewin