aws/amazon-freertos #3381: [General] how to do amazon-freertos certificate rotation at production?

Raghav3107, 2021-09-06:
Hello developers,

I was learning on the amazon-freertos certificate rotation, I read some blog [https://aws.amazon.com/blogs/iot/provisioning-with-a-bootstrap-certificate-in-aws-iot-core/](url) where  they explain how to do the certificate rotation and also found some code to implement on amazon-freertos [https://github.com/zanussbaum/Amazon-FreeRTOS-Certificate-Rotation/blob/10cccd16c5835a4fdaac8c05cdcb4e665c6f7535/demos/ota/csr.c](url).  But here [https://github.com/zanussbaum/Amazon-FreeRTOS-Certificate-Rotation/blob/10cccd16c5835a4fdaac8c05cdcb4e665c6f7535/demos/ota/csr.c](url) they saving the certificate at ram, if system reboot the certificate will gone. I want to save this certificate which i receive using the `custom ota `in nvs.

For that, I was searching how and from where amazon reads the certificate  so i can save this at that location. I found that amazon read the certificate from `cred` partition. The thing is the file which i receive in from `custom ota `that might me in `bin` format. But using the esp function I cannot write the bin file in `nvs cred partition.`

Question:

1. Little explanation on how amazon-freertes reads certificate from `nvs cred partition.`
2. How will I write the receive the file in nvs instead of ram.

**System information**
- Hardware board: [ ESP32 ]
- IDE used: [ esp-idf ]
- Operating System:  [ Linux ]
- Code version: (amazon-freertos v2019)
- Project/Demo: [ amazon-freertos certificate rotation ]

**References**

- [https://aws.amazon.com/blogs/iot/provisioning-with-a-bootstrap-certificate-in-aws-iot-core/](url)
- [https://github.com/zanussbaum/Amazon-FreeRTOS-Certificate-Rotation](url)

paulbartell, 2021-09-07:
Hello @Raghav3107 ,

The AWS IoT[ Fleet Provisioning service](https://docs.aws.amazon.com/iot/latest/developerguide/provision-wo-cert.html) provides some of the functionality you are looking for. An example of the service side of fleet provisioning written in python is available in [this repository](https://github.com/aws-samples/aws-iot-fleet-provisioning). This repository also has an example provisioning template which can be used for certificate rotation.

Our team has recently released a c library for interacting with the fleet-provisioning service. Some demos are available in the [AWS iot C-SDK repository](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/main/demos/fleet_provisioning/fleet_provisioning_with_csr) which is targeted at posix systems. This should give you a good starting point for how you can implement a certificate rotation solution on your FreeRTOS devices with a PKCS11 provider handling key and certificate storage.

Please let me know if you have any further questions and I will be happy to assist.

archigup, 2021-09-08:
Hey, regarding your first two questions, for reading the certs, corePKCS11 is used. To write them you would also use corePKCS11, which in the ESP32 port stores the data in nvs. An example for provisioning amazon-freertos credentials such as keys/certs using corePKCS11 can be found here: [https://github.com/aws/amazon-freertos/tree/main/demos/dev_mode_key_provisioning](https://github.com/aws/amazon-freertos/tree/main/demos/dev_mode_key_provisioning). You will likely have to convert the bin to PEM format.

For the third, as Paul mentioned above, AWS IoT has the Fleet Provisioning feature which could be used for fleet provisioning.

Raghav3107, 2021-09-15:
Hello @archigup @paulbartell 
Thanks for the reply.
@archigup, Right now I m receiving the certificate using the mqtt msg, and it is already in pem format. I checked the dev provisioning method, there it uses the default certificate of `aws_clientcredential.h` . Instead of the I tried to pass the certificate which i received the using the mqtt msg. 
```
char *private_key =  
-----BEGIN RSA PRIVATE KEY-----MIIEpAIBAAKCAQEAsIqY5P8Is+NDrgWAxC755nrwWsE2FAQ3Rj+GIVjEcliKmV19DCGxQustpUrH5mG6BhH+f/zsf9ylkC0i+sF1UYSkpCNl2sZweKnmYgKu2WHtxiJq9ZD6UbpwEv4lX9aUr3apRk5AblSNlgECO7gOM17h/m8hdybJH8jfscVttRRMx9OI6ksHrARDFnawdA+PH8lC6jTxqSVAmGvV/Q3TbhQcgZaXasS5kb6AxJZID034g4gmowxODCubPKCPo+xAP8HTyNYrO5ZLYufnXYeg3ZmdKGR/ndzZDFys0xw3PYQ3qHKQX3jfAebDKEyRcwalaoaFS3O5OTGGbSNmPJ+MOQIDAQABAoIBAHjbyWQr/Ic6NKEz/z1Pc6GH1HgtAZqecEDVSDwYCxTE7M2lpO/jOPk5C5y3Hzbm/HUfaH2yZy35nhaf7+x++JQAHL6JxTcK1wCkP6YldBI56o/UDCllQRkma4d1GZ0Mx+l/cM9CzvU5SVzCAAdmL8KXWFvZNK3fjZ0G5a1MHZjy0ycOJIwf9L2as5twNEX1g0Z8nfD4hDocKSJprDO+VbEYgGjnRzQOfsn4MTtYniUJlinRM2xwqbBz07XfhWIoPgnFFD2qxh0PQjnBCfMk9rqIfQsxfP6tffV6PRZmJmxASOZ41MzaSztzNeyktCujWfVNJOc2QyhFDYYaXIK1igECgYEA4fG4s5yPkgpF0/wjTP7XeXouCvMBYAPvwDSYSqki406FLrL6NJdaBtMomm0xRd4HtQoTc38TlXDFrX+ivD0ZOZ8ShjGrnr5rhj5GMHkf20NsBdls0VAuTcIv0Oik9iSfVuumhHdIH6Lt515xy2gIF4esHWUFb7Od0mJz5JNSiVkCgYEAyAaEfE1C8kYH34fbXHv/tv9tzCB5pEb1Aa6RmoUq9g+GEJfwXVREyJV8d68m39tBEHqLKZXQldFPYMsW4+enX/KL3phwQ7Snlacp4ZgOM6Mm9pZJuAxS6lsHfmSTdZ+PaJzs8HmU1vAPBhTQmeCi+ig95HUa+q+yUnzH74E73eECgYEAhCcZ3RQpdUXE6eSD0TkNVuP7ojPYDgy/YjBhEuoymbkHdPUCG1ibM3q2OPy84RyobV9jWR5ume3H8V7CFE+4PVepQgwzUToXMgGilB+EXn7k49b0Ksk3DAwUY6BvO/jXqcNTNg9+g+KMO9Olw08pzHlyVg8MSvVHzhH/ryXsFKECgYEAxTbn2Ua0AvtZN+NTPNRlQ6CEOQar6HStjIL5vEggG4siu6mSq/Ix/zKkDbz7rysV85FfNBkqFwzQHrceUoZMWCVkEoxp49dyg/LVMEAjgHYULHvxIy3Dcp1dvhib6xTu6E4Qruiw/1qg9XYAVjdNeSfEHkP33cLtJrcwycu5owECgYBkh5YP4WFn66jcrPzUJwc0uBA3orb/meWUnTyA0VxXkmc3RiQhHAIi7HePfXWWGhzAG/7jj4XjdI1A1rZZ1AhF2K16Bwhvo5Om7hpQw4IcNfPweKaXUDn3VgsmYcvGI2ucbOKY6z67BBy3QrrVjw6g1oeq6yH4XcmRuHPWJj3NYA==-----END RSA PRIVATE KEY-----
char* public_key =  
-----BEGIN CERTIFICATE-----MIIDMzCCAhsCFB0A7LD9wnEw7b9z4GQaIMHF3YoLMA0GCSqGSIb3DQEBCwUAMIGRMQswCQYDVQQGEwJJTjELMAkGA1UECAwCVFMxDDAKBgNVBAcMA0hZRDESMBAGA1UECgwJRmlyZXdpcmVzMRUwEwYDVQQLDAwgUiZEIFJPT1QgQ0ExGTAXBgNVBAMMEGlvdC5maXJld2lyZXMuaW4xITAfBgkqhkiG9w0BCQEWEmFkbWluQGZpcmV3aXJlcy5pbjAeFw0yMTA5MTQxMzMzNTZaFw0yMzAxMjcxMzMzNTZaMBoxGDAWBgNVBAMMD0ZXX0ZDRjVDNDZFODkzQzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALCKmOT/CLPjQ64FgMQu+eZ68FrBNhQEN0Y/hiFYxHJYipldfQwhsULrLaVKx+ZhugYR/n/87H/cpZAtIvrBdVGEpKQjZdrGcHip5mICrtlh7cYiavWQ+lG6cBL+JV/WlK92qUZOQG5UjZYBAju4DjNe4f5vIXcmyR/I37HFbbUUTMfTiOpLB6wEQxZ2sHQPjx/JQuo08aklQJhr1f0N024UHIGWl2rEuZG+gMSWSA9N+IOIJqMMTgwrmzygj6PsQD/B08jWKzuWS2Ln512HoN2ZnShkf53c2QxcrNMcNz2EN6hykF943wHmwyhMkXMGpWqGhUtzuTkxhm0jZjyfjDkCAwEAATANBgkqhkiG9w0BAQsFAAOCAQEAaaio6feD8zu+6pZ7ZOGebVCQ4A3P6Ug9rq2cSRKZK/9jcBooHaSehmPBuKNLbzfh7W5IUeh839fvcDXMdzUnXhi1/tXaM1Ya0/bsBSnAO0SAcTJXAA0JCfSyMzzDoJABtsuBjpobRuXAhAXc0XTZx1b5jcy7DlfmPcdYLr/JD/wuQeBVFiVDg97PV0PYHsyYQqViuOZXQGe8EUHGE6Nkn8mA5snwJev7B4OtckEtamrB8WKGxguaLbCVCnJTcIDeOkZr/oHGrelzTlOuMd3HIp6+HLoWyr36HR8cZA7jYsy5fgXHttdol18BqgOiOguVYtZCrfIPY4CgIYqIaC41Jg==-----END CERTIFICATE-----
```
```
xParams.pucJITPCertificate = (uint8_t *)keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM;
xParams.pucClientPrivateKey = (uint8_t *)private_key;
xParams.pucClientCertificate = (uint8_t *)public_key;
```

But this is not working, 
```
E (42178) mqtt.c: vCustomDevModeKeyProvisioning return error
97 4187 [iot_thread] Write certificate...
98 4187 [iot_thread] INFO: xProvisionCertificate() Success 0 
99 4187 [iot_thread] Unable to parse private key.
100 4187 [iot_thread] ERROR: Failed to provision device private key with status 7.
```
I check the error code  status 7,If i m not wrong its says wrong argument? 
I also, tried to convert this pem into der and pass it on structure. But it throw the same issue.
So, is the format wrong or some other thing?

muneebahmed10, 2021-09-17:
Hi @Raghav3107,

Please do not post credentials in a public forum. We have deleted them from your post for you, and your original post from the history. However, you should still ensure that you revoke these credentials and not reuse this certificate key pair.

From what you posted, it didn't look like your keys included newlines. PEM encoding requires newlines after every 64 characters in the body, and you should also have them after your `-----BEGIN RSA PRIVATE KEY-----` and before `-----END RSA PRIVATE KEY-----`. You can refer to this [python script](https://github.com/aws/amazon-freertos/blob/main/tools/certificate_configuration/PEM-to-C-string.py) to see how the C string should look like.

Can you try your workflow again after adding newlines to the strings?

Raghav3107, 2021-09-18:
Hello @muneebahmed10,
Thanks for the concern
The credential which I uploaded it was a dummy certificate, we are not using that one. In certificate there is newline characters, I thing I  upload the wrong format certificate. And do we need the quote after every new line character?
**Like this**
```
"-----BEGIN CERTIFICATE-----\n"	
"xxxxxxxxxxxxxxxxxxxxxxxxxxxx\n"
"yyyyyyyyyyyyyyyyyyyyyyyyyyyy\n"
"zzzzzzzzzzzzzzzzzzzzzzzzzzzz\n"
"-----END CERTIFICATE-----\n"
```
**Or Simple this will work**

```
-----BEGIN CERTIFICATE-----\n	
xxxxxxxxxxxxxxxxxxxxxxxxxxxx\n
yyyyyyyyyyyyyyyyyyyyyyyyyyyy\n
zzzzzzzzzzzzzzzzzzzzzzzzzzzz\n
-----END CERTIFICATE-----\n
```
Because, when I was testing the, this way able to embedded the certificate in nvs. So, I just want to make sure. 
```
Thanks for the help!
"-----BEGIN CERTIFICATE-----\n"	
"xxxxxxxxxxxxxxxxxxxxxxxxxxxx\n"
"yyyyyyyyyyyyyyyyyyyyyyyyyyyy\n"
"zzzzzzzzzzzzzzzzzzzzzzzzzzzz\n"
"-----END CERTIFICATE-----\n"
```

Raghav3107, 2021-09-20:
Hello, @muneebahmed10  thanks for the help,  
Now I am able to write on certificate using dev provisioning method in NVM.

closed 2021-09-20 by Raghav3107